JMRI/JMRI #5575: LE : Scale Track Tool does not work with Beziers

icklesteve, 2018-07-26:
The x and y values for the control points are unaffected by the scale operation.
All other track objects + the positional points all scale fine.

```
<tracksegment ident="T1" connect1name="EB1" type1="1" connect2name="EB2" type2="1" dashed="no" mainline="no" hidden="no" bezier="yes" hideConLines="no" class="jmri.jmrit.display.layoutEditor.configurexml.TrackSegmentXml">
      <controlpoints>
        <controlpoint index="0" x="50.0" y="80.0" />
        <controlpoint index="1" x="50.0" y="20.0" />
      </controlpoints>
    </tracksegment>
```

Current workaround ( if scaling is an absolute requirement, eg x-1 or y-1 scaling ), is to operate the panel via web server, which has not implemented beziers.

Translate track tool ( add or subtract ) works fine with the bezier control points, it's just the scale ( muliplication ) control points which are non-functioning,

Steve.

geowar1, 2018-07-26:
Fixed via PR #5578

icklesteve, 2018-07-27:
Thanks @geowar1 , confirm resolved.

closed 2018-07-27 by icklesteve