JMRI/JMRI #5584: Incorrect Tooltips when adding to Sensor, Turnout or Light Table

icklesteve, 2018-07-28:
System Name Tooltips are dependent on the top connection tab in the Sensor, Turnout or Light tables.

The tips work fine when All ( connection types ) are selected in the tab
, the connection dropdown select within Add new displays the correct tip when changed.

The issue occurs when the table tab is not sent to All ( Connections ).
The tooltip does not change when changing the drop-down select to a connection type, defaulting to the connection type of the open table tab.

Tested with multiple connection types with the same behaviour,
Steve.

icklesteve, 2018-11-08:
Issue was that hardware types with no range were not getting the tooltip updated.
The class type being used for the add new item validation different to class type used in the table table.
Also discovered issue with Reporter hardware type not being displayed correctly.
Moving to instance manager seems to have done the trick.

KenC57, 2019-05-24:
The OpenLCB tool tip for create are not showing valid form of event ids. It is showing "enter 1 to 9999" as a form and that's not right. Instead it should show a pair of events like: 02.01.57.00.00.54.00.72;02.01.57.00.00.54.00.73

icklesteve, 2019-05-24:
I was looking for a mini change to make to test something, good timing @KenC57 !
See #6987 wording ok?

silverailscolo, 2019-05-24:
If the default is not fit, for OpenLCB the default xManager tooltips should be overridden with specific format strings (in the OpenLCB Bundle). Please look at can.cbus or Acela.

icklesteve, 2019-06-01:
Closed by #6987

closed 2019-06-01 by icklesteve