JMRI/JMRI #5591: Switchboard panels limited input range

bobjacobsen, 2018-07-30:
The excellent "[Switchboard](http://jmri.org/help/en/package/jmri/jmrit/display/SwitchboardEditor.shtml)" panels seem to have a limited range of possible inputs:

```
    private int rangeMin = 1;
    private int rangeMax = 24;
    private int _range = rangeMax - rangeMin;
    private final JSpinner minSpinner = new JSpinner(new SpinnerNumberModel(rangeMin, rangeMin, rangeMax, 1));
    private final JSpinner maxSpinner = new JSpinner(new SpinnerNumberModel(rangeMax, rangeMin, rangeMax, 1));
```

C/MRI input and output numbers get very large; CT9031 is a typical value.  It would be good to be able to type in one of these large numbers.

On the other hand, you don't want to accidentally create a switchboard with thousands of elements.  So there's some logic needed to constrain the two values (min and max) to a reasonable interval.

silverailscolo, 2018-07-30:
I coded the spinners to jump to a next block (of say 24) items when you click the ``>>``. So the rangeMin and rangeMax are flexible and could also respond to the connection that is active. Switchboards are intended for quick and easy use. We might add a small icon to the interface that let's you enter any number you desire (and check the entry with the Manager).

bobjacobsen, 2018-08-02:
Thanks!  I hadn't understood what the "<" and ">" did (they're single arrows on the Mac).  But those are still a bit inconvenient a way to get up to 8032-8056, even 24 at a time.

Could we remove the upper limit from the spinners, and then have each one follow the other?

- If you change upper, then lower is checked and set to to be zero to 24 below

- If you change lower, then upper i schecked and set to to be zero to 24 above

So I could just type 8024 in lower, and have upper jump to a useful value (8056) while still being adjustable downward, or vice-versa.

pabender, 2018-08-02:
@bobjacobsen perhaps I am missing something in what you are trying to do here, but I thought you could set the specific turnout for each button in the switchboard?

In other words, unless you really have thousands of outputs you need buttons for, you don’t need thousands of buttons.

bobjacobsen, 2018-08-02:
Definitely don't want thousands of buttons!  Just 24 (or maybe 32). But I want to make buttons for CT8025 to CT8056 as easily as one can now make CT25 to CT56

silverailscolo, 2018-08-03:
@bobjacobsen suggested: "Could we remove the upper limit from the spinners, and then have each one follow the other?”
We sure could, but than depending on the connection type you might create out of bounds items.
You could just respond to that, instead of preventing it, using the bean mangers “maxturnoutnumber” field if set.

Would be an improvement in the UI if you could type in those larger numbers.
For MERG I decided to just support the +1;-2 format. So a -238;+21 Light can’t be controlled, currently.

As for the customisation per button @pabender mentions, that’s not what the Switchboard currently can do. Instead, it is more like a canvas full of buttons in a continuous range, some of which _might_ have an item defined in JMRI. If not, it shows dimmed. So it’s more a quick starter or test bed than your ultra compact keyboard.

We could provide the option to create the buttons from the Turnout/Light/Sensor table instead, but a new user is not helped as the table is empty at first.

New features like per key customisation would require to store individual properties for each key. You might even support dragging the buttons around like your apps on iPad and store their index. But for now only the range start and end values, the connection and the bean type (plus button shape and background color) are saved in the panel xml. The menu option to hide unconnected buttons just checks the bean. The Inverted property is passed on and stored in the bean itself.
The turnout icon is also deliberately generic, no LH/RH or Wye shapes. I wanted to present something at the start, and I think that is what makes Switchboards unique against the blank slate of the other panel types. But from there, feel free to create new uses.

Egbert

> Op 3 aug. 2018, om 00:07 heeft Paul Bender <notifications@github.com> het volgende geschreven:
> @bobjacobsen <https://github.com/bobjacobsen> perhaps I am missing something in what you are trying to do here, but I thought you could set the specific turnout for each button in the switchboard?
> 
> In other words, unless you really have thousands of outputs you need buttons for, you don’t need thousands of buttons.
>

silverailscolo, 2018-10-21:
In #6031 address range spinner limits can be set at will (up to 10.000) with an option AutoRange to keep the same number of items (default checked). Number of items within rangel is limited to 400, with a warning starting at 250. Adding 1000 switched caused a stack overflow.
By checking Hide unconnected you could still create a board with just 2 connected items in a range from 2000 - 3000, as requesed by user Jerry.

closed 2021-04-14 by stale[bot]