JMRI/JMRI #5599: openlcb.PackageTest hangs

bobjacobsen, 2018-08-02:
When run as a separate test group, `openlcb.PackageTest` hangs. See test output and thread dump below.  

This is completely reproducible in hash c9bf062e05b0516bb995460621bb00be558af921 Do `./runtest.csh jmri/jmrix/openlcb/PackageTest` to recreate.

More  [here](https://bugs.java.com/bugdatabase/view_bug.do?bug_id=4917751) on what might be happening to cause a conflict between AWTAutoShutdown and threading.

Output and trace:

```
JUnit version 4.12
Using org.netbeans.jemmy.drivers.APIDriverInstaller driver installer
.....................I........................................AWT blocker activation interrupted:
java.lang.InterruptedException
	at java.lang.Object.wait(Native Method)
	at java.lang.Object.wait(Object.java:502)
	at sun.awt.AWTAutoShutdown.activateBlockerThread(AWTAutoShutdown.java:350)
	at sun.awt.AWTAutoShutdown.notifyThreadBusy(AWTAutoShutdown.java:176)
	at java.awt.EventQueue$5.run(EventQueue.java:1107)
	at java.awt.EventQueue$5.run(EventQueue.java:1098)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.EventQueue.initDispatchThread(EventQueue.java:1097)
	at java.awt.EventQueue.postEventPrivate(EventQueue.java:287)
	at java.awt.EventQueue.postEvent(EventQueue.java:262)
	at java.awt.EventQueue.invokeAndWait(EventQueue.java:1341)
	at java.awt.EventQueue.invokeAndWait(EventQueue.java:1324)
	at javax.swing.SwingUtilities.invokeAndWait(SwingUtilities.java:1353)
	at jmri.implementation.AbstractSensor$1.run(AbstractSensor.java:133)
	at java.lang.Thread.run(Thread.java:748)
..........................................................................................................................................................................................................................................Could not add close window listener
WARN  - Cleaning up frame "Locomotive CDI Demonstration" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
WARN  - Cleaning up frame "Configuration Demonstration" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
WARN  - Cleaning up frame "After Message" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
.Could not add close window listener
WARN  - Cleaning up frame "Configuration Demonstration" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
WARN  - Cleaning up frame "3rd has PIP && 1st SNII" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
.Could not add close window listener
WARN  - Cleaning up frame "2nd has protocol id" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
.Could not add close window listener
WARN  - Cleaning up frame "TreePane Test" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
.Could not add close window listener
WARN  - Cleaning up frame "Prior Message" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
.Could not add close window listener
WARN  - Cleaning up frame "listener test" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
....II.

^\2018-08-02 07:50:51
Full thread dump Java HotSpot(TM) 64-Bit Server VM (25.181-b13 mixed mode):

"Olcb-Pool-49-Thread-24" #549 prio=5 os_prio=31 tid=0x00007fea3bc11000 nid=0x27b1f waiting on condition [0x000070000c51e000]
   java.lang.Thread.State: TIMED_WAITING (parking)
	at sun.misc.Unsafe.park(Native Method)
	- parking to wait for  <0x00000006c0400a28> (a java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject)
	at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:215)
	at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.awaitNanos(AbstractQueuedSynchronizer.java:2078)
	at java.util.concurrent.LinkedBlockingQueue.poll(LinkedBlockingQueue.java:467)
	at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1073)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1134)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

"Olcb-Pool-49-Thread-23" #548 prio=5 os_prio=31 tid=0x00007fea3ba0e000 nid=0x1b713 waiting on condition [0x000070000d9e6000]
   java.lang.Thread.State: TIMED_WAITING (parking)
	at sun.misc.Unsafe.park(Native Method)
	- parking to wait for  <0x00000006c0400a28> (a java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject)
	at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:215)
	at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.awaitNanos(AbstractQueuedSynchronizer.java:2078)
	at java.util.concurrent.LinkedBlockingQueue.poll(LinkedBlockingQueue.java:467)
	at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1073)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1134)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[... 837 lines cut ...]

"VM Thread" os_prio=31 tid=0x00007fea3c848800 nid=0x5203 runnable 

"GC task thread#0 (ParallelGC)" os_prio=31 tid=0x00007fea3c800800 nid=0x1d07 runnable 

"GC task thread#1 (ParallelGC)" os_prio=31 tid=0x00007fea3b80a800 nid=0x1b03 runnable 

"GC task thread#2 (ParallelGC)" os_prio=31 tid=0x00007fea3b80b000 nid=0x2b03 runnable 

"GC task thread#3 (ParallelGC)" os_prio=31 tid=0x00007fea3d000000 nid=0x5303 runnable 

"VM Periodic Task Thread" os_prio=31 tid=0x00007fea3c810800 nid=0x4403 waiting on condition 

JNI global references: 851

Heap
 PSYoungGen      total 82944K, used 11428K [0x000000076ab00000, 0x0000000770300000, 0x00000007c0000000)
  eden space 75776K, 13% used [0x000000076ab00000,0x000000076b501138,0x000000076f500000)
  from space 7168K, 16% used [0x000000076fc00000,0x000000076fd28000,0x0000000770300000)
  to   space 7168K, 0% used [0x000000076f500000,0x000000076f500000,0x000000076fc00000)
 ParOldGen       total 65536K, used 9867K [0x00000006c0000000, 0x00000006c4000000, 0x000000076ab00000)
  object space 65536K, 15% used [0x00000006c0000000,0x00000006c09a2fb8,0x00000006c4000000)
 Metaspace       used 32013K, capacity 32904K, committed 33024K, reserved 1077248K
  class space    used 4100K, capacity 4295K, committed 4352K, reserved 1048576K
```

pabender, 2018-08-03:
I can't replicate this on my Linux machine.

Before #5595 was merged, I was seeing this:

```
bash-4.4$ ./runtest.csh jmri.jmrix.openlcb.PackageTest
JUnit version 4.12
.Using org.netbeans.jemmy.drivers.DefaultDriverInstaller driver installer
....................I.................................................................................................................................................................................................................................................................................Could not add close window listener
WARN  - Cleaning up frame "Locomotive CDI Demonstration" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
WARN  - Cleaning up frame "Configuration Demonstration" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
WARN  - Cleaning up frame "After Message" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
.Could not add close window listener
WARN  - Cleaning up frame "3rd has PIP && 1st SNII" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
.Could not add close window listener
WARN  - Cleaning up frame "2nd has protocol id" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
.Could not add close window listener
WARN  - Cleaning up frame "TreePane Test" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
.Could not add close window listener
WARN  - Cleaning up frame "Prior Message" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
.Could not add close window listener
WARN  - Cleaning up frame "listener test" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
....II........................................WARN  - Pool did not terminate [main] org.openlcb.can.CanInterface.dispose()
............
Time: 101.727

OK (355 tests)
```

After #5595 was merged I am seeing this when I run the package tests locally:

```
bash-4.4$ ./runtest.csh jmri.jmrix.openlcb.PackageTest
JUnit version 4.12
Using org.netbeans.jemmy.drivers.DefaultDriverInstaller driver installer
.....................I........................................AWT blocker activation interrupted:
java.lang.InterruptedException
	at java.lang.Object.wait(Native Method)
	at java.lang.Object.wait(Object.java:502)
	at sun.awt.AWTAutoShutdown.activateBlockerThread(AWTAutoShutdown.java:350)
	at sun.awt.AWTAutoShutdown.notifyThreadBusy(AWTAutoShutdown.java:176)
	at java.awt.EventQueue$5.run(EventQueue.java:1107)
	at java.awt.EventQueue$5.run(EventQueue.java:1098)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.EventQueue.initDispatchThread(EventQueue.java:1097).
	at java.awt.EventQueue.postEventPrivate(EventQueue.java:287)
	at java.awt.EventQueue.postEvent(EventQueue.java:262)
	at java.awt.EventQueue.invokeAndWait(EventQueue.java:1341)
	at java.awt.EventQueue.invokeAndWait(EventQueue.java:1324)
	at javax.swing.SwingUtilities.invokeAndWait(SwingUtilities.java:1353)
	at jmri.implementation.AbstractSensor$1.run(AbstractSensor.java:133)
	at java.lang.Thread.run(Thread.java:748)
.........................................................................................................................................................................................................................................Could not add close window listener
WARN  - Cleaning up frame "Configuration Demonstration" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
WARN  - Cleaning up frame "After Message" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
.WARN  - Cleaning up frame "3rd has PIP && 1st SNII" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
.WARN  - Cleaning up frame "2nd has protocol id" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
.Could not add close window listener
WARN  - Cleaning up frame "TreePane Test" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
.Could not add close window listener
WARN  - Cleaning up frame "Prior Message" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
.Could not add close window listener
WARN  - Cleaning up frame "listener test" (a class javax.swing.JFrame) in jmri.jmrix.openlcb.swing.networktree.TreePaneDemo. [main] jmri.util.JUnitUtil.resetWindows()
.Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at java.awt.Component$BltBufferStrategy.contentsLost(Component.java:4479)
	at javax.swing.BufferStrategyPaintManager.flushAccumulatedRegion(BufferStrategyPaintManager.java:413)
.	at javax.swing.BufferStrategyPaintManager.endPaint(BufferStrategyPaintManager.java:377)
[... cut ...]
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
.II..............................................WARN  - Pool did not terminate [main] org.openlcb.can.CanInterface.dispose()
..............
Time: 119.288

OK (364 tests)

bash-4.4$ 
```

So #5595 appears to have changed something that is causing a couple of exceptions, but I don't see a hang.

pabender, 2018-08-09:
@bobjacobsen can you check and see if you still see the hanging behavior now that #5627 has been merged?

You probably will still see the InterruptedException.

bobjacobsen, 2018-08-09:
Seems to be running fine, thanks!

Yes, the InterruptedException exception is still there.

Bob

> On Aug 9, 2018, at 11:14 AM, Paul Bender <notifications@github.com> wrote:
> 
> @bobjacobsen can you check and see if you still see the hanging behavior now that #5627 has been merged?
> 
> You probably will still see the InterruptedException.

rhwood, 2020-05-12:
This appears to have been addressed.

closed 2020-05-12 by rhwood