JMRI/JMRI #5615: Some translated items still show in system language

bobjacobsen, 2018-08-06:
From various jmriusers comments, some items (that have had their strings internationalized) are still showing in the default Locale.  Typically, this is perferences menus and other items that are created early, probably before the Locale has been set.

silverailscolo, 2018-08-06:
Nice improvements! In build 2619 the Preferences pane list to the left fully translated:
[image]

In the macOS application menu the default (US English) locale items still show when Locale in JMRi set to German/Dutch:
[image]

Not sure anything can be done about that by Java once it runs. Perhaps only by adding some resources to the macOS DecoderPro application package, since other apps show the OS locale.

bobjacobsen, 2018-08-06:
Thanks!  Will look into translating the Mac menu items, might be an interesting challenge.  In the JMRI preferences window, should "JSON Server" and "SRCP Server" be translated?

silverailscolo, 2018-08-08:
I see little added value as the word Server is well understood and using a synonym will make the function harder to find from the help.

Egbert

bobjacobsen, 2018-08-10:
Sorry, I probably wasn’t clear.  Happy to have them translated or not, just wanted to know if there was a bug on those items that was preventing translated text from being displayed.

> On Aug 8, 2018, at 10:38 AM, Egbert Broerse <notifications@github.com> wrote:
> 
> I see little added value as the word Server is well understood and using a synonym will make the function harder to find from the help.
> 
> Egbert
> 
> > Op 7 aug. 2018, om 00:59 heeft Bob Jacobsen <notifications@github.com> het volgende geschreven:
> > 
> > Thanks! Will look into translating the Mac menu items, might be an interesting challenge. In the JMRI preferences window, should "JSON Server" and "SRCP Server" be translated?

silverailscolo, 2018-08-10:
No sorry, those items come across fine if/when a key is in the props file.

closed 2018-08-18 by bobjacobsen