JMRI/JMRI #5618: strClass  jmri.util.JmriJFrame OK? Root cause of Hardware save window size issues?

icklesteve, 2018-08-06:
Issue affecting : Auto-save Window sizes
Possibly affecting : Help page address links used by getHelpTarget() having to use the long link?

While having a look at issue #4859 ( Window size preferences in a MERG CBUS Tool + see OpenLCB Issue #3091 ), I found that it wasn't a tool class specific issue, then not a hardware MERG CBUS issue.

I then added a log.debug to setWindowSize in jmri/managers/JmriUserPreferencesManager.java:542 to look at the strClass. 
This is consistently "jmri.util.JmriJFrame" for the affected screens.

Also shows  "jmri.util.JmriJFrame" for windowFrameRef in JmriJFrame.java:881

I proceeded to test with MERG CBUS, Loconet Simulator + OpenLCB Simulator, with a few positives.

Was thinking perhaps frames / panes ?

Working when added to user-interface.xml

### Working
**Loconet Simulator**
Startup Window jmri.jmrix.loconet.hexfile.HexFileFrame
Monitor Slots	jmri.jmrix.loconet.locomon.LocoMonPane.FilterField
Loconet over TCP Server	jmri.jmrix.loconet.loconetovertcp.LnTcpServerFrame:LocoNet over TCP Server

**OpenLCB Sim**
None

**MERG Sim**
Event Filter Highlighter ( Popout window of MERG CBUS Console Log )) jmri.jmrix.can.cbus.swing.console.CbusEventFilterFrame

### Not Working
**Loconet Sim**
Monitor
Monitor Clock
Stats
Send Trottle Message
Send Loconet Packet
Select PR3 node
Firmware Downloader
Download Sounds
Config BDL16
Config LocoIO
Config PM4
Config Command Station
Config Loconet ID
Config Duplex Group
Configure DS64

**OpenLCB Sim**
Monitor ( 
Key + value "java.lang.String" inserted by jmri.jmrix.openlcb.swing.monitor.MonitorPane.FilterField
 )
Send Frame
Configure Node Network Tree
Hub Control
Firmware Downloader

**MERG CBUS**
Console Log
Send Frame
Event Capture
Event Table
Node Config

May be easier for someone to do a full list of affected pages via a local ant test?

Is there a one-liner I could add to the non-working ones or would that just be a sticking plaster over a potentially bigger issue?

Over to you folks,
Steve.

Pugwash1, 2018-08-12:
Something like this may get you somewhere. A lot of those frames listed are using a JmriPanel which creates a frame and does caching, so everything finish up with out name, which is what preferences uses. What happens with multiple instances is that the last window closed gets its size and position saved. 

```diff --git a/java/src/jmri/util/swing/sdi/JmriJFrameInterface.java b/java/src/jmri/util/swing/sdi/JmriJFrameInterface.java
index 0150b48348..25bc66bfe1 100644
--- a/java/src/jmri/util/swing/sdi/JmriJFrameInterface.java
+++ b/java/src/jmri/util/swing/sdi/JmriJFrameInterface.java
@@ -36,7 +36,7 @@ public class JmriJFrameInterface implements jmri.util.swing.WindowInterface {
         }
          // create frame
-        frame = new jmri.util.JmriJFrame();
+        frame = new jmri.util.JmriJFrame(child.getClass().getName());
         // cache if single instance
         if (!child.isMultipleInstances()) {
```

Steve G.

icklesteve, 2018-08-24:
Thanks to @Pugwash1 and @devel-bobm 
The window sizes and positions I've tested all re-open as I left them.
Auto launch window action on startup looks loads better,
Steve

closed 2018-08-24 by icklesteve