JMRI/JMRI #5633: RaspberryPiShutdown.py not finding shutDownManagerInstance

mstevetodd, 2018-08-10:
Running https://github.com/JMRI/JMRI/blob/master/jython/RaspberryPiShutdown.py from a button, I get this error:

> 2018-07-06 02:53:25,798 script.JmriScriptEngineManager        INFO  - ECMAScript ECMA - 262 Edition 5.1 is provided by Oracle Nashorn 1.8.0_65 [AWT-EventQueue-0]
> 2018-07-06 02:53:25,818 script.JmriScriptEngineManager        INFO  - python 2.7 is provided by jython 2.7.0 [AWT-EventQueue-0]
> 2018-07-06 02:54:06,226 jython.RunJythonScript                ERROR - Unable to execute script. [AWT-EventQueue-0]
> javax.script.ScriptException: AttributeError: type object 'jmri.InstanceManager' has no attribute 'shutDownManagerInstance' in <script> at line number 16
> 	at org.python.jsr223.PyScriptEngine.scriptException(PyScriptEngine.java:202)
> 	at org.python.jsr223.PyScriptEngine.eval(PyScriptEngine.java:42)
> 	at org.python.jsr223.PyScriptEngine.eval(PyScriptEngine.java:47)
> 	at javax.script.AbstractScriptEngine.eval(AbstractScriptEngine.java:249)
> 	at jmri.script.JmriScriptEngineManager.eval(JmriScriptEngineManager.java:316)
> 	at jmri.jmrit.jython.RunJythonScript.invoke(RunJythonScript.java:111)
> 	at jmri.jmrit.jython.RunJythonScript.actionPerformed(RunJythonScript.java:84)

So shutDownManagerInstance has been removed. What has replaced it?

bobjacobsen, 2018-08-10:
There’s a predefined “shutdown” accessor, which also shows how to reference the manager:

jython//jmri_bindings.py:shutdown     = jmri.InstanceManager.getDefault(jmri.ShutDownManager)

> On Aug 10, 2018, at 12:31 PM, mstevetodd <notifications@github.com> wrote:
> 
> So shutDownManagerInstance has been removed. What has replaced it?
>

mstevetodd, 2018-08-11:
Thanks, @bobjacobsen 
Changed per your suggestion, plus added a restart version.
Closing via #5634

closed 2018-08-11 by mstevetodd