JMRI/JMRI #5656: Tracks and turnouts not loading properly, and Logix not acting logically.

Swisschris52, 2018-08-17:
I've just re-opened my layout and discovered that some of the tracks, and turnouts had greyed out as shown in the screen shot. I've tried to edit the turnout by right clicking, but the edit box does not open or anything else for that matter. All tracks and turnouts had been tested for simple logic on the signals before closing, with no problems, I did a save before closing as well. How do i get round this without entering in the whole lot again.

On the right hand side of the screen,there is a signal with lit head beside it. I set that head to be lit using logix only when the mast was Amber and the turnout at the bottom right of the image is closed. Neither state is true when the head lights.. I appreciate that someone might want to see the xml file, and will send it to whoever.

[image]

bobjacobsen, 2018-08-17:
With the program running, please go to Help -> Upload Debugging Info and fill out that form to send in your profile, logs and panel file.  Then mention here (in a comment) that you did that so we can link them up.  (Also, please mention in the Debug Info that this is for Issue 5656) Thanks!

geowar1, 2018-08-17:
1st off you may want to make your edit/control circles smaller; You can do that via the “Options”, “Turnout Optionss”,  “Set Turnout is le Size” menu.
The tracks are colored by block and there’s a default color for tracks without blocks (IIRC it’s grey). If you contextually (right- or control-) click on a gray’d out track segment and choose “Edit”. A block should be specified in the “Block Name” combobox. If one is there click the “Create/Edit Block” button and in the “Edit xxx Block” dialog that appears select the “Layout Editor” tab and set the track colors (for unoccupied, occupied & allocated).
You can change the default (no block) track color by selecting “Options”, “Track Options”, “Default Track Colors” submenus.

geowar1, 2018-08-17:
You may also want to use the "Tools", "Check" sub-menus to find and fix any un-connected tracks, un-blocked tracks, non-contiguous blocks and un-necessary anchors.

Swisschris52, 2018-08-17:
I've now made the circles smaller. Thanks for that. When I created that particular set of blocks, I had created the name, set the colour, and allocated sensors to the blocks and turnouts. As I set the colours, I could then see any sections (Turnout switches) that I'd not allocated properly. I then added the signals and set the SML (Signal Mast Logic) for all with a view to coming back to Logix for any feather / split heads. Each Mast was tested so that it gave me the result I wanted (a few failed because I'd made a mistake which I then corrected). A few failed even though the SML was correct.
I closed down JMRI and came back to it later to see the panel as shown in the screenshot.  I could not right click on the turnouts, although I could throw them, and the signal masts did not show any change of state. The grey blocks had been created and sensors added to them and the colour for that area had been set before closing. 2 of the greyed out blocks were showing no name, and one was showing no sensor, si I re-named the blocks, the block name showing as having been created, and re-added the missing sensor.
The grey turnouts were showing the correct position and the sensors were inactive, so I should have shown a proceed or caution, but a stop was displayed.
Before the problem with the panel load, and also since the problem corrected itself, see image
[image]
I've also been having trouble with the Logix of the 3 way. Logix was used so that both sides could not be in the thrown position simultaneously. It tends to bend the turnout blades when that happens. Selecting the left throw brought up the error. If you check the image, the tracks show paths where the masts should show a proceed aspect, but don't. I have got around this by manually setting the turnouts to closed, then throwing the required side. This the shows all the correct aspects.

closed 2021-04-14 by stale[bot]