JMRI/JMRI #5657: Programming Accesoryes on the main

ohojvat, 2018-08-17:
Is it posible with JMRI? I can't find how
In NMRA 9.21 it exists
I am using version 3.10.1 and Java 1.8 (because 4.12 did not work on the first attempt) - I use Windows XP SP3.
------------------------------------------------------------------------------------------------
**This is from NMRA 9.21**
Accessory Decoder Configuration Variable Access Instruction
Accessory decoders can have their Configuration variables changed in the same method as locomotive decoders using the Configuration Variable Access Instruction - Long Form instruction defined above. For the purpose of this instruction, the accessory decoders’ address is expanded to two bytes in the following method. If operationsmode acknowledgement is enabled, the receipt of an Accessory Decoder  Configuration Variable Access instruction
must be acknowledged in the same manner as the Configuration Variable Access Instruction – Long Form.
 Basic Accessory Decoder Packet address for operations mode programming
10AAAAAA 0 1AAACDDD
Where DDD is used to indicate the output whose CVs are being modified and C=1.
 If CDDD= 0000 then the CVs refer to the entire decoder. The resulting packet would be
{preamble} 10AAAAAA 0 1AAACDDD 0 (1110CCVV 0 VVVVVVVV 0 DDDDDDDD) 0 EEEEEEEE 1
Accessory Decoder Address (Configuration Variable Access Instruction) Error Byte
--------------------------------------------------------------------- 
Any help would be appreciated
Osvaldo Hojvat

dheap, 2018-08-17:
It is not possible with JMRI 3.10. It is possible with current JMRI versions.

Despite information to the contrary, it is possible to install current JMRI versions with Windows XP. You need to install a special version of Java. Instructions at:
<http://jmri.org/install/WindowsNew.shtml#sysreq>

Programming accessories and signals on the main is then possible, using either these options in Single CV programmer or with a specially crafted decoder definition.
http://jmri.org/help/en/html/apps/DecoderPro/CreateDecoderAdvanced.shtml#accessoryops

Dave

> On 18 Aug 2018, at 4:07 AM, ohojvat <notifications@github.com> wrote:
> 
> Is it posible with JMRI? I can't find how
> In NMRA 9.21 it exists
> I am using version 3.10.1 and Java 1.8 (because 4.12 did not work on the first attempt) - I use Windows XP SP3.

ohojvat, 2018-08-18:
Thank you for your reply - I am improving a small DCC control system (LDH) and today I have  founded a method for sending the necesary ExpressNet command.  I didn' found it in Lenz 3.0 or 3.6 version.
It is what I was finding in JMRI. for on the main accessories programming (PoM)
I did the following: as PoM. ( If it may be helpfull for somebody:)    
With existing  PoM command  E6 30 the device  send the two following addres bytes, and the ommand station copy they to DCC without changes,  So I wrote these bytes as NMRA 9.21 requests and it works all OK.
Now if a I can install the recent JMRI version, I will look at XpressNet traffic which codes are using. 
Also LDH decoders definitions recently added.
Perhaps JMRI uses other solution.

Regards, Osvaldo

dheap, 2018-08-18:
I was the developer who worked on getting the (then incomplete) JMRI code for Accessory POM working in both Decoder Definitions and the Single CV Programmer.

That code uses NMRA 9.2.1 packet requests.

However the TrafficController code for at least one system (NCE USB) that cannot accept NMRA packet commands from JMRI does reverse engineer the NMRA packet requests to a command-station-specific POM command.

> @ohojvat So I wrote these bytes as NMRA 9.21 requests and it works all OK.

pabender, 2018-08-18:
Sent from my iPad

> On Aug 18, 2018, at 2:32 PM, ohojvat <notifications@github.com> wrote:
> Thank you for your reply - I am improving a small DCC control system (LDH) and today I have founded a method for sending the necesary ExpressNet command. I didn' found it in Lenz 3.0 or 3.6 version.
> 

This is correct, officially, Lenz does not support Operations mode programming of accessory decoders.
> It is what I was finding in JMRI. for on the main accessories programming (PoM)
> I did the following: as PoM. ( If it may be helpfull for somebody:)
> With existing PoM command E6 30 the device send [...]
I can’t guarantee this will work, at least not with all computer interfaces.  I’ll follow up shortly with more details as to why.

pabender, 2018-08-18:
> On Aug 18, 2018, at 6:35 PM, Dave Heap <notifications@github.com> wrote:
> I was the developer who worked on getting the (then incomplete) JMRI code for Accessory POM working in both Decoder Definitions and the Single CV Programmer.
> 
> That code uses NMRA 9.2.1 packet requests.
> 
> However the TrafficController code for at least one system (NCE USB) that cannot accept NMRA packet commands from JMRI does reverse engineer the NMRA packet requests to a command-station-specific POM command.
> 
Since @ohojvat is using an XPressNet system, I’ll expand here has to why this may or may not actually work.

First, as @ohojvat indicated, the XPressNet command for operations mode programming requests are basically the NMRA DCC packets for operations mode programming requests.  Once I discovered that, I implemented the command station code for XPressNet systems.

This code has always been a little flakey because it may not work correctly with packets that are not the same length as the operations mode programming packets ( which are 6 bytes if you include the address and the checksum ).  Shorter packets sometimes work and sometimes don’t.

Lenz actually made the situation worse with the version 3.6 software update for the LI-USB.  The version 3.6 LI-USB software actively filters anything it does not interpret as being an approved command.  This change does not impact how the command station interprets the commands, so if you have one of the Lenz serial interfaces, you have a better chance of this working than with the Lenz USB ( or Ethernet ) interfaces.

Given that @ohojvat indicated he was able to send the raw packets, this should work with his system.

dheap, 2018-08-18:
@pabender Agreed, things are likely to get a bit flakey when the command station interface doesn't officially support raw packets, but incomplete workarounds in the CommandStation code are possible.

We have a slightly similar situation with NceTrafficController:
- The Serial interface has commands to send 3-6 byte raw DCC packets to the track.
- The USB interface doesn't have any 'send raw DCC packet' commands, but we can intercept some packets (accessory throw, signal aspect & accessory POM) and convert them to NCE interface commands. But there is no NCE interface command for signal (extended accessory) POM.

ohojvat, 2018-08-19:
At the moment I have not Istalled current version of JMRI - I wil try again later with received suggestions
When using my improved programm I get this commands, as example for accessory Nº1 , CV2 , value 3
ExpressNet  E6  30  80  80  EC  01  03
and from command station  80  80  EC  01  03
For CV1023 , CV515, value3
E6  30  BF B0 EE  02  03
BF  B0  EE 02  03

I ask if some of you can do the same with JMRI 4.12 accessory on the main programming and get similar results.  Using ExpressNet.

----------------------------------------------------

Now I will add this programming mode to a braking module I am makeing.
I was improving it to permit using of all functions when is in brake state. Similar as Lenz do with asymetrical DCC but this is for any kind of DCC movil decoder. 
I guess I will permit to modify also the decoder address with PoM

I am using a rs-232 interfase and have no problem.  I expect my parts by compatible with other manufacturers ExpressNet products, but am happy if at the least my LDH products works as  like.

----------------------------------------------------------

NCE product information: 

I have tested only one NCE Power Cab (after fixing it - many uP zocket pins bad solders) All buttoms OK
At power-on: display  NCE  PROCAB V1.3

There is prog.  accessory on the main option (press 7 times prg/esc)
I tested about 20 times and only once I have a probabely OK  DCC command:
accessory Nº 2044 ( the max. admited)  , CV2 value3
DCC           BF  BF  EC  01  03
all the other times I get for the same example:      BF  0C  01  03
and for acc.Nº1  CV2  value3:        81  70  01  03

For the contrary, PoM for movile decoders works OK.
----------------------------------------------------------

Thank you for the help
Osvaldo

dheap, 2018-08-20:
NCE DCC systems send four repetitions each of both the Legacy format (S9.2.1 Appendix A) and Current (S9.2.1 Section D Basic Accessory Decoder Packet Format ) Ops Prog Accessory commands.

What you are being is perfectly normal. From my Track Packet Analyser:
- BF 8F EC 01 03 DE is A2044 CV0002=003 (current format, with error check byte)
- BF 0C 01 03 B1 is A2041*CV0002=003 (legacy format, with error check byte)

Note the legacy address format differs as it is based on the underlying board address (511), so 2041-2044 are all sent to 2041. This is an NCE firmware choice and is usually not a problem as modern accessory decoders should ignore the legacy format commands.

JMRI only sends the legacy format if [Address Type is set to "legacy"](http://jmri.org/help/en/html/apps/DecoderPro/CreateDecoderAdvanced.shtml#accessoryops) in a decoder definition , but some DCC systems (such as the NCE Power Cab) intervene and send the legacy format as well.

Hope this helps clarify things.

Dave

closed 2021-03-30 by ohojvat