JMRI/JMRI #5660: JMRI 4.12 can't read from SRC162

rm5248, 2018-08-18:
Version 4.10 of JMRI can read from Team Digital SRC162 when programming on the main:

[image]

However in 4.12, it's no longer possible to read the entire sheet when selecting 'programming on main':

[image]

devel-bobm, 2018-08-18:
This problem is likely a result of functional changes to LocoNet "Programming on the Main" at/around JMRI version 4.11.5.  See Pull Request #5152.

When those changes were made, the Team Digital SRC162 "decoder definition" probably needed a minor tweak that would allow "Programming on the Main" read operations to be provided on LocoNet.  That tweak was not made.

There may be other devices (Team Digital or otherwise) which can be read via LocoNet using "Programming on the Main" LocoNet messaging which also need this type of a fix.

The fix relates to the `LOCONETOPSBOARD` feature, and the fact that that XML attribute(?) is missing from the TD_SRC162.xml file.  I presume that it should look like the typical entry for RR-CirKits products like the MotorMan, which reads:

```
    <programming direct="no" paged="yes" register="no" ops="no">
        <mode>LOCONETOPSBOARD</mode>
    </programming>
```

except that the SRC162 supports `direct` mode programming where the RR-CirKits MotorMan does not.

Since I am away from hardware for the near future, I cannot test any attempt I might make to fix this issue.  So a fix will have to wait, unless another developer wishes to work this issue.  If so, please feel free to remove my self-assignment.

devel-bobm, 2018-08-18:
@rm5248 , as a workaround, change your connection configuration to reflect that "transponding present:" is set to "Yes".  This will allow the "Programming on the Main" LocoNet messages to be processed during CV reads.

You would probably want to restore the setting to "No" after configuring your SRC162 devices.

devel-bobm, 2018-08-18:
It looks like the XML files for the following Team Digital devices need updates for `LOCONETOPSBOARD`:

- TD_BlocD8.xml
- TD_CSC.xml
- TD_SC8.xml
- TD_SC82.xml
- TD_SIC24.xml
- TD_SIC24AD.xml
- TD_SIC24E.xml
- TD_SMD84.xml
- TD_SRC16.xml
- TD_SRC162.xml
- TD_SRC8.xml

Note that the following devices already are modified for `LOCONETOPSBOARD` functionality:
- RR-CirKits-LNCP-basic.xml
- RR-CirKits-MotorMan.xml
- RR-CirKits-SignalMan-basic.xml
- RR-CirKits-SwitchMan.xml
- RR-CirKits-TC-64-basic.xml
- RR-CirKits-TowerController-Mark-II.xml
- RR-CirKits-TowerMan-basic.xml
- RR-CirKits-WatchMan-basic.xml
- TamValleyDepot_QuadLN.xml
- TamValleyDepot_QuadLN_S_11.xml
- TamValleyDepot_QuadLN11.xml

rm5248, 2018-08-19:
I will try that when I get a chance, both the XML and transponding preset, but it may not be for a bit.  Using 4.10 as a workaround for now is not a problem.

rm5248, 2018-08-25:
I tried both ways(editing the XML file to add `<mode>LOCONETOPSBOARD</mode>` and setting "transponding present") and both ways did work for reading data back from the SRC162.

rm5248, 2018-09-11:
Have these changes been done yet?  Otherwise I can do a PR later this week.

devel-bobm, 2018-09-11:
Changes have not yet been made.  My experimentation shows that making the change will cause some undesirable effects in Ops-mode programming of the various Team Digital devices.  I have not yet been able to experiment on possible solutions to the undesirable effects, and I have not yet made contact with the manufacturer on the issues.

bobjacobsen, 2018-09-11:
> making the change will cause some undesirable effects in Ops-mode programming of the various Team Digital devices.

A bit more detail, please. Is this a JMRI bug?

devel-bobm, 2018-09-12:
>  making the change will cause some undesirable effects in Ops-mode programming of the various Team Digital devices.
A bit more detail, please. Is this a JMRI bug?

It may be a bug, or just a "feature".  _I haven't had the time to run the issue to ground_.  A JMRI Users post today about a possibly-related user issue may bump the priority a little higher, though.

My recollection is that, when programming over a LocoNet connection, the `LOCONETOPSBOARD` change will result in the "ops mode programmer" showing a "read all sheets" button and a "read full sheet" button.

This is ok for the typical user with LocoNet-connected Team Digital hardware.  

But it is awkward for any LocoNet user who has Team Digital products which are track-connected instead of LocoNet-connected, or any Team Digital products which are "downstream" of a Team Digital device which is configured as a "DCC track signal-to-Serial bus Gateway".  With such connections, reading the device is not possibe, but the JMRI programmer will display the "read" buttons.  The presence of those "read" buttons on the symbolic programmer become a support problem, where such a support problem did not previously exist.

In other words, we have no good way, within JMRI, to understand that the device being programmed is track-connected rather than LocoNet-connected, and that the read operation is therefore not possible.  

The one thing JMRI has going is that we do have the LocoNet "connection preference" for "transponding hardware present/absent".  I think this can act as a surrogate for actual knowledge, but this requires more research.

This issue does not affect track-connected devices when the command station is a non-LocoNet command station (because `LOCONETOPSBOARD` does not apply to non-LocoNet connections).

bobjacobsen, 2018-09-12:
This doesn't seem to be different from the situation before the option was added:  You can't read back things that aren't connected in a way that brings back the read reply messages.

devel-bobm, 2018-09-12:
> This doesn't seem to be different from the situation before the option was added: You can't read back things that aren't connected in a way that brings back the read reply messages.

On the surface, this is true.  But we also have a manufacturer (and, likely, more than one) who:

- produced his own decoder definition files;

- is unaware that JMRI functionality changed;

- is unaware of a new JMRI connection configuration option that influences how JMRI interacts with his hardware;

- is unaware that his files need modification because of the JMRI changes;

- is unaware that his users are having problems with his hardware because of these changes;

- is unaware that he needs to change his talking-points for answering his user's questions about how to use new versions of JMRI with his products.

From the perspective of an affected device manufacturer, I'd say that's some non-trivial difference.  (Not an insurmountable difference, though!)

bobjacobsen, 2018-09-12:
I must not have been clear. 

I agree that adding the new type to the file is a change. That should be done. 

My point was about your concern there are “undesirable effects” of adding that. I don’t see any, because once that’s done it restores the behavior that was there before. 

We should just do the update.

devel-bobm, 2018-09-13:
Pull-request #5790 should resolve this issue.

Installable versions of a JMRI "developer's" build should be available within an hour or two via the JMRI "Jenkins" web site at http://jmri.tagadab.com/jenkins/job/Development/job/Packages/ .  I believe that any build starting with build number 2726 should have these changes in place.

@rm5248 , if you can, please try a new JMRI install from the Jenkins server noted above, after build 2726 becomes available.  With these changes, it should be possible to Read and Write SRC162 CVs using the JMRI symbolic programmer without having to configure the LocoNet connection as if "Transponding" hardware is present.

rm5248, 2018-09-18:
I will take a look at it this weekend

rm5248, 2018-09-23:
Confirmed working with build 2763, JMRI.4.13.4ish+jenkins+20180923T1403Z+Re8c2c28.exe.

Thanks!

closed 2018-09-23 by rm5248