JMRI/JMRI #5671: JMRI 4.12 with Windows XP

ohojvat, 2018-08-20:
I had using version 3.10.1 with Java 8.101  and Windows XP SP3 
I try to install V4.4 - 4.8 - and recently 4.12 with Java8.151 as recommended with no error.
But when opening PanelPro , when arriving at main JMRI window, a bigg error page arises, with only opcion to close JMRI.   Main JMRI closes but PanelPro.exe  131MB and LaunchJMRI..exe 4.8M are still running
With DecoderPro is the same
If I return to the old version 3.10.1, it works.
Attached a print screen of error page.
[image]

Please help about this matter.
Osvaldo

ohojvat, 2018-08-20:
I tryed to open again and after a warning (JMRI is running) I can obtain to more windows.
Two secons later the error one.
[image]
[image]

dheap, 2018-08-20:
@ohojvat The first problem you are seeing occurs sometimes when JMRI is migrating from the legacy V3.x settings files to the new 4.x format.

The second problem you are seeing  is because you have some stuck JMRI processes from the first error. Never try launching any JMRI application when there are stuck processes. Kill them first.

To solve your problem:
- Kill any stuck JMRI processes.
- Try again. It may work the second time.
- If not, kill any stuck JMRI processes.
- Search your "C:\Documents and Settings\<yourusername>\JMRI" folder and its subfolders for any legacy "UserPrefsProfileConfig.xml" files and delete or rename them.
- Try again. You may have to recreate some preference items. 

Dave

closed 2021-03-30 by ohojvat