JMRI/JMRI #5679: Create web versions of Locations and YardMaster windows

mstevetodd, 2018-08-21:
Per https://groups.io/g/jmriusers/topic/24875955 it would be useful to have web versions of the Locations and YardMaster windows (from JMRI Operations). These could be similar to the existing Trains, Manifest and Conductor web apps.

mstevetodd, 2018-08-22:
Sample screenshots of yardmaster and conductor windows. Quite similar, except for train selector on Yardmaster, and Move button on Conductor.
[image]
[image]

mstevetodd, 2018-08-22:
matching screenshot of Conductor Web page.
[image]

mstevetodd, 2018-08-30:
I've worked on this a bit, and it appears most of the Conductor code can be used, if I explicitly pass the RouteLocation instead of assuming currentLocation. So basically YardMaster will be Conductor with additional navigation.

mstevetodd, 2018-09-05:
I've made some progress on this, visible on my demo site: http://jmri.mstevetodd.com:1080/

I'd welcome any input, particularly from folks who would use this feature.

KenC57, 2021-06-28:
It should be mentioned that the current Conductor screen only allows approving all moves at a location. If a pickup wasn't found, or a drop could not be performed. You should be able to mark as 'unknown' for location when the car isn't found for pickup. Likewise if you can't spot a car correctly, you should be able to put it somewhere as off-spot or carry to destination. Then the system would find another train to get it where it needs to go later.

KenC57, 2022-11-09:
I've been seeing some activities related to this sort of thing and need to investigate how far other parts have changed.

KenC57, 2023-06-27:
I've seen some discussion of the changes needed in the web protocols to support this. I'm trying to find that thread and connect it here. It sounded like it would be making the transport part of this work. Namely allowing the option to find or lose cars from a train remotely. That would leave the web page edits as the next step.

KenC57, 2023-06-29:
PR #12138 added to the json at least how to remove a car from a train and set a car as unknown location. That's have the battle and more likely the one that gets used the most. You get to a location and find the car listed for pickup isn't there.