JMRI/JMRI #5696: Reporter Table NPE

devel-bobm, 2018-08-24:
Seen in "head of master" code just now:

Upon reception of a LocoNet "Transponding" message, if the associated `Reporter` does not exist, one is created.  When the `Reporter Table` is asked to display that created bean, and that bean was the first Reporter bean, the ReporterDataModel issues an NPE.

If __that specific__ `Reporter` is then manually added, it appears in the `Reporter Table` as a reporter number one higher than the one specified in the dialog box, but the original bean (from the LocoNet message `Reporter` creation) also shows up at its correct `Reporter` number.

If instead JMRI is started and then a fake (unrelated) reporter is created manually using the `Reporter Table` "add" button, then the LocoNet message is received, the automatically created Reporter displays correctly, without NPE.

Steps to reproduce the problem:

- Start JMRI in LocoNet Simulator mode

- Open "LocoNet"->"Send LocoNet Packet" tool.

- In the uppermost "Packet" field, enter the following string
`D0 20 05 00 0B 01`

- Actuate the "send" button immediately below the uppermost "Packet" field.

- Open "Tools"->"Tables"->"Reporters".

The NPE will appear immediately, and the Reporter table will remain empty.

- Actuate the "Reporter Table" "Add" button.  Specify hardware address 6.  Actuate "Create" button.

The "Reporter Table" will now show "Reporter 6" (from the LocoNet message) and "Reporter 7" (from the manual add operation).

This issue results from a rather long string of user-level problems reported by "halaszgabor" on the JMRI user's list.  The specific experiment stems directly from the LocoNet Transponding message HEX values posted by @pabender (see [https://groups.io/g/jmriusers/message/151087](url)) during the lifetime of that thread.

pabender, 2018-08-24:
The actual log file that data line came from was provided by the user.  Those are here:
https://groups.io/g/jmriusers/files/Scripts

bobjacobsen, 2018-08-24:
Likely fix in #5697, would appreciate somebody checking it.

klk32003, 2018-08-24:
> Likely fix in #5697, would appreciate somebody checking it.

Tested under 
(1) JMRI 4.13.3ish, Windows 7, LocoNet Simulator
(2) JMRI 4.13.3ish, Windows XP, LocoBuffer-USB, connected to my layout

All good. 

I didn't notice the NPE, because I'm using a XML file like...
```
<?xml version="1.0" encoding="UTF-8"?>
<?xml-stylesheet href="/xml/XSLT/panelfile-2-9-6.xsl" type="text/xsl"?>
<layout-config xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://jmri.org/xml/schema/layout-2-9-6.xsd">
  <reporters class="jmri.jmrix.loconet.configurexml.LnReporterManagerXml">
    <reporter>
      <systemName>LR1</systemName>
    </reporter>
    <reporter>
      <systemName>LR2</systemName>
    </reporter>
    ...
  </reporters>
</layout-config>
```
...at JMRI Startup to fill the Reporter table.

Another note (not related to the NPE, just wanted to mention it): 
```
Send LocoNet Packet: D0 20 05 00 0B 01 --> [D0 20 05 00 0B 01]  Transponder address 11 (short) present at LR6 () (BDL16x Board ID 1 RX4 zone C).
Send LocoNet Packet: D0 20 05 7D 0B 7C --> [D0 20 05 7D 0B 7C]  Transponder address 11 (short) (or long address 16011) present at LR6 () (BDL16x Board ID 1 RX4 zone C).
```

Both seems to work. Digitrax [documents](http://www.digitrax.com/media/apps/products/detection-signaling/bdl168/documents/transponding%20bdl16xrx4%20app%20note_1.pdf) only the second variant.

devel-bobm, 2018-08-25:
Looks ok when JMRI configured for LocoNet Simulator.  I hope to test with real hardware in the next 24 hours.

Regards,
Bob

closed 2018-09-10 by bobjacobsen