JMRI/JMRI #5707: Track Current Meter Tool Nullpointer

icklesteve, 2018-08-26:
Hardware Type : MERG CBUS ( Network + Simulation )

On selecting the Track Current Meter Tool

```
java.lang.NullPointerException: Required nonnull default for jmri.MultiMeter does not exist.
	at jmri.InstanceManager.getDefault(InstanceManager.java:203)
	at jmri.jmrit.ampmeter.AmpMeterFrame.<init>(AmpMeterFrame.java:58)
	at jmri.jmrit.ampmeter.AmpMeterAction.actionPerformed(AmpMeterAction.java:29)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
```

CBUS ( and I'm sure a few others  ), don't currently report track current, perhaps a dialogue with a not currently supported message?

nb, no other hardware types tested,
Steve.

bobjacobsen, 2018-08-27:
Better might be to have the AmpMeterAction code check whether there's an instance of `MultiMeter` available, and deactivate the action (make it gray in the menu) to show it's not going to do anything.

pabender, 2018-09-14:
closed via #5797

closed 2018-09-14 by pabender

icklesteve, 2018-09-14:
Thanks both, deactivate option much better solution than a dialogue.