JMRI/JMRI #5714: XML validation error for rotated multisensoricon

dsand47, 2018-08-27:
The jmriusers group message "Down loading 4.12 Erroir" is actually an XML validation error.

When a multisensoricon has been rotated, the degrees attribute is added.  This attribute is not included in EditorMultiSensorIconType (editors.xsd).

bobjacobsen, 2018-08-28:
The current code doesn't write "degrees" any more, but it reads it if present. 

Not sure when was was removed.

I'll add it back to the schema so old files can be read.

bobjacobsen, 2018-08-28:
Can you append a file (or file fragment) to add as a test?  Thanks.  For old-release files like this, I prefer using real ones to synthetic test cases.

dsand47, 2018-08-28:
The degrees attribute is still being written in PositionableLabelXml#storeCommonAttributes:

        if (p.getDegrees() != 0) {
            element.setAttribute("degrees", "" + p.getDegrees());
        }

I was able to re-create the error with a brand new CPE panel with just the multi sensor icon.  Apparently nobody else has ever rotated a multi sensor icon.

bobjacobsen, 2018-08-28:
Well, that's unfortunate.  Looks like the inheritance tree got pretty messed up in that code.

I don't have time to fix it correctly at the moment.  I'll put up a PR to add it individually in the Schema (which will effectively bypass the problem) later this evening, and add it to the long queue of jmrit.display cleanups that are needed.

closed 2020-01-13 by bobjacobsen