JMRI/JMRI #5721: Multiple reporter managers can’t coexist?

pabender, 2018-08-28:
I am trying to figure out an issue tangentially related to #5696 in that the same user group thread has triggered looking into this.

The Roco Z21 creates three connections:
A Z21 Connection
An XPressNet connection
A LocoNet connection.
The actual device has physical LocoNet and XPressNet connections, along with a few others.  The LocoNet and XPressNet connections are tunneled through the Z21 protocol ( hence the multiple connections )

This generally all works, but there appears to be an issue with registering the Reporter Managers.  

We can create the one reporter possible with the Z21’s own reporter manager. 

What isn’t working is the LocoNet reporter manager.  I have verified the LocoNet Manager is actually created by the LocoNet system Connection memo and that setReporterManager is called for both Reporter Managers.  After initialization, it is only possible to create the single reporter for the Z21 ( it has one railcom detection zone...).  It isn’t possible to create a LocoNet reporter.

I’ll also note that the reporter table doesn’t have tabs like the turnout and sensor tables do.  In addition to Z21 and LocoNet, ?I expect to see an internal 

I think this all points to an issue with the instance manager or the proxy manager for reporters, but It all looks basically the same as the TurnoutManager and SensorManager code, which is working, and I can’t spot what is going on.

Does anyone have any insights or suggestions?

bobjacobsen, 2018-08-28:
What types are being registered?

Bob

pabender, 2018-08-28:
The ones created by the Z21 connection are a jmri.jmrix.roco.z21.Z21ReporterManager, and a jmri.jmrix.lovonet.LnReporterManager.  These are created and registered by a  Z21SystemConnectionMemo and a LocoNetSystemConnectionMemo, respectively.

These are both registered with the instance manager using the setReporterManager method ( which follows the same pattern as the SetTurnoutManager and setSensorMsnager methods.

There should also be an internal reporter manager created.

bobjacobsen, 2018-08-29:
There is a ProxyManager being created for Reporters, which is they key to having multiple reporters with _different_ system prefixes working.  Are the multiple managers being created with different prefixes?

> I’ll also note that the reporter table doesn’t have tabs like the turnout and sensor tables do. In addition to Z21 and LocoNet, ?I expect to see an internal

There’s no jmri.jmrit.beantable.ReporterTableTabAction action class.  The FooTableTabAction class handles multiple Foo implementations.  FooTableAction (i.e. ReporterTableAction) just provides a single combined table.

pabender, 2018-08-29:
@bobjacobsen yes, there are two different prefixes involved.

The default Z21 prefix is "Z" and the LocoNet prefix for that connection beomese ZL. In other words, ZR1 is a Z21 reporter, and ZLR1 should be a LocoNet reporter.

Could this be as simple as the proxy managers not following the principle of longest substring?  In other words, rather than matching the first matching prefix, we should be matching the longest prefix.  If that's the case, this is a pretty subtle bug in the implementation.

I'm more concerned with getting the managers recognized than the tables.  I had found the missing TableTabAction after posting this.

bobjacobsen, 2018-08-29:
ZL is one of those “Unparsable system prefixes” that we’ve been trying to get rid of.  See #4670 They don’t work in _lots_ of places.

System prefixes need to be a single letter optionally followed by digits.

> @bobjacobsen yes, there are two different prefixes involved.
>
> The default Z21 prefix is "Z" and the LocoNet prefix for that connection beomese ZL. In other words, ZR1 is a Z21 reporter, and ZLR1 [...]

pabender, 2018-08-29:
Ok, I was wrong.  The loconet tunnel has been registering itself as Just L.

https://github.com/JMRI/JMRI/blob/f502632a1650c077fc023d0c3858e213254c76d2/java/src/jmri/jmrix/roco/z21/Z21LocoNetTunnel.java#L229

(The XPressnet tunnel used to register itself as ZX...)

Running this script:

import jmri
jmri.InstanceManager.reporterManagerInstance().provide("LR1")

Produces this output:

     [java] java.lang.IllegalArgumentException: java.lang.IllegalArgumentException: Invalid Reporter name: ZRLR1 - only one reporter supported 

so something isn't happening correctly here.

bobjacobsen, 2018-08-29:
That message comes from jmri/jmrix/roco/z21/Z21ReporterManager  createNewReporter(String systemName, String userName) when it finds an internal mismatch between the system name it’s being asked to create (here ZRLR1) and what it expects.  So the problem is whoever is calling that.

I don’t see a call to that from jmrix/roco/ or jmrix/lenz or jmri/loconet, so it’s probably from the java/src/jmri//managers/AbstractReporterManager.java

> import jmri
> jmri.InstanceManager.reporterManagerInstance().provide("LR1")
> 
> Produces this output:
> 
>  [java] java.lang.IllegalArgumentException: java.lang.IllegalArgumentException: Invalid Reporter name: ZRLR1 - only one reporter supported 
> 
> so something isn't happening correctly here.

pabender, 2018-08-29:
@bobjacobsen I know where that is generated.

The problem is that asking for LR1 SHOULD be asking the ProxyReporterManager for LR1, which should send it off to the LocoNet Reporter manager, and that isn't happening, but I don't see why that isn't happening.

bobjacobsen, 2018-08-29:
Maybe not the problem, but this code doesn’t work (from java/src/jmri//jmrix/roco/z21/Z21ReporterManager.java)

    public Z21ReporterManager(Z21SystemConnectionMemo memo){
        _memo = memo;
        if(InstanceManager.getDefault(RailComManager.class)==null){
              // there is no RailComManager, so create a new one
              InstanceManager.setDefault(RailComManager.class,
                                     new jmri.managers.DefaultRailComManager());
        }

That getDefault throws an NPE instead of returning null.

pabender, 2018-08-29:
@bobjacobsen I have fixed that locally, but that getDefault doesn't ever throw an NPE or return null.  A DefaultRailComManager object is instantiated through the DefaultInstanceInitializer.
 it isn't the Z21ReporterManager that's the problem.

Fixing that locally (both with a try catch and calling getNullableDefault instead of getDefault ) doesn't change what happens here.

bobjacobsen, 2018-08-29:
Where’s the start of automatically creating a Z21Reporter?  I.e. where does it say “This message says I need a Reporter with system name …”?

I can see in Z21Reporter where it creates tags, but not where the Reporter itself is created.

pabender, 2018-08-29:
@bobjacobsen,

The reporter isn't currently created automatically.  It has to be done manually.

The thought was that not everyone wants to use it, so while we create the manager automatically, we don't create the reporter just in case someone wants to ignore that traffic.

We could actually create it when the Z21ReporterManager is created.  Every z21 and Z21 system has, the hardware built in.

pabender, 2018-08-30:
So, apparently, I've been looking at code that wasn't in sync with master.
The Z21SystemConnectionMemo in master was actually registering the Z21ReporterManager using store instead of setReporterManager.

I've fixed that and added a tabbed pane for the Reporters as part of #5722.  As soon as I fix a couple of tests, I'll get that merged in.

pabender, 2018-08-30:
closing via #5722

closed 2018-08-30 by pabender