JMRI/JMRI #5727: No Turnout Table!

esfeld, 2018-08-31:
Opening Tools\Tables\Turnouts nothing happens ....... opening Window\Turnouts I get a blank Turnouts table with no "Add" etc  ...... I DLed JMRI ver 4.12 and still have the problem. I can operate turnouts from Engine Driver so they must exist in JMRI but no TO table in JMRI on Win 10 computer.  In addition while the blank TO table is open "Quit" in Panel Pro is frozen and can only close using Task Manager.  Any ideas?

bobjacobsen, 2018-08-31:
(This is a duplicate of the comment on #4599 just a moment ago)  Are you loading a panel file when the program starts up?  Sometimes problems in that file can cause similar symptoms.  

Also, are there any ERROR or WARN messages in the Console log?  (It's in the Help menu)

KenC57, 2018-08-31:
The next step is to see if anything shows in the Help->System Console, have
this open before trying to open the tables. If something shows up then, that
would be an error or warning related to why the table can't display.

esfeld, 2018-08-31:
Thank you for your response. No, no panel being opened ........ In the interim (approx 1 hour) my turnout table has opened.  As an added complication, I'm sure, I am using dropbox on a JMRI simple network connection.  While the TO table opened on the layout laptop it still will not open on one of the other networked boxes. Complicated I know but up until about a month ago everything worked just fine.   Go figure.  Any ideas as to why it would delay so drastically before populating?

KenC57, 2018-08-31:
When you have the case of the table opening, a long time later, check the
Help->System Console, if the opening happens with a line in the log about:

2018-08-31 11:29:46,386 audio.JoalAudioFactory                INFO  -
Initialised JOAL using OpenAL: vendor - OpenAL Community version - 1.1
ALSOFT 1.15.1 [Listed Table Generation]

That says your system has changed and it is failing due to a library issue.
Windows updates causes this to happen on some systems. The latest code has
options set that should avoid the issue. But if an install isn't working
right and some parts from an old install (startup links, and similar), it
can get this problem. A clean install of JMRI usually clears the issue.

KenC57, 2018-08-31:
Were there additional lines with the 'unexpected state' message, usually
some form of a trace back? This shows us exactly where in the code the
problem happened.

bobjacobsen, 2018-08-31:
> On the layout laptop I get a series of Java string index out of range errors.
> On the client console is showing :2018-08-31 11:17:22,078 jmrix.AbstractMRTrafficController ERROR - left timeout in unexpected state: 50 [jmriclient.JMRIClientTrafficController Transmit thread]
> 2018-08-31 11:17:22,192 jmrix.AbstractMRTrafficController ERROR - left timeout in unexpected state: 50 [jmriclient.JMRIClientTrafficController Transmit thread]

Which machine are you trying to open the table on?

More information on the Java String errors would be helpful. Is there a traceback that shows where it happened?

Bob

bobjacobsen, 2018-08-31:
Ken’s suggestion about the audio library is a good one. That’s used to be the common cause of that on Windows. But Randall thought he’d fixed that, so it’s not entirely clear that this is the same thing. 

> On Aug 31, 2018, at 8:10 AM, esfeld <notifications@github.com> wrote:
> 
> Any ideas as to why it would delay so drastically before populating?

esfeld, 2018-08-31:
I think that a complete uninstall\install is probably my best bet ... albeit a lot of reconfiguration.  BTW  I am on Long Island ... where in Central NY are you located?  Just curious ... as I  just saw eldest G'son off to Syracuse U.

bobjacobsen, 2018-08-31:
As a workaround, you can try setting presences on that machine to open the turnout table at startup. That can bypass the audio issue, because it opens -just- the turnout table, not the tabbed window with all the tables. 

Bob

bobjacobsen, 2018-08-31:
Uninstall/ reinstall of JMRI shouldn’t require redoing the JMRI configuration 

Bob

esfeld, 2018-08-31:
Bob, thank you ...... thought I was responding to Ken ..... thought that if I did complete uninstall via IObit Uninstall all config files would be lost as well

KenC57, 2018-08-31:
There are two parts, one is the program space that is usually C:\Program
Files (x86)\JMRI, that is what the installer deals with. All you custom
setup, panels, roster, etc... that is your user space, usually
C:\Users\your_login_name\JMRI, that's where your stuff lives and stays even
if the program is uninstalled.

dheap, 2018-09-01:
Sometimes a Recreate Roster Index is all that is needed.

If your roster folder is on Dropbox, there are 30 days of history of deleted and changed files to recover your work. I use that technique sometimes. But again, a rebuild of the Roster Index is necessary/advisable.

> On 1 Sep 2018, at 3:23 AM, STEPHEN FELDMAN <notifications@github.com> wrote:
> 
> had to use a backup of Roster and will need to update

closed 2021-04-14 by stale[bot]