JMRI/JMRI #5728: CAN Send Frame in MERG CBUS does not add priority to CBUS messages.

icklesteve, 2018-08-31:
Priority needs to be added to the CAN message to meet CBUS spec.
Am thinking add a checkbox to enable this behaviour to all outgoing messages so can be used for both CAN and CBUS,

Steve.

bobjacobsen, 2018-08-31:
A bit out of date on the CBUS spec and now can't find it, sorry.  What does "priority" mean in this context?

icklesteve, 2018-08-31:
Hi Bob,

Latest dev manual at
https://www.merg.org.uk/merg_wiki/lib/exe/fetch.php?media=public:cbuspublic:developer_6b.pdf

+ Pg. 9

> The following nomenclature for the CAN header will be used MjPri MinPri ID 

 + Pg. 30

> MjPri bits 9 – 10 of the CAN header. Dynamic Priority, elevated by the node to gain access
> based on a transmit fail count. Values:
> 0 – Emergency priority
> 1 – High priority
> 2 – Normal priority
> MinPri bits 7 - 8 of the CAN header. Static priority based on message and node type. Values:
> 0 – High access
> 1 – Above Normal access
> 2 – Normal access
> 3 – Low access

There's a checkbox in the CBUS Console log to view message priority in CBUS messages, has been there as long as I've known JMRI.

The priorities are added in other JMRI CBUS code by CbusMessage.setPri ( or getPri in reverse ) http://jmri.org/JavaDoc/doc/jmri/jmrix/can/cbus/CbusMessage.html

Approx ln. 160 of 
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/can/cbus/CbusMessage.java

Messages sent from the CAN Send CAN Frame Tool are only JMRI-sent messages I've noticed without a CBUS priority, which is fair enough as it's in the CAN directory, not the CBUS!

I thought a checkbox to add the priority could keep the tool within CAN but also enable CBUS compliance without duplicating whole pane?

Not a major issue, only developers likely to notice the difference, however likely to help developers who may be using the send frame tool to observe the output in order to create their own CBUS frames,

Steve

icklesteve, 2018-09-06:
resolved in #5755

closed 2018-09-06 by icklesteve