JMRI/JMRI #5729: Light Control PanelPro Tool can cause Stack O, MERG CBUS

icklesteve, 2018-08-31:
Hi Folks,

No tables / panels loaded, just PanelPro from Head.
Likely to be long-term unreported issue.

MERG CAN Simulator - Stack Overflow
Enter incorrect hardware address ( eg. "123" without quotes ) as light address.
Click On, sends cbus message to event 123, though generally as many times as possible, eg 1277 on events in less than a second.
For each CAN frame sent, this is added to JMRI console

```
	at jmri.implementation.AbstractLight.setState(AbstractLight.java:428)
	at jmri.jmrix.can.cbus.CbusLight.message(CbusLight.java:91)
	at jmri.jmrix.can.adapters.loopback.LoopbackTrafficController.forwardMessage(LoopbackTrafficController.java:30)
	at jmri.jmrix.AbstractMRTrafficController.notifyMessage(AbstractMRTrafficController.java:176)
	at jmri.jmrix.can.adapters.loopback.LoopbackTrafficController.sendCanMessage(LoopbackTrafficController.java:51)
	at jmri.jmrix.can.cbus.CbusLight.doNewState(CbusLight.java:79)
```

MERG Actual connection - System Freezes.
Enter incorrect address, sends CAN message in loop from incorrect address, unable to re-gain control of JMRI, have to force shutdown.
Essentially same as above only as happening a lot slower!

Need to do more investigation to see if hardware specific / general.
Have got a feeling system specific as no issues when address entered correctly, eg "+123".

Full trace 

```
2018-08-31 19:23:24,760 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.StackOverflowError
	at com.sun.java.swing.SwingUtilities3.getDelegateRepaintManager(SwingUtilities3.java:123)
	at javax.swing.RepaintManager.getDelegate(RepaintManager.java:1736)
	at javax.swing.RepaintManager.addInvalidComponent(RepaintManager.java:346)
	at javax.swing.JComponent.revalidate(JComponent.java:4849)
	at javax.swing.plaf.basic.BasicTextUI$RootView.preferenceChanged(BasicTextUI.java:1406)
	at javax.swing.text.View.preferenceChanged(View.java:288)
	at javax.swing.text.PlainView.updateDamage(PlainView.java:566)
	at javax.swing.text.PlainView.insertUpdate(PlainView.java:451)
	at javax.swing.text.FieldView.insertUpdate(FieldView.java:293)
	at javax.swing.plaf.basic.BasicTextUI$RootView.insertUpdate(BasicTextUI.java:1610)
	at javax.swing.plaf.basic.BasicTextUI$UpdateHandler.insertUpdate(BasicTextUI.java:1869)
	at javax.swing.text.AbstractDocument.fireInsertUpdate(AbstractDocument.java:201)
	at javax.swing.text.AbstractDocument.handleInsertString(AbstractDocument.java:748)
	at javax.swing.text.AbstractDocument.insertString(AbstractDocument.java:707)
	at javax.swing.text.PlainDocument.insertString(PlainDocument.java:130)
	at javax.swing.text.AbstractDocument.replace(AbstractDocument.java:669)
	at javax.swing.text.JTextComponent.setText(JTextComponent.java:1669)
	at jmri.jmrix.can.cbus.swing.console.CbusConsolePane.message(CbusConsolePane.java:1172)
	at jmri.jmrix.can.adapters.loopback.LoopbackTrafficController.forwardMessage(LoopbackTrafficController.java:30)
	at jmri.jmrix.AbstractMRTrafficController.notifyMessage(AbstractMRTrafficController.java:176)
	at jmri.jmrix.can.adapters.loopback.LoopbackTrafficController.sendCanMessage(LoopbackTrafficController.java:51)
	at jmri.jmrix.can.cbus.CbusLight.doNewState(CbusLight.java:79)
	at jmri.implementation.AbstractLight.setState(AbstractLight.java:428)
	at jmri.jmrix.can.cbus.CbusLight.message(CbusLight.java:91)
	at jmri.jmrix.can.adapters.loopback.LoopbackTrafficController.forwardMessage(LoopbackTrafficController.java:30)
	at jmri.jmrix.AbstractMRTrafficController.notifyMessage(AbstractMRTrafficController.java:176)
	at jmri.jmrix.can.adapters.loopback.LoopbackTrafficController.sendCanMessage(LoopbackTrafficController.java:51)
	at jmri.jmrix.can.cbus.CbusLight.doNewState(CbusLight.java:79)
	at jmri.implementation.AbstractLight.setState(AbstractLight.java:428)
	at jmri.jmrix.can.cbus.CbusLight.message(CbusLight.java:91)
	at jmri.jmrix.can.adapters.loopback.LoopbackTrafficController.forwardMessage(LoopbackTrafficController.java:30)
	at jmri.jmrix.AbstractMRTrafficController.notifyMessage(AbstractMRTrafficController.java:176)
	at jmri.jmrix.can.adapters.loopback.LoopbackTrafficController.sendCanMessage(LoopbackTrafficController.java:51)
	at jmri.jmrix.can.cbus.CbusLight.doNewState(CbusLight.java:79)
	at jmri.implementation.AbstractLight.setState(AbstractLight.java:428)
	at jmri.jmrix.can.cbus.CbusLight.message(CbusLight.java:91)
	at jmri.jmrix.can.adapters.loopback.LoopbackTrafficController.forwardMessage(LoopbackTrafficController.java:30)
	at jmri.jmrix.AbstractMRTrafficController.notifyMessage(AbstractMRTrafficController.java:176)
	at jmri.jmrix.can.adapters.loopback.LoopbackTrafficController.sendCanMessage(LoopbackTrafficController.java:51)
	at jmri.jmrix.can.cbus.CbusLight.doNewState(CbusLight.java:79)
	at jmri.implementation.AbstractLight.setState(AbstractLight.java:428)
	at jmri.jmrix.can.cbus.CbusLight.message(CbusLight.java:91)

```

Red Herring alert - 
The behaviour seems to change slightly depending if the get status button is clicked first ( status requests haven't yet been implemented in JMRI for CBUS ).

Steve.

bobjacobsen, 2018-09-11:
The problem is here:
```
	at jmri.jmrix.can.cbus.CbusLight.doNewState(CbusLight.java:79)
	at jmri.implementation.AbstractLight.setState(AbstractLight.java:428)
	at jmri.jmrix.can.cbus.CbusLight.message(CbusLight.java:91)
```

The `doNewState` method in CbusLight _always_ sends a message, even if its being asked to go to the same state it already has i.e. `oldState.equals(newState)`.  That then immediately sends the message, which on a simulator (but not a real layout) immediately comes back to the `message()` method which calls `setState`, which calls `doNewState`.   That loop needs to be broken by one of those pieces of code stopping and saying "yes, yes, I know we're ON, thank you very much" and not going any further.

icklesteve, 2018-09-11:
Thanks Bob,
Will take a proper look into this in the next week or so,
Steve

icklesteve, 2018-09-26:
Hi Bob,

Have tried out suggestion of comparing oldstate and newstate.
The current state would make sense if the output was 100% known to be a physical output.
It ocurred to me though, that as CBUS is essentially a message protocol, people are using on events as starters for actions.
Eg, an action on a module is triggered by sending an on event according to a logix, no off event is ever sent.

What I did notice however, was that if a 100% valid event was sent either before or after adding the check there was no change in behaviour, however after adding the light status difference comparison there were still 4-6 attempts at sending an on message, not a single off message was sent.

By 100% valid, I mean "+1" , "+123" , "+n456e789".

By NOT 100% valid I mean "1" , "123" . "456e789", all without leading "+" or "-"  
( I'd class these as 90% as have found them buggy, eg in this case! )

0% valid eg "Badger" or "+WWWW" return the correct response, ie error message within simple light control.

What I think might be happening is that as the on ( or off ) message to switch 123 off ( or on ) is 123 etc, loop grows.

It's been validated successfully by the light manager ( by the cbus lm adding the leading + or -
), however the validation is a failing on the auto-generated 2nd half of a not 100% valid message.

Any thoughts on what to look at next before I suggest a Plan B?

Thanks,
Steve

bobjacobsen, 2018-09-27:
> On Sep 26, 2018, at 6:31 PM, steve young <notifications@github.com> wrote:
> 
> Have tried out suggestion of comparing oldstate and newstate.
> The current state would make sense if the output was 100% known to be a physical output.
> It ocurred to me though, that as CBUS is essentially a message protocol, people are using on events as starters for actions.
> Eg, an action on a module is triggered by sending an on event according to a logix, no off event is ever sent.

Not sure what you’re saying here, sorry.

The bean&listener methodology in JMRI (and in Java in general) relies on old!=new to avoid creating loops in event propagation.  There may be other problems, but if something is firing events with old==new it will eventually drive loops that cause stack overflows.

> What I did notice however, was that if a 100% valid event was sent either before or after adding the check there was no change in behaviour, however after adding the light status difference [...]

I’d look at the code that interprets the name string to see whether its handling the “123” case inconsistently.  Perhaps in some places it interprets that as “+123 on, -123 off”, but in others it thinks it means “123 on, no off”.

Bob

--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

icklesteve, 2018-10-11:
I'm struggling to re-create this when running JMRI via ant ( consistently reproducible when installed from package. )
Would this represent a potential threading issue?