JMRI/JMRI #5754: Updated Hornby TTS decoder file

brianjackson1, 2018-09-05:
Hello 

I recently bought a few new TTS decoders, but found that the decoder file in Decoder pro had not been updated since May 2017 so I thought I would add them myself.

With help from RMWeb members I have been able to add an additional 5 locomotives:
Class 08, Class 20, Class 31, Class 43 HST and Class 60.

I have also updated the following:
Class 37 separated from Class 40, Class 66 renamed to Class 67 as No class 66 chip issued so far.

Added factory reset option of only resetting sound volume levels using a value of 5 on CV8.

The decoders have been tested and all have been reported as being correctly identified in decoder pro..
All the Sound volume level CV entries have been updated so that the sound volumes can be updated under the sound levels tab.

Please see the the following RMWeb thread for details:

http://www.rmweb.co.uk/community/index.php?/topic/136936-tts-decoders-help-required-to-update-tts-decoder-profile-in-jmri/

I am looking to add additional steam locomotives to the file but with a lack of TTS manuals available on the Hornby site I am finding it difficult to source the manuals.  

I am not the original author of the file, that was Nigel Cliffe, but I did want to update the file so my TTS decoders were properly recognised and decided to add as many as I could provided the information was available.   Nigel suggested I post it here once the file had been tested.
Regards
Brian

[Hornby_TTS_Decoders.zip](https://github.com/JMRI/JMRI/files/2354516/Hornby_TTS_Decoders.zip)

bobjacobsen, 2018-09-06:
This duplicates some model names that are in the existing xml/decoders/HornbyTTS.xml file.  Should this new one just replace that one (i.e. by changing the name of the new one to xml/decoders/HornbyTTS.xml )?

brianjackson1, 2018-09-06:
I used a copy of the original HornbyTTS.xml file to carry out the update,  so yes it could be renamed and replace the original file.

bobjacobsen, 2018-09-10:
Closed by #5773, which merged this to JMRIR 4.13.3.  Thanks for the update!

closed 2018-09-10 by bobjacobsen