JMRI/JMRI #5764: Handling of LocoNet Reporter numbering is off by 1 for BXP88

devel-bobm, 2018-09-08:
The BXP88 (and presumably BXPA1) provides a slightly-different transponding location report LocoNet message than the BDL16x devices.  

This means that JMRI reports a transponding zone number which is "off-by-1" versus a Digitrax throttle for the BXP88, but reports a transponding zone number which is identical to a Digitrax throttle for the BDL16x.

This was reported on the "LocoNet_Hackers" list as part of the user issues reported in post number 10079.

devel-bobm, 2018-09-08:
Note that fixing this bug will result in discrepancies for any BXP88 (and perhaps BXPA1) user who created "Reporters" using previous versions of JMRI,  

Fixing this bug will not affect users of BDL16x-based transponding.

bobjacobsen, 2018-09-08:
On the number change: is there enough difference in the message formats from these devices that JMRI can tell the origin?  If so, we could add code to either warn the user or do a one-time conversion.

devel-bobm, 2018-09-08:
@bobjacobsen, My initial investigation, based solely on BXP88 transponding "location report" messages reported by users, is that the LSbit of byte 2 of BXP88 OPC_MSEN messages is a '1' for BXP88 and a '0' for BDL16x+RX4.  I have not yet confirmed this suspicion.  And I have not yet confirmed that the BXPA1 behaves like the BXP88.  It is my intention to confirm this with physical hardware within the next few days.

The "LnReporter" code and the "LocoNet Monitor" code are both grabbing all 7 LSbits of byte 2 to use as the LS bits of the "Reporter" number and the "Zone" number, respectively.   If we simply mask the offending bit, the problem may go away and solve the functional problem.

An automatic conversion of "wrong" Reporters may be possible, by identifying LRx values where x is odd, and simply subtracting one.

Pugwash1, 2018-09-08:
@devel-bobm @bobjacobsen  I have just done some testing with BDL168+RX4, and on my dt402(firmware .21) it reports zone 22 and JMRI reports LR23, zone 20 is LR21 so both BXP/BDL168+RX4 report the same way, a difference of 1 between the dt402 and jmri. I think digitrax allows 2048 sensors so using binary they use 0-2047. By convention JMRI uses 1-2048, so it uses 1-2048 on the reporters, but digitrax uses 0-2047.  I checked and the first zone on the throttle is Zone Zero. If you change it, the sensors wont align with the zones.

devel-bobm, 2018-09-09:
A bit of experimentation tonite shows that **assumptions previously made are wrong**.

Experimentation reveals that byte 2 of the OPC_MSEN (0xD0) opcode is used differently between the BDL16x and the BXP88.  The BDL168 encodes its eight "zones" in bits `3-1` of byte 2, while the BXP88 encodes its eight "zones" in bits `2-0` of byte 2.

This means that it is NOT possible to discriminate between the two sources of messages based on the message content across all zones - four BXP88 zones do not overlap at all, but four BDL16x zones overlap with zones related to a BXP88 device which is programmed for a different Board ID.

I believe that we _do not_ need to discriminate between the two.  We probably just need to live with the odd nature of the numbering!  (And provide a suitable warning to users.)  

I believe that BDL16x `Reporter` numbers are likely to remain as-is, incrementing by 2s, while BXP88 `Reporter` numbers will continue to increment by 1s.  

`LocoNet Monitor` will need some changes, however.  There part of the message which atttempts to define the "source" of the report will likely get longer.

I must do some more experimentation, and check the BXPA1, before I make any promises, though.

devel-bobm, 2018-09-09:
I can confirm that the BXP88 uses the lower 3 bits of byte 2 of the OPC_MSEN message to determine which "section" the transponder is occupying.

I also know that the BXP88 uses bit 4 of byte 2 of the OPC_MSEN as part of the "board ID" value.

This contradicts how the BDL16x device handles its "zones" and its "board id" value - it uses bits 3-1 of byte 2 for "transponding zone" and bits 6-4 of byte 2 as part of the "board ID" value.  For the BDL16, bit 0 is apparantly always zero and does not have anything to do with "zone" or "board ID".

The above implies that the current `LnReporter` implementation need not change in any way.  It also implies that `Llnmon` needs some tweaks.  

I have implemented some LocoNet Monitor changes in `Llnmon` in my local repository.  I have verified with BDL16x and BXP88 hardware, and I am in the process of building testcases for these changes.  The messages I am implementing are shown below.

Note that I have thus far been unable to get my BXPA1 to issue any transponding messages, so I haven't been able to check its mapping from Board ID value to JMRI "Repeater" number.  I am also experiencing difficulty with its block detection function.

Here are the `LocoNet Monitor` messages which will be emitted when a BDL168 board with Board ID=2 sends transponding messages when a transponding decoder with short address 3 is connected sequentially to transponding zone detectors A thru H:

```
[D0 20 10 7D 03 61]  Transponder address 3 (short) (or   long address 16003) present at LR17 () (BDL16x Board ID 2 RX4 zone A or BXP88   Board ID 3 section 1).
[D0 00 10 7D 03 41]  Transponder address 3 (short) (or long   address 16003) absent at LR17 () (BDL16x Board ID 2 RX4 zone A or BXP88 Board   ID 3 section 1).
[D0 20 12 7D 03 63]  Transponder address 3 (short) (or long   address 16003) present at LR19 () (BDL16x Board ID 2 RX4 zone B or BXP88   Board ID 3 section 3).
[D0 00 12 7D 03 43]  Transponder address 3 (short) (or long   address 16003) absent at LR19 () (BDL16x Board ID 2 RX4 zone B or BXP88 Board   ID 3 section 3).
[D0 20 14 7D 03 65]  Transponder address 3 (short) (or long   address 16003) present at LR21 () (BDL16x Board ID 2 RX4 zone C or BXP88   Board ID 3 section 5).
[D0 00 14 7D 03 45]  Transponder address 3 (short) (or long   address 16003) absent at LR21 () (BDL16x Board ID 2 RX4 zone C or BXP88 Board   ID 3 section 5).
[D0 20 16 7D 03 67]  Transponder address 3 (short) (or long   address 16003) present at LR23 () (BDL16x Board ID 2 RX4 zone D or BXP88   Board ID 3 section 7).
[D0 00 16 7D 03 47]  Transponder address 3 (short) (or long   address 16003) absent at LR23 () (BDL16x Board ID 2 RX4 zone D or BXP88 Board   ID 3 section 7).
[D0 20 18 7D 03 69]  Transponder address 3 (short) (or long   address 16003) present at LR25 () (BDL16x Board ID 2 RX4 zone E or BXP88   Board ID 4 section 1).
[D0 00 18 7D 03 49]  Transponder address 3 (short) (or long   address 16003) absent at LR25 () (BDL16x Board ID 2 RX4 zone E or BXP88 Board   ID 4 section 1).
[D0 20 1A 7D 03 6B]  Transponder address 3 (short) (or long   address 16003) present at LR27 () (BDL16x Board ID 2 RX4 zone F or BXP88   Board ID 4 section 3).
[D0 00 1A 7D 03 4B]  Transponder address 3 (short) (or long   address 16003) absent at LR27 () (BDL16x Board ID 2 RX4 zone F or BXP88 Board   ID 4 section 3).
[D0 20 1C 7D 03 6D]  Transponder address 3 (short) (or long   address 16003) present at LR29 () (BDL16x Board ID 2 RX4 zone G or BXP88   Board ID 4 section 5).
[D0 00 1C 7D 03 4D]  Transponder address 3 (short) (or long   address 16003) absent at LR29 () (BDL16x Board ID 2 RX4 zone G or BXP88 Board   ID 4 section 5).
[D0 20 1E 7D 03 6F]  Transponder address 3 (short) (or long   address 16003) present at LR31 () (BDL16x Board ID 2 RX4 zone H or BXP88   Board ID 4 section 7).
[D0 00 1E 7D 03 4F]  Transponder address 3 (short) (or long   address 16003) absent at LR31 () (BDL16x Board ID 2 RX4 zone H or BXP88 Board   ID 4 section 7).
```

Here are the `LocoNet Monitor` messages which will be emitted when a BXP88 board with Board ID=2 sends transponding messages when a transponding decoder with short address 3 is connected sequentially to detection/transponding sections 1 thru 8:

```
[D0 20 08 00 03 04]  Transponder address 3 (short) present at   LR9 () (BDL16x Board ID 1 RX4 zone E or BXP88 Board ID 2 section 1).
[D0 00 08 00 03 24]  Transponder address 3 (short) absent at LR9   () (BDL16x Board ID 1 RX4 zone E or BXP88 Board ID 2 section 1).
[D0 20 09 00 03 05]  Transponder address 3 (short) present at   LR10 () (BXP88 Board ID 2 section 2).
[D0 00 09 00 03 25]  Transponder address 3 (short) absent at   LR10 () (BXP88 Board ID 2 section 2).
[D0 20 0A 00 03 06]  Transponder address 3 (short) present at   LR11 () (BDL16x Board ID 1 RX4 zone F or BXP88 Board ID 2 section 3).
[D0 00 0A 00 03 26]  Transponder address 3 (short) absent at   LR11 () (BDL16x Board ID 1 RX4 zone F or BXP88 Board ID 2 section 3).
[D0 20 0B 00 03 07]  Transponder address 3 (short) present at   LR12 () (BXP88 Board ID 2 section 4).
[D0 00 0B 00 03 27]  Transponder address 3 (short) absent at   LR12 () (BXP88 Board ID 2 section 4).
[D0 20 0C 00 03 00]  Transponder address 3 (short) present at   LR13 () (BDL16x Board ID 1 RX4 zone G or BXP88 Board ID 2 section 5).
[D0 00 0C 00 03 20]  Transponder address 3 (short) absent at   LR13 () (BDL16x Board ID 1 RX4 zone G or BXP88 Board ID 2 section 5).
[... cut ...]
[D0 00 0D 00 03 21]  Transponder address 3 (short) absent at   LR14 () (BXP88 Board ID 2 section 6).
[D0 20 0E 00 03 02]  Transponder address 3 (short) present at   LR15 () (BDL16x Board ID 1 RX4 zone H or BXP88 Board ID 2 section 7).
[D0 00 0E 00 03 22]  Transponder address 3 (short) absent at   LR15 () (BDL16x Board ID 1 RX4 zone H or BXP88 Board ID 2 section 7).
[D0 20 0F 00 03 03]  Transponder address 3 (short) present at   LR16 () (BXP88 Board ID 2 section 8).
[D0 00 0F 00 03 23]  Transponder address 3 (short) absent at   LR16 () (BXP88 Board ID 2 section 8).
```

pabender, 2018-09-09:
How do we handle this when the board is neither a BDL16x or BXP88?

I'm asking because I am thinking about the case of the Roco 10808 on a Z21 when we ask it to report as a LocoNet device.

On a related note on this, when did we start putting the JMRI device names into the translated messages?

Requiring that the monitor code knows about the loconet managers involved in the communication makes it a lot harder to translate these messages when the managers can't be provided.  (on the Z21, the messages are tunneled, and the tunnel doesn't know anything about the managers, so we can't embed the correct LocoNet translations into the Z21 monitor. )

I would like to add a toMonitorString method to the jmri.jmrix.Message interface (which LocoNet messages don't currently implement, but probably should ).  This method would handle all the translations into human readable text.  There are several systems that already implement this (Ieee 802.15.4 systems, powerline, XPressNet, RFID, and Roco ). 

I think from an OO design perspective, this puts the translation in the right spot and it allows us to 1) simplify the monitors and 2) put the translation into a place where it can more easily be accessed and tested.  

To implement this on loconet requires breaking the connection to the managers, because there is no reason why a message should know about a manager.

devel-bobm, 2018-09-09:
@pabender,

Since I do not have any Roco hardware, I am unable to report any inferences of Roco hardware identity when interpreting LocoNet transponding messages.  As such, `LocoNet Monitor` will show the messages as if they were coming from the comparable Digitrax hardware.

Device names have been in `LocoNet Monitor` information for about as long as I have been an active deveoper - that's perhaps 8 to 10 years now.  That addition was based on a user's request, on account of LocoNet device "addressing" being so bizarre.

If I understand the history correctly, the `LocoNet Monitor` was one of the first "monitor" functions to exist, so it uses very early JMRI methodologies.  Including device-specific information was bolted on over time.  The LocoNet Monitor code has grown and grown as new LocoNet features were introduced or existing ones "figured out".  It is now a monster.

I cannot speak to the `jmri.jmrix.Message` concepts you mention - I have never studied the other code you mention, or the `jmri.jmrix.Message` interface.  

Decoding and interpreting LocoNet messages into human-readable text is a huge chunk of code, and any systematic changes to the existing scheme will take a great deal of painstaking work.  I am not against the idea, but I am not looking forward to the amount of rework necessary!

The biggest issue I have is that there are many LocoNet messages which do not have associated JMRI "managers".  How would these be handled?

Another concern I have would be how to handle the cases where one LocoNet message can be interpreted in two (or more) different ways by different pieces of hardware, as is the case for variants of OPC_PEER_XFER?  A wide variety of users have appropriated this opcode for their own use, without negotiating with Digitrax.  How would the "negotiation" work between "managers" and whatever manager-proxy is put into place to handle "non-managed" messages, to ensure that each message gets interpreted exactly once, and to ensure that the interpretation is predictable and not subject to something like "class-load order"?

In a "big picture" view of LocoNet-related JMRI functionality, refactoring `LocoNet Monitor` is fairly low on my personal list of priorities.  If someone else comes along to work on this, I'll support it.  I just don't see myself putting any significant work into it in the next year or so.

devel-bobm, 2018-09-09:
I have managed to get my BXPA1 to emit transponding devices, albeit somewhat erratic.  I have determined that there is a 1:1 relationship between BXPA1 Device ID and the associated JMRI `Reporter` Number.

pabender, 2018-09-09:
Sent from my iPad

> On Sep 9, 2018, at 3:40 PM, Bob Milhaupt <notifications@github.com> wrote:
> 
> @pabender,
> 
> Since I do not have any Roco hardware, I am unable to report any inferences of Roco hardware identity when interpreting LocoNet transponding messages. As such, LocoNet Monitor will show the messages as if they were coming from the comparable Digitrax hardware.
> 
My argument here is that we shouldn’t be including specific device information for every possible device in the translated strings.  That gets very unwieldy, and it already is with only two devices listed in the translated strings.
> Decoding and interpreting LocoNet messages into human-readable text is a huge chunk of code, and any systematic changes to the existing scheme will take a great deal of painstaking work. I am not against the idea, but I am not looking forward to the amount of rework necessary!
> 

Right, it is a huge chunk of code.  I had the same problem with the XpressNet monitor a couple of years ago.

Because of the way the code was interlaced with the monitor, it was nearly impossible to test the message translation code.  I basically lifted the message translation code verbatim from the monitor into the toMonitorString methods for XNetReply and XNetMessage

The LocoNet monitor is much better tested, but it appears that the Llnmon format() method implements what would be the toMonitorString() method in LocoNetMessage.
> The biggest issue I have is that there are many LocoNet messages which do not have associated JMRI "managers". How would these be handled?
> 
They are all messages.  They should be handled by toMonitorString.
> Another concern I have would be how to handle the cases where one LocoNet message can be interpreted in two (or more) different ways by different pieces of hardware, as is the case for variants of [...]
All of this SHOULD be irrelevant.  toMonitorString is an operation performed on a message object.  The message object should be the arbiter of how that message is interpreted.

devel-bobm, 2018-09-09:
At least part of my confusion stems from the idea that "a Manager" must be responsible for a `toString` or `toMonitorString` implementation.  I immediately thought that the `LnReporterManager` would be responsible for interpreting the transponding messages, and that `LnTurnoutManager` would be resonsible for interpreting turnout-related messages, and that `SlotManager` would be responsible for interpreting all slot-related messages, and ...  From an OO standpoint, this makes a great deal of sense to me.  I further assumed that `LocoNetMessage#toMonitorString()` would be responsible for directing the message to the correct manager.  I now understand that this is _not_ the intention.

`My argument here is that we shouldn’t be including specific device information for every possible device in the translated strings. That gets very unwieldy, and it already is with only two devices listed in the translated strings.`

Yes it is unwieldy.  But it is important in a world where manufacturers provide minimal technical details about how their equipment works, except to say that users should "avoid having two devices which use the same address" without ever defining what addresses the device uses, or how differing address spaces relate to each other or even overlap.  

Note that the device-specific information is not provided for "all" devices - just the ones where it is not obvious how the addressing is configured.  For hardware where a "magic board ID" applies, the code attempts to reverse-engineer and report the possible source or destination in the message.  Where there is no "magic board ID" or similar concept, such as for RR-CirKits products, there is no text, even though a RR-CirKits product may either generate or listen to the particular message.

A fairly large number of JMRI users have praised how LocoNet Monitor reports board ID and port number, and the way it can report user names for various items.  I will not make a guess of how many would complain if those features were to disappear.

pabender, 2018-09-10:
> At least part of my confusion stems from the idea that "a Manager" must be responsible for a toString or toMonitorString implementation. I immediately thought that the LnReporterManager would be [...]

Right, the managers shouldn't be involved in building the messages in the monitor at all.  They aren't for any other system, at least as far as I can tell.

Even for the existing implementation of the LocoNet monitor, I believe the only place the monitor asks the managers to do anything is to get the device names for the messages.