JMRI/JMRI #5774: Refactor LocoNet Monitor to use jmri.jmrix.Message

devel-bobm, 2018-09-09:
Improve the implementation of LocoNet Monitor by making use of the `jmri.jmrix.Message` interface.  This would make it possible for "tunneling" connections to use "LocoNet Monitor" message interpretations without the "tunneling" mechanism having to know about LocoNet managers.  This would specifically help interpret "tunneled" LocoNet-style messages as seen with the Roco Z21.

Ref: the first @pabender comment in #5764.

bobjacobsen, 2018-09-09:
(As a preliminary thing, [jmri.jmrix.Message](http://jmri.org/JavaDoc/doc/jmri/jmrix/Message.html) is a base for lots of message types, but not yet for [LocoNetMessage](http://jmri.org/JavaDoc/doc/jmri/jmrix/loconet/LocoNetMessage.html) - it would be good to add that.)

I think that the "[concise but informative representation](https://docs.oracle.com/javase/8/docs/api/java/lang/Object.html#toString--)" contract of `toString()` should be honored.  The existing [LocoNetMessage#toString()](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/loconet/LocoNetMessage.java#L118) code provides "a String representation of the entire message in hex" which seems reasonable for that.

Hence I'd prefer that we add another "getSummaryVersion()" method (with perhaps an even better name) that provides the equivalent of the LocoNet Monitor display text. For ease of migration, that could `default` to the `toString()` result in the interface at first, but in the long run having the decoding go through a different method is best.

One possible fly in the ointment:  LocoNetMessage and similar don't retrain any information about the system they're associated with, but a separate object like LocoNetMonitor can.  If that's needed for the display, this might get complicated and/or inefficient.

devel-bobm, 2018-09-09:
@bobjacobsen  wrote: `One possible fly in the ointment: LocoNetMessage and similar don't retrain any information about the system they're associated with, but a separate object like LocoNetMonitor can. If that's needed for the display, this might get complicated and/or inefficient.`

So far as I remember, the existing LocoNet Monitor implementation does not retain any state from one message to the next.

devel-bobm, 2018-09-09:
@bobjacobsen wrote: 

> I think that the "concise but informative representation" contract of toString() should be honored. The existing LocoNetMessage#toString() code provides "a String representation of the entire message in hex" which seems reasonable for that.

Where would `LocoNetMessage#toString()` be used?  Are you proposing that LocoNet Monitor use that instead of the verbose text currently being displayed?  Are you proposing that a "tunneling" implementation would use `LocoNetMessage#toString(), while LocoNet Monitor would use `LocoNetMessage#getSummaryVersion()`?  Or do you have something else in mind?

bobjacobsen, 2018-09-09:
> So far as I remember, the existing LocoNet Monitor implementation does not retain any state from one message to the next.

Sorry, you missed my point.  

Looking a little more deeply (based on Paul’s comments), jmri.jmrix.loconet.locomon.Llnmon has a reference to the LocoNetSystemConnectionMemo which is uses to customize the message output.

LocoNetMessage itself does _not_ have one of those.

bobjacobsen, 2018-09-09:
> Where would LocoNetMessage#toString() be used?

That’s not the point.  toString() does what it does, and it should keep doing that.

> Are you proposing that LocoNet Monitor use that instead of the verbose text currently being displayed?

No, quite on the contrary.  I’m proposing that you _NOT_ use jmri.Message#toString for the current verbose text. Instead, add a new method to the jmrix.Message interface for that.

> Are you proposing that a "tunneling" implementation would use LocoNetMessage#toString(), while LocoNet Monitor would useLocoNetMessage#getSummaryVersion()`? Or do you have something else in mind?

I’m proposing that both use the same method, not called toString.  Paul’s comment suggested calling it toMonitorString(), which is fine with me,

devel-bobm, 2018-09-09:
> Looking a little more deeply (based on Paul’s comments), jmri.jmrix.loconet.locomon.Llnmon has a reference to the LocoNetSystemConnectionMemo which is uses to customize the message output. LocoNetMessage itself does _not_ have one of those.

Presumably, this means that `LocoNetMessage#toMonitorString()` would need to make use, internally, of `InstanceManager.getDefault(beantypeMananger.class)` in order to get a manager which is responsible for a given bean type, then use that manager to get further information, such as the "user name" for a given "system name" of that bean type.  Correct?

bobjacobsen, 2018-09-09:
> Presumably, this means that LocoNetMessage#toMonitorString() would need to make use, internally, of InstanceManager.getDefault(beantypeMananger.class) in order to get a manager which is responsible [...]

I haven’t looked at Llmon.java in detail.  If it’s using the prefix for a particular system (“L” vs “L2” vs whatever the user called it), then the specific SystemConnectionMemo has to be found.  You can’t just ask the InstanceManage for e.g. the LocoNet connection, because there might be more than one.

But it’s not clear to me that’s necessary.  Having the monitor text say “LS1” vs “L2S1” seems an over-specialization to me.

devel-bobm, 2018-09-09:
Yes, Llnmon.java does use specific prefixes.  Sensor messages seen on the L2 connection are decoded with respect to prefix "L2".  That way the displayed "user names" match both the connection and the sensor number.  This is currently only done for Turnouts, Sensors, and Reporters.  

Presumably, the current code's concept could be extended to figure out user names for SE8C Signal Heads.  And perhaps some other Bean types I haven't thought of.

pabender, 2018-09-09:
Sent from my iPad

> On Sep 9, 2018, at 6:05 PM, Bob Jacobsen <notifications@github.com> wrote:
> > Are you proposing that a "tunneling" implementation would use LocoNetMessage#toString(), while LocoNet Monitor would useLocoNetMessage#getSummaryVersion()`? Or do you have something else in mind?
> 
> I’m proposing that both use the same method, not called toString. Paul’s comment suggested calling it toMonitorString(), which is fine with me,

The reason for calling it toMonitorString() is that there are at least 5 existing packages where the messages and replies call it toMonitorString().

To further Bob’s point on this, each one of these implementations provides both toString and toMonitorString.  The respective monitors use the toString method for the RAW message and the toMonitorString method for text translated to a human readable form.

The Z21 monitor, where this matters today, calls the toMonitorString for XPressNet messages.  For LocoNet messages, it currently creates an Llnmon object to translate those messages, but it passes null for the memo, so it can’t translate everything.

Paul

pabender, 2018-09-09:
Sent from my iPad

> On Sep 9, 2018, at 6:53 PM, Bob Milhaupt <notifications@github.com> wrote:
> Yes, Llnmon.java does use specific prefixes. Sensor messages seen on the L2 connection are decoded with respect to prefix "L2". That way the displayed "user names" match both the connection and the sensor number. This is currently only done for Turnouts, Sensors, and Reporters.
> 

I would actually prefer that it doesn’t have that hard link to the system name at all.  It seems to be a very heavyweight process that prevents some legitimate uses.

devel-bobm, 2018-09-09:
I propose a "lightweight" and a "heavyweight" version of `toMonitorString()`.  The "system prefix" is a required parameter of the "heavyweight" version, while the "lightweight" version omits the parameter.  The heavyweight verion would use the prefix parameter it to look up "user names" where applicable.  The "lightweight" version does not have the prefix parameter, and will not ever look up "user names".  

The interface could, by default, pass the "heavyweight" invocation to the "lightweight" method and simply drop the prefix parameter into the bit-bucket.  The concrete implementation gets the "no heavyweight implementation" for free, or can choose to do the extra work to provide the "heavyweight" functionality.

Perhaps the LocoNet implementation would be the first (only?) "heavyweight" version.

pabender, 2018-09-09:
@devel-bobm to the best of my knowledge, there are no other monitors that currently try to associate any devices with their system names.

I'm working on updating the interface to include the toMonitorString() method, which includes a default that just calls toString().

Once that is done, I'm going to start do some refactoring to the monitors and messages for the systems that already use the interface.