JMRI/JMRI #5783: Edit sensor debounce -- java.lang.ClassCastException

dsand47, 2018-09-10:
As of 4.13.3, when saving updated debounce values using the sensor edit dialog a ClassCastException occurs:
```
2018-09-10 15:02:59,726 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.ClassCastException: java.lang.Integer cannot be cast to java.lang.Long
	at jmri.jmrit.beantable.beanedit.SensorDebounceEditAction.saveDebounceItems(SensorDebounceEditAction.java:113)
	at jmri.jmrit.beantable.beanedit.SensorDebounceEditAction$2.actionPerformed(SensorDebounceEditAction.java:94)
	at jmri.jmrit.beantable.beanedit.BeanItemPanel.saveItem(BeanItemPanel.java:19)
	at jmri.jmrit.beantable.beanedit.BeanEditAction.save(BeanEditAction.java:381)
	at jmri.jmrit.beantable.beanedit.SensorEditAction.save(SensorEditAction.java:1)
	at jmri.jmrit.beantable.beanedit.BeanEditAction.applyButtonAction(BeanEditAction.java:284)
	at jmri.jmrit.beantable.beanedit.BeanEditAction$6.actionPerformed(BeanEditAction.java:256)
```
A recent change to **SensorDebounceEditAction** changed the spinners from **Long** to **Integer** (commit 6d87aec).

I have identified two possible solutions:
- Long time = Long.valueOf(String.valueOf(sensorDebounceActiveSpinner.getValue()));
- Long time = new Long(String.valueOf(sensorDebounceActiveSpinner.getValue()));

There are most likely other solutions that I am not aware of.

bobjacobsen, 2018-09-10:
That's my commit, so I'll get right on this.

bobjacobsen, 2018-09-10:
PR #5784 should be a fix for this. It also adds a bunch of test coverage.

bobjacobsen, 2018-09-10:
@dsand47 Would you have a chance to check that fix in PR #5784?  it works for me, and I've added tests for two cases, but since I missed it the first time...

It's enough of a regression that I should respin the 4.13.3 test release before publishing it.

dsand47, 2018-09-10:
I tested with a range number up to 9999999 and it works as expect. Looks good.

dsand47, 2021-03-12:
Fixed by #5784.

closed 2021-03-12 by dsand47