JMRI/JMRI #5829: fast clock changes not always represented in WiThrottle protocol.

davidzuhn, 2018-09-19:
If the Fast Clock Rate is changed on the Fast Clock Setup screen, the current time and the new rate are immediately pushed out on the WiThrottle connection.  That makes perfect sense to me -- something changed, let everyone know about it.

However, just changing the time (via the Set button for Fast Clock Time) does NOT send any immediate update to WiThrottle connection.   The next periodic update will be correct, but that could be 5 (real, 1:1) minutes away.   

I think any change to the time settings ("current fast time" or the clock rate) should warrant an immediate message on the WiThrottle connection.   If the rate changes, it needs to be the full TIME<;>RATE message.   If only the time changes, it could be either the TIME<;>RATE or just the TIME message.

davidzuhn, 2018-09-19:
I'll note that this is with release 4.12

devel-bobm, 2018-09-19:
I wonder if this is system-connection-type-dependant.  What system is this seen with?

davidzuhn, 2018-09-19:
This is with LocoNet Simulator.

devel-bobm, 2018-09-19:
Although I have no direct knowledge of how the Fast Clock tool interacts with the WiThrottleServer, I would not be surprised to find that this problem might be a nuance of JMRI's Fast Clock support via LocoNet.  I have seen odd behavior of the JMRI FastClock tool in a case where only LocoNet-connected clocks were used (without WiThrottleServer) during operating sessions, but was not able to debug, and resolved the problem by changing the clock using a throttle, so ignored JMRI's FastClock tool.

I do not have any WiThrottle-capable devices, so cannot easily do any testing.  Is this something which you can easily reproduce?  Can you provide a LocoNet Monitor trace showing the LocoNet messages around the problematic FastClock tool activities?

davidzuhn, 2018-09-19:
This has been completely repeatable.   You don't really need a WiThrottle device, if you've got a tool like 'telnet' or 'nc' available to make the TCP connection (I can write up the sequence to manually type to do this, it'll be pretty simple since it's a text based protocol and this is a very isolated issue).   

I'll check it again on Loconet (non-Simulator, DCS-100) and PiSprog, and if there's any change in behavior I'll document it and provide trace files.

bobjacobsen, 2018-09-19:
The problem may be in jmri.jmrit.simpleclock.SimpleTimebase  

I haven't put a debugger on it, just read the code, so this might not be the correct class nor the correct understanding of the class.  Given that, it seems to fire property change events on `setRate(..)` and `setRun(..)` but not send them on `setTime(..)`

davidzuhn, 2018-09-19:
BobJ, that does look suspiciously like the root of the problem to me.

BHoffman351, 2018-09-20:
So I had looked into setting values in the fast clock previously and ran into a few problems.  I believe that things worked better if JMRI is used as a clock source, but problems occur when the source is external.  

There is a race condition that occurs.  If WiThrottle sends a command to set time and rate, JMRI separates it, sets time, sends it to LocoNet, then sets rate and sends it to LocoNet.  The Loconet echos the message, but only is addressing the first message with a time change.  That echoed message has the new time, but the old rate.  This sets the JMRI rate back to what it was, which then gets sent out to LocoNet again.  Now the command station handles the first rate change which now gets echoed to JMRI and changes it to the new rate, while the old rate is staged again to go to the command station.

Long story short:  As fast as it can the LocoNet traffic will be fast clock messages alternating between two rates.

The JMRI clock control window does not have the issue because there is enough lag time between pressing a time set button and pressing a rate set button.  Put the commands close enough together and the loop begins.

I bring this up as a cautionary, just in case adding a property change event on setTime() creates a similar loop.

Other systems than LocoNet are less likely to have an issue as the time and rate should be in separate packets.

Regards,
Brett

davidzuhn, 2018-09-20:
I'll note that while I have a loconet based system, I have no interest in using the loconet system clock.

So any options available (I'm away from the setup screen right now) to remove the hardware synchronization would useful to me to avoid this possible race condition.   I am only concerned about JMRI host time as master and the messages via the WiThrottle protocol.

devel-bobm, 2018-09-20:
I've spawned new issue #5835 to note @Bhoffman's issue (above).

devel-bobm, 2018-09-20:
@davidzuhn , are you ever using a WiThrottle agent to send a "set the time" message back to JMRI via WiThrottle protocol, or is fast time set only thru non-WiThrottle means?

davidzuhn, 2018-09-20:
I don't believe that the WiThrottle protocol has any means to set the time -- as far as I've seen the protocol just has the server (JMRI) send periodic state messages to the clients, either the current time every 5 minutes, or the current time and the time rate at startup or when the rate changes.  

My use case doesn't require any additions to this -- I'm just distributing clocks around the layout, but instead of needing them all to be wired in to some layout communication bus, all I need on my device is 3-5V DC (which I'm providing via USB power chargers or battery).

bobjacobsen, 2018-09-23:
PR #5857 has a check that the jmri/jmrit/simpleclock/SimpleTimebase class _does_ send the minute events if the new time is different.  (It's done in a different place than I thought).  So that was a red herring.  Sorry for the confusion.

How do you have the "Set Up Fast Clock" window set?  Something like this? (Not recommending this one, just showing what I was checking with)

[image]

davidzuhn, 2018-09-25:
I've run my examples with the loconet monitor running, and there's NO loconet traffic whatsoever.   Since I've got nothing else running on the box right now (no throttles, etc.), that's not too surprising to me.  

I also see the same behavior when using the PiSprog as my command station.

[image](https://user-images.githubusercontent.com/1703466/45985623-b86c4f00-c02c-11e8-91ab-8b96fe504def.png)

devel-bobm, 2018-09-25:
Be aware that the JMRI "LocoNet Simulator" connection type can have cases where JMRI LocoNet activity is not reported in the LocoNet Monitor window.  I do not know if the Fast Clock Setup tool is one of those cases.

I do not know the root-cause of this behavior.  I have not worked with LocoNet simulator much of late can cannot give other examples.  

It may be possible to expose the "missing" traffic by setting DEBUG-level logging for the LnPacketizer class.  That level of logging could be overwhelming for some use-cases, but could be adequate for one-time debugging of this problem.

bobjacobsen, 2018-09-25:
New hypothesis (why not go for three wrong ones in a row?) is that the following code in `java/src/jmri/jmrit/withrottle/FastClockController.java` is sending every Nth _update_, not every Nth _minute_.  It's called when the timebase fires a minutes-changed event. It counts those to decide when to send over the network. It should be changed to keep track of how much the time has changed, not how many updates it's gotten.

```
    /**
     * Send just time.
     * <P>
     * Use to synchronize time on Wi-Fi devices to nearest second. Send no rate.
     */
    public void sendFastTime() {
        updateMinuteCount++;
        if (updateMinuteCount >= updateMinsSetpoint) {
            if (listeners != null) {
                for (ControllerInterface listener : listeners) {
                    listener.sendPacketToDevice("PFT" + getAdjustedTime());
                }
            }
            updateMinuteCount = 0;
        }
    }
```

mstevetodd, 2020-02-13:
closing via #8081

closed 2020-02-13 by mstevetodd