JMRI/JMRI #5835: Issue setting Fast Clock from WiThrottle

devel-bobm, 2018-09-20:
As mentioned in #5829 by @BHoffman351, WiThrottle can have an issue setting the Fast Clock time on LocoNet-based systems.

> There is a race condition that occurs. If WiThrottle sends a command to set time and rate, JMRI separates it, sets time, sends it to LocoNet, then sets rate and sends it to LocoNet. The Loconet echos the message, but only is addressing the first message with a time change. That echoed message has the new time, but the old rate. This sets the JMRI rate back to what it was, which then gets sent out to LocoNet again. Now the command station handles the first rate change which now gets echoed to JMRI and changes it to the new rate, while the old rate is staged again to go to the command station. Long story short: As fast as it can the LocoNet traffic will be fast clock messages alternating between two rates.

devel-bobm, 2018-09-20:
@BHoffman351 - does the WiThrottle send two different messages - one to set the rate and one to set the time, or just one message?

If two messages, then perhaps three would have a better result?  Set the rate to 0, then set the desired fast-clock time, then set the desired rate?

(I have no idea how this would affect other systems which have fast-clock implementations.)

BHoffman351, 2018-09-20:
> On Sep 20, 2018, at 12:39 PM, Bob Milhaupt <notifications@github.com> wrote:
> does the WiThrottle send two different messages - one to set the rate and one to set the time, or just one message?
> 
> If two messages, then perhaps three would have a better result? Set the rate to 0, then set the desired fast-clock time, then set the desired rate?
> 
It sends no messages to set time or rate.  It does not work so I did not add another problem to the mix.

The problem can be solved by adding a delay between messages in JMRI, but adding a delay simply masks the real issue.

devel-bobm, 2018-09-20:
I see the problem a little bit differently.  With LocoNet, there is no need to send two different messages, one for the rate and one for the time - they are both encoded in the same type of LocoNet message.

So, the problem would appear to be more like "why does a WiThrottle attempt to set the fast rate and fast time cause a 'race' condition when used with LocoNet?" in the problematic case mentioned above.  

Is this problematic case reproducable from the JMRI Fast Clock tool?  If so, then I could look into it.

Presumably, this type of problem could be avoided altogether with LocoNet if a single request is made to LocoNet and a single LocoNet message is sent.  Or by requesting that the clock be stopped, then, after a dely, requesting that the clock be re-set, then requesting that the clock be started at some rate.

bobjacobsen, 2018-09-20:
I think there’s a bit of confusion here. 
> So, the problem would appear to be more like "why does a WiThrottle attempt to set the fast rate and fast time cause a 'race' condition when used with LocoNet?" in the problematic case mentioned above.
> 
AFAIK WiThrottle makes no request. 

david sees the delayed response after he makes the change directly in JMRI. 

The timing concern is about setting locally, while connected to LocoNet. 
> Is this problematic case reproducable from the JMRI Fast Clock tool? If so, then I could look into it.
> 
david is seeing it from the fast clock tool.

BHoffman351, 2018-09-21:
> On Sep 20, 2018, at 5:53 PM, Bob Jacobsen <notifications@github.com> wrote:
> 
> AFAIK WiThrottle makes no request.

That is correct.  Perhaps it would be more appropriate to restrict not only WiThrottle but JMRI from making fast clock changes when it is using an external time source such as a LocoNet clock?  I’m not sure that would be the right thing to do, but it would seem to fit with the terms used to describe it.

The problem is that a change to the time is not promptly sent to devices connected through the WiThrottle Server.  I was just bringing up the race condition as something to watch for if a property change is added to the setTime().

Regards,
Brett

devel-bobm, 2018-09-21:
I guess I am confused.  @BHoffman351 reported:

> If WiThrottle sends a command to set time and rate, JMRI separates it, sets time, sends it to LocoNet, then sets rate and sends it to LocoNet.

So WiThrottle can (or did) make a "set the fast clock" request.  Apparantly that request caused the reported failure.

Pugwash1, 2018-09-21:
When a new time is set in the GUI, SimpleClockFrame.java, it ambles its way down thru SimpleTimeBase.java which fires a property change for "minutes".
This is picked up by LCD clock for instance, which takes a Brick House approach and always updates everything regardless of what changed, but wifithrottle/FastClockController looks at the specific event "minutes" and then says, "I am only updateing this every 5 minutes", and doesn't send it out, because the five minutes aren't up.

Now. On the off topic of LnClockControl.java. This does some truly awesome stuff, but its not obvious and not always right. So starting from "internal clock" in the time source and switching to loconet, it immediately updates the rate and part of the time to whats in the gui. And never ever asks for the time again...... But as a slave if it doesnt see a time packet "E7 0E ......" go by its supposed to ask for one after 90 secs or so "BB  7B....."   

Further. A race condition exists if you say source "LocoNet Fast Clock" and check "Correct hardware clock", but who who try that?  "Internal Comp.." and "Synchronized..." results in "E7 0E.." and "EF 0E.." messages (here's the time, set the time)

Loconet clocks are either Slaves or  The Master. And that choice is missing from the GUI. If its a Slave it process the time packet "E7 0E.." (if it doesnt see one in 90secs it should request one) and obeys it. (new time, new rate, start counting, no fancy adjustments). If its the master it should send the "E7 0E..." message every 60 secs say, enough to stop anyone asking for the time, ignore any incoming "E7 0E..." and, optionally, change parameters based on "EF 0E...". 

In the middle of all this is the use of the old java date stuff resulting in this "@SuppressWarnings("deprecation")" the most common statement in the code.

FWIW.

BHoffman351, 2018-09-21:
I am the author of WiThrottle. No released version ever had fast clock control in it because it could create the problem. 

Regards,
Brett 

> On Sep 21, 2018, at 6:41 AM, Bob Milhaupt <notifications@github.com> wrote:
> 
> I guess I am confused. @BHoffman351 reported:

devel-bobm, 2018-09-21:
And this is why I am confused.  I cannot work on a problem that cannot be quantified or reproduced.

bobjacobsen, 2018-09-21:
This thread is messed up enough that I think we should close it and start a new one.  Will do that shortly.

closed 2018-09-21 by bobjacobsen

bobjacobsen, 2018-09-21:
See #5845