JMRI/JMRI #5849: Add "Restart" button to "JMRI is unable to connect" window

devel-bobm, 2018-09-22:
I really like the idea behind the new  #5771 "JMRI is unable to connect" window feature!

But it does not seem to allow easy recovery from the thing I usually do.  If I simply forget to plug in the USB cable, I will get the "JMRI is unable to connect" menu.  But I have to jump thru a few hoops to get JMRI to simply retry the same connection after I have plugged in the USB cable.  First I "start with a new profile" then "cancel".  Then I get back to the "Set active profile" window where I can simply wait 10 seconds or press "enter" to accept the selected profile.

Can this new window feature be enhanced with a new button that will simply re-try the same connection as before?  Or simply to return to the "Set active profile" window?

bobjacobsen, 2018-09-22:
> But it does not seem to allow easy recovery from the thing I usually do. If I simply forget to plug in the USB cable, I will get the "JMRI is unable to connect" menu. But I have to jump thru a few [...]

Why do you do all that instead of “Exit Program”, plug cable, restart?

devel-bobm, 2018-09-22:
Because it is faster to do this when running in my IDE of choice.  I see the message, plug in the cable, then press a couple of buttons and JMRI is happy.  If I have to quit JMRI, I then need to get the IDE to re-build JMRI, which takes 15 or more seconds to begin launching JMRI...

So what I'm asking is to make my "press a couple of buttons" in to "press a button", for the purpose of shaving 15 seconds down to a couple seconds.  Call me lazy!

bobjacobsen, 2018-09-22:
The first place to add that button is probably in the ErrorDialog constructor that starts at line 364 in `jmri.implementation.JmriConfigurationManager`.  You can see how the three existing buttons are created there. They return an enum value to the switch statement that handles the operation around line 235.  I think you just want your new button & new case to do `AppsBase.handleRestart();`

danielb987, 2018-09-22:
PR #5855 fixes this.

danielb987, 2018-09-26:
@devel-bobm : Does my PR #5855 do what you want it to do? If so, this issue could be closed.

devel-bobm, 2018-10-01:
Yes - it works admirably.  Closing this issue.

closed 2018-10-01 by devel-bobm