JMRI/JMRI #5862: Jacoco report - sort by missed lines

danielb987, 2018-09-24:
I have made a [script](http://mjtrafik.se/jmri/parse_jmri.php) that reads the Jacoco report and sort it by number of missed lines. The original report sorts by percentage which means that a class with 0% coverage and 10 missed lines comes before a class with 10% coverage and 1000 missed lines.

Is this of interest for other JMRI developers?
If so, is it of interest to host it on JMRI web servers?

The [source code](http://mjtrafik.se/jmri/parse_jmri.php.txt).

danielb987, 2018-09-24:
It may be worth noting that it's the package list and class list that is sorted by number of missed lines, not individual classes.

pabender, 2018-09-24:
Interesting.  I still need to be able to sort by different columns though, which is I don’t appear to be able to do from the output of your script.

Also, I should note that if you produce a jacoco report locally ( using the ant alltest-coveragereport target ) the report has more sorting options than the Jenkins Jacoco report allows.

danielb987, 2018-09-24:
I can easily add sorting of different columns if you want that. I was lazy and only did sorting on missed lines, but it would be easy to add sorting of the other columns as well.

bobjacobsen, 2018-09-24:
Very nice!  

I recommend you put it in the scripts/ subdirectory of the JMRI projects. Then
- we can link it from e.g. the head of the Jenkins JaCoCo project.  That'll make it a clickable option on that page.
- people can run it locally on their own jacoco.exec file. That needs a bit of infrastructure, i.e. a script that does something like (on macOS)
```
php scripts/parse_jmri.php > temp.html
open temp.html
```

danielb987, 2018-09-24:
Thanks!

PR #5864 have this script. I changed the name of the script to make it more clear what it does.

[jacoco_sort_report.php](http://mjtrafik.se/jmri/jacoco_sort_report.php)
[jacoco_sort_report.php?package=jmri.jmrit.logix](http://mjtrafik.se/jmri/jacoco_sort_report.php?package=jmri.jmrit.logix)

danielb987, 2018-09-25:
Does the PHP script works for you now? I followed the link on the [CI-status.shtml](http://jmri.sourceforge.net/help/en/html/doc/Technical/CI-status.shtml) page to the [script](http://jmri.sourceforge.net/help/en/html/doc/Technical/jacoco_sort_report.php).

I got the error message:
Fatal error: Call to a member function getElementsByTagName() on a non-object in /home/project-web/jmri/htdocs/help/en/html/doc/Technical/jacoco_sort_report.php on line 192

It seems that the script either cannot load the Jacoco web page, or that the script cannot parse the page.

bobjacobsen, 2018-09-25:
That server is obsolete.  Use the jmri.org version of the link, please.

> On Sep 24, 2018, at 10:42 PM, Daniel <notifications@github.com> wrote:
> 
> Does the PHP script works for you now? I followed the link on the CI-status.shtml page to the script.
> 
> I got the error message:
> Fatal error: Call to a member function getElementsByTagName() on a non-object in /home/project-web/jmri/htdocs/help/en/html/doc/Technical/jacoco_sort_report.php on line 192
> 
> It seems that the script either cannot load the Jacoco web page, or that the script cannot parse the page.

danielb987, 2018-09-26:
I close this issue since the script is now on [JMRI CI status page](http://jmri.org/help/en/html/doc/Technical/CI-status.shtml).

closed 2018-09-26 by danielb987