JMRI/JMRI #5869: Questions about writing tests

danielb987, 2018-09-24:
Can there be several test classes testing one single class file? If so, which naming convention is recommended? [jmri.jmrit.display.layoutEditor.LayoutEditorTools](http://jmri.tagadab.com/jenkins/job/Development/job/JaCoCo/lastStableBuild/jacoco/jmri.jmrit.display.layoutEditor/LayoutEditorTools/) has over 9 000 lines of code and over 8 000 lines of these are untested. 4% of the total number of untested lines in JMRI is in this class.

Is it allowed to use Reflection to test private methods and access private fields in a test?
LayoutEditorTools has lots of private methods. It is probably possible to test these methods without Reflection, but it would make it easier to focus on the private methods directly.

pabender, 2018-09-25:
Sent from my iPad

> On Sep 24, 2018, at 6:54 PM, Daniel <notifications@github.com> wrote:
> Can there be several test classes testing one single class file? If so, which naming convention is recommended? jmri.jmrit.display.layoutEditor.LayoutEditorTools has over 9 000 lines of code and over 8 000 lines of these are untested. 4% of the total number of untested lines in JMRI is in this class.
> 
The entire jmri.jmrit.display.layoutEditor package is a miserable mess that is largely unmaintainable as it sits today.

I think there are some huge opportunities for refactoring and code cleanup that will reduce the testing load, but we really need tests in place to reduce the risk of breaking anything in the process.

I’ve been picking at adding tests to the JMRI.jmrit.display package and it’s sub packages for a while.

If you want to add more test files, please add what makes sense to you.  As long as the class name ends in Test, and the name is logical, we can figure out what it tests by the content.
> Is it allowed to use Reflection to test private methods and access private fields in a test?
> LayoutEditorTools has lots of private methods. It is probably possible to test these methods without Reflection, but it would make it easier to focus on the private methods directly.
> 

Yes.  We already have some tests that use reflection, both for accessing methods and variables.

Paul

bobjacobsen, 2018-09-25:
if the eventual goal is to improve the LayoutEditor code (which is a _very_ _good_ goal), then it'd be a good thing to modify it to make it more testable.  

- If an internal private method is a testable unit, then just change it to package access and test it directly. 

- But don't get too wrapped up in testing each micro-component routine. Consider whether a better implementation of LayoutEditor might not even have that routine, in which case that test wouldn't really be a long term contribution.  When refactoring, it's often better to test from the API, even if that's incomplete, and then add to the testing of _refactored_ code as that's done.

- If some chunk of functionality looks like a coherent piece of (some future) implementation, then write the tests for that as a separate *Test class.  That way, when the code gets refactored into another class, the tests are already somewhat separate.   But don't separate tests into multiple files just based on volume, because you'll likely have to redo that when you do figure out a better way to segment the code.

danielb987, 2018-09-25:
When I started writing tests for Conditional and friends, I didn't had a clue about what it did. And for many of those tests I don't know why a test should give a certain result, I only assumed that the current result is correct and did a test that ensured that the result would be the expected. Now I start to understand much, but not all, of the Conditional code.

My question was about if I could do the same thing with LayoutEditor. Start to write tests about small parts of the code, without understanding what that piece does in the bigger part.

But there are many other parts of the code that I could work on, so it's no problem. Maybe take a look at jmri.Section for example.

pabender, 2018-09-25:
Sent from my iPad

> On Sep 25, 2018, at 8:45 AM, Daniel <notifications@github.com> wrote:
> 
> When I started writing tests for Conditional and friends, I didn't had a clue about what it did. And for many of those tests I don't know why a test should give a certain result, I only assumed that the current result is correct and did a test that ensured that the result would be the expected. Now I start to understand much, but not all, of the Conditional code.
> 
> My question was about if I could do the same thing with LayoutEditor. Start to write tests about small parts of the code, without understanding what that piece does in the bigger part.
> 
You certainly can.  I would appreciate any new tests added to this part of the code that helps our understanding of how the pieces fit together.

As Bob said,  testing the public API is probably best, at least as a starting point.
> But there are many other parts of the code that I could work on, so it's no problem. Maybe take a look at jmri.Section for example.
> 
I really would appreciate new tests anywhere you want to add them.  With 200,000 untested lines of code, there is a lot of work to do here.

Eventually, I would like to practice test driven development, but my focus of late ( the last 2 years at least )has been on improving test coverage.

bobjacobsen, 2018-09-25:
@danielb987 Extending your very nice work on Conditionals to Logix might be a next step.

Although you ended up checking Conditionals for current behavior, you also uncovered some things that didn't make sense. That's valuable.  And those tests will now make sure that any changes are _intentional_, which is also valuable.  So there's more hope of being able to add things to that code without breaking it, or even (more importantly) eventually make that code more robust.

The place that both needs extension and is fragile is the LayoutEditor.  It works, it does things that people want, but it's so brittle that it's very hard to extend it.  (There have been issues around slips, duplicate turnouts and other things that I've repeatedly bounced off of because I can't see how to work on them without breaking lots of other stuff inadvertently, and it looks like changes won't converge) So if you want a big project, I suggest that area.

danielb987, 2018-09-25:
Thanks for your answers.

I have an idea on how to efficient test LayoutEditor from a user perspective, but that would require doing a lot of tests between setUp() and tearDown(). That could be solved by having a test class calling methods in other classes which do the actual testing.

```
class LayoutEditorTest {
    @Test
    testOne {
        TestClassOne.testOneThing();
        TestClassOne.testAnotherThing();
        TestClassOne.testThirdThing();
    }

    @Test
    testTwo {
        TestClassTwo.testOneThing();
        TestClassTwo.testAnotherThing();
        TestClassTwo.testThirdThing();
    }

    @Before
    setUp() { .... };

    @After
    tearDown() { ... };
}
```

In the example above, the classes TestClassOne and TestClassTwo do the actual testing. In the example above, I have excluded parameters and local variables, but a LayoutEditor instance would have been created in the setUp() method and then been passed to the methods.

@pabender told me about “Scaffold” in PR #5837, but adding “Scaffold” doesn't seem correct in this case. Do you have any other word that I could add to TestClassOne and TestClassTwo to mark them as part of the tests. Maybe something like “Worker”, like in TestClassOneWorker, to mark that TestClassOne do the work for the LayoutEditorTest class.

What I want is to do a huge number of tests between setUp() and tearDown(). These tests are related to each other and the whole point is that all these tests should use the same setup (same managers, turnouts, sections, ...)

pabender, 2018-09-25:
@danielb987

There are several things in your last message that we should think about.

First, we have to write tests so they are independent.   The primary reason for that is we don't control the order in which the tests execute, so you can't assume one test occurs before another.

Second, The marked @Before and @After methods are run before each test.  There are also @BeforeClass and @AfterClass methods that run before and after EVERY test in the class respectively.  The general rule of thumb is that only things that are costly to construct (such as database connections should be constructed in the @BeforeClass and torn down in the @AfterClass methods.   

Third, Since we're talking about testing a graphical user interface, we really should be writing the tests from the UI perspective.  Our current framework for doing this is a testing framework called Jemmy.  Jemmy works with an Operator concept.  You should take a look at my recent additions to the tests for throttles and consists which drive the interaction with the frame using task specific operators.

You also want to look at: 
https://github.com/JMRI/JMRI/blob/master/java/test/jmri/jmrit/display/EditorFrameOperator.java

That operator already has the ability to close EditorFrame derived objects.  This includes LayoutEditor objects.  

The operators are basically helper classes that hold GUI manipulation routines.

If you're going to start adding operators for LayoutEditor, start by extending the EditorFrameOperator class.  You can also add operators for more specific components if you like.  (I'm close to breaking the operator apart for the Throttle, so I have operators for each Internal frame object rather than a single operator).

danielb987, 2018-09-26:
I see.

When I wrote 'test LayoutEditor from a user perspective' I meant testing the UI. So on that part I fully agree. And my plan is to use Jemmy, but I have not studied it yet. Therefore my examples is not adopted to Jemmy yet.

I could make my idea work by having one single test doing all the work. This will be safe if the initialization also is in the same test.

```
@Test
public void testSomething() {
    LayoutEditor l = new LayoutEditor();
    TestClassOne.testOneThing(l);
    TestClassOne.testAnotherThing(l);
    TestClassOne.testThirdThing(l);

    TestClassTwo.testOneThing(l);
    TestClassTwo.testAnotherThing(l);
    TestClassTwo.testThirdThing(l);
}
```
.

> The general rule of thumb is that only things that are costly to construct (such as database connections should be constructed in the @BeforeClass and torn down in the @afterclass methods.

If each test is going to create and dispose the LayoutEditor window, I think that is expensive. I think a good testing of LayoutEditor will require maybe a couple of hundred tests and if all these show and dispose the LayoutEditor window, it will be a lot of windows coming up and closed.

I do agree with what you are saying, but my problem with LayoutEditor is that it has a huge amount of functionality, which makes it very difficult and labor intesive to fully test it.

My idea is basicly to have two complete sets of managers. Either by saving and restoring the managers themselves, or by having two InstanceManager instances. (I'm fully aware that InstanceManager is a singleton).

What I want to do is to have a complete JMRI setup which I manipulate by the UI. Like having the mouse drag a turnout in the LayoutEditor. And then have another identical complete JMRI setup that I manipulate manually by acting on the turnouts and other objects directly.

I first do a UI change on the first JMRI setup. I then do a manual change on the objects directly on the second JMRI setup. After that, I do a complete check that every object on both setups are equal.

This would catch things like a turnout gets duplicated, a change on one item changes something unrelated on another object, and so on. And most important, I think that once the staffold of this is in place, it would be quite easy to do a lot of testing on LayoutEditor.

A meta code example:

```
JUnitUtil.setInstanceManager(layoutEditorInstanceManager);
// Move turnout LT21 by dragging it
testUI_drag(10, 20, 50, 3);

JUnitUtil.setInstanceManager(manualInstanceManager);
turnoutLT21.x = 50;
turnoutLT21.y = 3;

Assert.assertTrue("Turnout has been moved to 50, 3",
        JUnitUtil.isSetupsEqual(layoutEditorInstanceManager, manualInstanceManager));
```

pabender, 2018-09-26:
Sent from my iPad

> On Sep 25, 2018, at 8:18 PM, Daniel <notifications@github.com> wrote:
> I do agree with what you are saying, but my problem with LayoutEditor is that it has a huge amount of functionality, which makes it very difficult and labor intesive to fully test it.
> 
Yes, it does, which is a big part of why it needs to be broken down into smaller units.

Perhaps what we need to do is think about what we really want to test here, and that goes to the types of tests we want to create.

We have two types of testing frameworks in place: JUnit tests and Cucumber tests.

JUnit tests really are about testing individual units.  I.e. do all the individual components do what they are supposed to do?

Cucumber tests focus on behavior.  ( given a set of starting conditions, when we take an action, then we get a specified result).  Behavior is provided by a group of components, rather than individual components.  

I really think the best approach to working with the Layout editor is to focus on the desired behavior rather than the individual unit functionality.

Let me see if I can mock up a framework for how a cucumber test would work here.

Paul

bobjacobsen, 2018-09-26:
> What I want is to do a huge number of tests between setUp() and tearDown(). These tests are related to each other and the whole point is that all these tests should use the same setup (same managers, turnouts, sections, …)

This sounds like a use for inheritance.  Create an AbstractLayoutEditorSpecialTestBase class with the setUp() and tearDown() methods, and any other common service methods needed.

Then create a bunch of LayoutEditorSpecialFooTest, LayoutEditorSpecialBarTest, LayoutEditorAndAnotherTest, etc all of which inherit from AbstractLayoutEditorSpecialTestBase.  By inheriting, each will run the common setUp and tearDown; you don’t have to separately inherit-and-reimplement those (unless you want to to add something).  

Note that if there’a a way to parameterize the variation you want across all those tests, that’s another option for getting lots of tests run.  

https://github.com/junit-team/junit4/wiki/parameterized-tests

For a local example, see java/test/jmri/jmrit/jython/SampleScriptTest.java

The idea is that you have a method that creates a bunch of parameters that are used to create a lot of variations on a single set of test code.

bobjacobsen, 2018-09-26:
> Perhaps what we need to do is think about what we really want to test here, and that goes to the types of tests we want to create.

Or perhaps what we want to eventually do, hence what the tests can help with.

For LayoutEditor, I would very much like to move toward an MVC separation.  The information about the underlying layout (the model of the model railroad) should be disentangled from teh Swing graphics and editor used to change it. The graphical display should be carefully separated from the editor.  Etc.

Once that’s done, we can start to tease apart the various aspects of that model, which are now really self-entangled.  It should be possible to work on how a slip works with it’s turnouts, without risking subtle bugs in graphics and signal logic assignments (yes, that’s a real example).

So I think it would be good to first test the layout logic, then test functional aspects like assignment of signals and connecting of blocks, _without_ going through the GUI.  If need be, add some package-level methods to let you test those below the graphical level.

danielb987, 2018-09-26:
I'm listening on what both of you are saying and I'm open to go that path.

But I don't think you have understood what I'm suggesting.

Think of it as two persons having two computers running JMRI. One person is using LayoutEditor and do all changes in the LayoutEditor. The other person is using a script to manually editing the objects on the panel, without using the LayoutEditor.

Now both persons do the exact same thing. They move the turnout LT20 from position (20, 3) to (50, 10). One person uses the LayoutEditor, the other person uses a script. But they do the exact same thing, which is moving the turnout.

If they now save their panels to a XML file and then compare the XML files, they should have the exact same result. One has used the LayoutEditor, the other has used a script. But the result should be the same.

This can be achieved within a single test if that test can work with two different InstanceManagers. First use one InstanceManager with managers, turnouts, etc, and use the UI of the LayoutManager to move the turnout.

Then switch to the other InstanceManager which has another set of managers, turnouts, etc, and change the turnout directly.

Then compare each object from both the InstanceManagers to see that they are the same.

A working example of what I mean is in PR #5881 . It's a very simple, non-UI, example,

I'm listening to both of you and I'm ready to follow you. But I still think my idea would do exactly what you want to achieve.

danielb987, 2018-09-26:
What I'm saying is that the _user_ has a clear picture of what he wants to do. He wants to move the turnout. My way of testing LayoutManager means that the test acts like an user by using the UI. But it also uses "another computer" with another JMRI instance to do the exact same thing, moving the turnout, by acting on the turnout directly, without using LayoutManager.

Now, if the LayoutManager does the right thing, these two operations should give the exact same result. If not, there is an error.

danielb987, 2018-09-26:
> The place that both needs extension and is fragile is the LayoutEditor. It works, it does things that people want, but it's so brittle that it's very hard to extend it. (There have been issues [...]

How do you test complex changes? How do you test that a change on turnout LT12 does NOT do a unwanted change on section 20 ? How do you catch EVERY unexpected change somethere else?

If LayoutEditor is working correctly, every operation must be predicable. And if it is, that means that you can change the turnouts, sections, lights, and so on, the exact same way as LayoutEditor would change them, but doing it on the items directly.

And by using "two differect computers", you can check that nothing unexpected would happen. For example, my way of testing LayoutEditor will catch if a turnout gets duplicated, if a change on a section causes an unexpected and unwanted change on a light, and so on.

pabender, 2018-09-26:
@danielb987 

I think @bobjacobsen and I both understand what you are saying you want to do.

The problem with the "two users using different methods" approach is that it only tells you if multiple methods of performing the actions are consistent.  The issue is that consistency does not imply correctness.

pabender, 2018-09-26:
> How do you test complex changes? How do you test that a change on turnout LT12 does NOT do a unwanted change on section 20 ? How do you catch EVERY unexpected change somethere else?
> 

Testing existing software is not about making sure that every unintended consequence is captured.  It is, to a large extent, about documenting existing behavior and making sure that (assuming it is the right behavior) you capture what that behavior is.

Test Driven or Behavior Driven Development works by collecting these tests that capture behaviors.  When you want to make a change (either because you want to add a new behavior or because a bug was found) you write a test that fails, and then make the smallest changes possible to the code so that the new tests, along with all the existing tests, pass.

> If LayoutEditor is working correctly, every operation must be predicable. And if it is, that means that you can change the turnouts, sections, lights, and so on, the exact same way as LayoutEditor would change them, but doing it on the items directly.
> 

And that is where the separation of MVC comes into play.   The objects are the model.  The GUI elements are the view, and the interaction between the two is where the controller comes into play.

> And by using "two differect computers", you can check that nothing unexpected would happen. For example, my way of testing LayoutEditor will catch if a turnout gets duplicated, if a change on a section causes an unexpected and unwanted change on a light, and so on.

but it won't catch the case when the turnout is duplicated because the underlying code duplicates it and not the GUI.

danielb987, 2018-09-26:
Yes, I agree. I understand EXACTLY what you and Bob Jacobsen want to achieve.

My tests will:
* document the current behavior
* ensure that changes done to LayoutEditor is not unintentional
* test that the code does that it is expected to do, test the correctness of the code
* work without changing the LayoutEditor itself

> but it won't catch the case when the turnout is duplicated because the underlying code duplicates it and not the GUI.

Yes, it will. Because I will test everything in the whole JMRI setup against everything in the other JMRI setup. My PR was a very simple example, but it's very easy to test that every object that is on one JMRI instance is also on the oher JMRI instace, and exactly the same.

MVC is probably a great thing to add. But if I understand you and Bob Jacobsen correctly, you want to do that change to LayoutEditor. Now, without tests, that seems to be a scary thing to do. If we add tests first, that test LayoutEdito from a user perspective, LayoutEditor could be changed to MVC when the tests are in place.

I suggest that you sleep on this and read my previous message again tomorrow.

danielb987, 2018-09-26:
> Because I will test everything in the whole JMRI setup against everything in the other JMRI setup.

I mean test for equality. That each object in one JMRI setup matches with exactly one object in the other JMRI setup.

pabender, 2018-09-26:
> Yes, it will. Because I will test everything in the whole JMRI setup against everything in the other JMRI setup. My PR was a very simple example, but it's very easy to test that every object that is [...]

This is true, but that also isn't what I'm saying here.

The your GUI instance will manipulate the objects by asking the GUI to do something, but the GUI then calls the underlying code on the objects.

Your non-GUI instance will manipulate the objects by directly calling the underlying code on the objects.

This might catch issues caused by the GUI components.  It will not catch issues caused by the underlying objects.

pabender, 2018-09-26:
> MVC is probably a great thing to add. But if I understand you and Bob Jacobsen correctly, you want to do that change to LayoutEditor. Now, without tests, that seems to be a scary thing to do. If we [...]

You mis-read what @bobjacobsen and I are saying.

This is a change we SHOULD make.  This is not a change we can make right now, because the code is too brittle and lacks tests.

danielb987, 2018-09-26:
> You mis-read what @bobjacobsen and I are saying.
> 
> This is a change we SHOULD make. This is not a change we can make right now, because the code is too brittle and lacks tests.

I apologize.

danielb987, 2018-09-26:
> The your GUI instance will manipulate the objects by asking the GUI to do something, but the GUI then calls the underlying code on the objects.
>
> Your non-GUI instance will manipulate the objects by [...]

My idea is to use as primitive methods as possible to do the manual manipulations. And yes, if these methods do the wrong thing, it will be an error on both setups. And in those cases there the only way to do the manipulation is to use higher level methods, there is a even a greater risk of errors.

But still, testing consistency would be a huge step forward compared to the current situation.

bobjacobsen, 2018-09-26:
@danielb987 I think you're going in an interesting direction with this.  I suspect that there will always be judgement calls about how much state to compare (everything&always is not really an option, because the test code has to navigate to do the comparison), but there are also judgement calls in our usual unit tests:  By testing what a unit is expected to do, we judge that it won't also do weird extra things.

If you want to check that nothing extra is being created, a starting point would be to check for extra Managers in the InstanceManager, and for extra items in the Managers you expect to find there.  I.e. if you created only a SensorManager and TurnoutManager for a test, you can check before and after about that.  (The InstanceManager will create some things automatically when you ask for them, which might make the counting a bit tricky at first) You can check for the number of NamedBeans in each manager, and record their before/after state.

LayoutManager _should_ only be using items from the `'jmri`, `jmri.util.*` and jmri.jmrit.*` packages (and ideally, only a few from `jmri.jmrit` outside of `jmri.jmrit.display` like `jmri.jmrit.roster`)  See http://jmri.org/help/en/html/doc/Technical/IntroStructure.shtml

There seem to be only a couple of references to `jmri.implementation` and `jmri.managers`:
```
% grep -r 'jmri\.im' java/src/jmri/jmrit/display/layoutEditor/
java/src/jmri/jmrit/display/layoutEditor//LayoutBlockConnectivityTools.java:     * <li>Called by {@link jmri.implementation.DefaultSignalMastLogic} using MASTTOMAST.</li>
java/src/jmri/jmrit/display/layoutEditor//LayoutEditorTools.java:import jmri.implementation.DefaultConditionalAction;
java/src/jmri/jmrit/display/layoutEditor//LayoutBlock.java:import jmri.implementation.AbstractNamedBean;
% grep -r 'jmri\.m' java/src/jmri/jmrit/display/layoutEditor/
java/src/jmri/jmrit/display/layoutEditor//configurexml/LayoutBlockManagerXml.java:public class LayoutBlockManagerXml extends jmri.managers.configurexml.AbstractNamedBeanManagerConfigXML {
java/src/jmri/jmrit/display/layoutEditor//LayoutBlockManager.java:import jmri.managers.AbstractManager;
```

The `LayoutBlock` class (like `OBlock`) is probably at the wrong level of the hierarchy; it might be good to just refactor it away before doing much else (_where_ a class is should be easy to change without breaking unexpected things,  although sometimes they have nasty package-level connections)

The `jmri.jmrit` aspects are more problematic:
```
% grep -r 'jmri\.jmrit' java/src/jmri/jmrit/display/layoutEditor/ | grep -v 'jmri\.jmrit\.display'
java/src/jmri/jmrit/display/layoutEditor//LevelXing.java:import jmri.jmrit.signalling.SignallingGuiTools;
java/src/jmri/jmrit/display/layoutEditor//LayoutBlockConnectivityTools.java:     * Called by {@link jmri.jmrit.signalling.SignallingPanel} using MASTTOMAST.
java/src/jmri/jmrit/display/layoutEditor//LayoutBlockConnectivityTools.java:     * <li>Called by {@link jmri.jmrit.entryexit.DestinationPoints} using SENSORTOSENSOR.</li>
java/src/jmri/jmrit/display/layoutEditor//LayoutBlockConnectivityTools.java:     * <li>Called by {@link jmri.jmrit.entryexit.EntryExitPairs} using SENSORTOSENSOR.</li>
java/src/jmri/jmrit/display/layoutEditor//BlockContentsIcon.java:import jmri.jmrit.roster.RosterEntry;
java/src/jmri/jmrit/display/layoutEditor//LayoutEditorTools.java:import jmri.jmrit.blockboss.BlockBossLogic;
java/src/jmri/jmrit/display/layoutEditor//LayoutEditorTools.java:import jmri.jmrit.catalog.NamedIcon;
java/src/jmri/jmrit/display/layoutEditor//LayoutEditorTools.java:import jmri.jmrit.signalling.SignallingGuiTools;
java/src/jmri/jmrit/display/layoutEditor//LayoutEditorTools.java:        if (!InstanceManager.getDefault(jmri.jmrit.entryexit.EntryExitPairs.class).deleteNxPair(sensor)) {
java/src/jmri/jmrit/display/layoutEditor//LayoutBlock.java:import jmri.jmrit.beantable.beanedit.BeanEditItem;
java/src/jmri/jmrit/display/layoutEditor//LayoutBlock.java:import jmri.jmrit.beantable.beanedit.BeanItemPanel;
java/src/jmri/jmrit/display/layoutEditor//LayoutBlock.java:import jmri.jmrit.beantable.beanedit.BlockEditAction;
java/src/jmri/jmrit/display/layoutEditor//LayoutBlock.java:import jmri.jmrit.roster.RosterEntry;
java/src/jmri/jmrit/display/layoutEditor//configurexml/LayoutEditorXml.java:import jmri.jmrit.dispatcher.DispatcherFrame;
java/src/jmri/jmrit/display/layoutEditor//PositionablePoint.java:import jmri.jmrit.signalling.SignallingGuiTools;
java/src/jmri/jmrit/display/layoutEditor//PositionablePoint.java:                    jmri.jmrit.blockboss.BlockBossLogic.getStoppedObject(removeremote);
java/src/jmri/jmrit/display/layoutEditor//PositionablePoint.java:                    jmri.jmrit.blockboss.BlockBossLogic.getStoppedObject(removelocal);
java/src/jmri/jmrit/display/layoutEditor//ConnectivityUtil.java:import jmri.jmrit.blockboss.BlockBossLogic;
java/src/jmri/jmrit/display/layoutEditor//MemoryIcon.java:import jmri.jmrit.catalog.NamedIcon;
java/src/jmri/jmrit/display/layoutEditor//MemoryIcon.java:import jmri.jmrit.roster.RosterEntry;
java/src/jmri/jmrit/display/layoutEditor//MemoryIcon.java:                if (val instanceof jmri.jmrit.roster.RosterEntry) {
java/src/jmri/jmrit/display/layoutEditor//MemoryIcon.java:                    jmri.jmrit.roster.RosterEntry roster = (jmri.jmrit.roster.RosterEntry) val;
java/src/jmri/jmrit/display/layoutEditor//LayoutEditor.java:import jmri.jmrit.catalog.NamedIcon;
[... cut ...]
java/src/jmri/jmrit/display/layoutEditor//BlockValueFile.java:import jmri.jmrit.roster.RosterEntry;
java/src/jmri/jmrit/display/layoutEditor//BlockValueFile.java:                                        if (blockList.get(i).getAttribute("valueClass").getValue().equals("jmri.jmrit.roster.RosterEntry")) {
java/src/jmri/jmrit/display/layoutEditor//BlockValueFile.java:                            val.setAttribute("valueClass", "jmri.jmrit.roster.RosterEntry");
java/src/jmri/jmrit/display/layoutEditor//LayoutBlockManager.java:import jmri.jmrit.roster.RosterEntry;
```

pabender, 2018-09-26:
> Which packages needs to be tested in order to make my test true correctness tests? Which non-GUI packages are used by LayoutManager and would be used by me manually in these tests?
> 

Bob has already largely answered the "which packages" question.  

Before I answer the correctness question, one thing to remember is that we really are looking at a portion of the code that builds and operates panels,  so there are two things we have to look at:

1. How are the panels built?
2. Do the panels manipulate the expected objects once built?

I think the second question is easier to tackle.  We can actually test that by loading a panel file and then making sure the values we expect to change while operating that panel change as expected.
This is where we should see the largest number of values change in non-GUI objects.

For the first question, almost everything should be manipulated within the jmri.jmrit.display package, but it isn't just testing to see that objects are correctly manipulated.  

To address your question about correctness, takes a much longer explanation (I teach a class in testing every year, and a large portion of the class is about how do we show correct behavior, especially of units).

The short version is that we have to answer the question of does the code do what we expect it too.  For that, you have to think about input ranges and output ranges and verify for a given set of inputs the expected outputs are produced.  Most of our existing tests fall into the category of "Special Value" tests in that we test a very small handful of values and see what happens.  We also should be concerned with boundary values (what happens when one or more inputs is out of range) and how they are handled, but we don't do that in a most places currently.

Where correctness is such a pain in the jmri.jmrit.display package is that we need to be able to see that a particular object has the visually expected results, which is hard to test programmatically, and that the objects on the panels have the expected internal state.

I will also add that there are other theories about testing and how it should be done.  There is a theory of path testing and a theory of complexity testing.  These exist because simply increasing the covered line count doesn't always help.  This is where the "branch" and "complexity" metrics on the Jacoco reports come into play.   Ultimately, these are looking at the program as a graph of statements.

The branch number is a measure of paths in that it is telling us whether or not we take every path from every control statement through the program.  In other words, every time we get to an if statement, do we test with the condition being both true and false.

Complexity is also looking at paths, but it is really looking at how well we test paths between two points in the program.  (I would have to look at how Jacoco does this, but the theoretical approach looks at how well we test the cyclomatic complexity of the program graph).

A part of both path testing and complexity testing is making sure that if we have an logical expression we test every possible way to make that true and every possible way to make it false.

bobjacobsen, 2018-09-26:
@danielb987  Bottom line, if you want to add some new methods of testing, please go ahead and do that. Any additional testing in this area will be useful, and I'll be interested in what you learn from this approach.

"Code wins.:  And even more than that, if you're interesting in working out a new approach, that's a good thing.

danielb987, 2018-09-26:
@bobjacobsen : Thanks.

Code wins, but there is good code and bad code. And I don't want to add bad code, which is why I asks first.

> If you want to check that nothing extra is being created, a starting point would be to check for extra Managers in the InstanceManager, and for extra items in the Managers you expect to find there. [...]

I have updated my PR #5881 with a method that compares all the beans in two different InstanceManagers. See `JUnitUtil.verifyInstanceManagerBeansAreEqual()` and `JUnitUtilTest.testVerifyInstanceManagerBeansAreEqual()`.

If I add a turnout to one instance manager but not the other, or if I have the same sensor in both but with different states, the method `verifyInstanceManagerBeansAreEqual` returns false and print the missing beans to the error log.

danielb987, 2018-09-26:
@pabender : Thanks for the explanation.

closed 2021-04-14 by stale[bot]