JMRI/JMRI #5880: Signal Mast for NCE DCC

Ted-IC, 2018-09-26:
Has anyone looked into the following issue?   I noticed when adding a signal Head using a DCC Signal Decoder, the box labeled "Offset Address" must be checked for an NCE DCC system.  However, when adding a signal Mast using a DCC Signal Mast Decoder, this box is not available for an NCE DCC system.    

Should a box be available for "Offset Address" when using a DCC Signal Mast Decoder with an NCE DCC system or is this offset address taken care of internally in JMRI? 

Ted in Iowa City

Ted-IC, 2018-09-29:
Hi,
On my workbench, I have a breadboard with Signal A made from green, yellow, and red LEDs.  The LEDs are connected to an NCE Light-It with an address of 136.  Signal B has the same arrangement on the breadboard but has an address of 137.  I can change the aspects of Signals A and B using an NCE ProCab.

I opened a panel for my layout in Panel Pro (JMRI V 4.12).  I selected Tables/Signal Masts and created Signal A using the following options:
	User Name:	MP137 136RA
	Signal Systems:	Basic Model Signals
	Mast Type:	Single searchlight high signal
	Mast Driver:	DCC Signal Mast Decoder
	System:	NCE
	DCC Accessory Address:	136

I created Signal B using the Signal Heads Table and checking the box “Offset address” and using the following options in the Signal Masts Table:
	User Name:	MP137 137RA
	Signal Systems:	Basic Model Signals
	Mast Type:	Single searchlight high signal
	Mast Driver:	Single Head Controlled Mast

The following are the results from setting the aspect of a signal from the Signal Masts Table:

*	From the NCE Command Monitor Window
Signal A:  MP137 136RA
	Reply: 1C 80 00 3A 00 00 84 00 00 00 00 00 10 00 01 04
	Signal 132 Aspect 1
	command completed successfully

	Reply: 3C 2A 0A D0 BC 3E 00 00 00 00 00 00 00 00 00 00
	Signal 132 Aspect 2
	command completed successfully

Even though the response is “command completed successfully”, the LEDs for Signal A did not change.

Signal B:  MP137 137RB
	Reply: 3C 2A 0A D0 BC 3E 00 00 00 00 00 00 00 00 00 00
	Signal 137 Aspect 2
	command completed successfully

The LEDs for Signal B changed as desired.

*	Note that for signal A, the address is 132 (not 136 as it should be) whereas for signal B, the address is 137 (correct).

*	I reconfigured Signal A by adding 4 to address 136 to get a DCC Accessory Address of 140.
	Reply: 3C 2A 0A D0 BC 3E 00 00 00 00 00 00 00 00 00 00
	Signal 136 Aspect 2
	command completed successfully

	This time the LEDs for Signal A changed as desired.

Does this make sense?  It seems that “4” should be added to the address when using the Signal Masts Table to configure a signal for an NCE system.

Ted in Iowa City

dheap, 2018-09-29:
@Ted-IC 
As you have discovered, the Signal Head Table allows you to specify an Offset Address when defining a DCC Signal Decoder, but the Signal Mast Table does not allow for the same when defining a DCC Signal Mast Decoder. This is a code deficiency that needs fixing.

The "Offset Address" checkbox actually selects between two different JMRI code methods that use different algorithms for creating the Extended Accessory (Signal) packets.

The "confusion" arises because of a perceived lack of clarity in NMRA S9.2.1 regarding exactly how the 11 address bits in a Signal decoder packet are to be constructed.
- Some manufacturers interpret these bits in the same manner as they are in a Basic Accessory (Turnout) packet. With a Turnout decoder packet; nine of the bits are interpreted as a decoder (board) address with valid values being 1-511 inclusive and the other two bits are interpreted as a channel address with valid values being 0-3 inclusive. So  the bit values for the first valid turnout/signal address (1) are 00000000100.
- Some manufacturers interpret these bits as a simple eleven bit address. So  the bit values for the first valid signal address (1) are 00000000000.

(Just to complicate matters, these bits are stored split up into groups in the packet, not in obvious order and some bits are inverted! But the user need to not worry about that.)

From the JMRI point of view, we just have to provide for the user be able to choose the method  that suits the decoder brand each time a DCC Signal address is created.

Ted-IC, 2018-10-01:
Dave,

Thanks for the comments.

It is interesting that the bits 100 correspond to 4, which is the offset DCC address I found to get the signal mast to work.  Will this be incorporated in a future version of JMRI?

BTW, at the Soo Line Historical and Technical Society convention banquet on Saturday, September  22 held in Stevens Point, Wisconsin, my wife and I were at a table with a couple from Australia.  We start talking about JMRI and I mentioned your name as a good reference.  He got on his smart phone and found that they live about 1000 miles from you.  I told him he should keep your name handy in case he needs any help with JMRI.

Ted in Iowa City

KenC57, 2021-03-30:
Current workaround is to use the wrong address that adjusts for the 4 offset.

dheap, 2021-05-15:
Still a needed feature.

Ted-IC, 2021-05-15:
Dave, 

I do not know what this message is for or who it is meant for. 

Yes, the "Offset Address" in JMRI for NCE LIght-It decoders is still needed. As I recall (I have not done this in the past year), the checking of "Offset Address" for signal heads works but there is no box for signal masts. It would be easier to create a signal mast. 

I have over 100 Light-Its on my layout. 

Ted Smith of Iowa City.

Ted-IC, 2021-06-29:
JMRI Developers,

I still feel that an "Offset Address" is still needed when configuring a signal mast,  There is already an Offset Address when configuring a signal head so why not for a signal mast?

Ted-IC

dheap, 2021-08-14:
Needs doing, but lacking a volunteer.

dheap, 2021-09-28:
Still needed.

dsand47, 2021-09-29:
A preliminary investigation indicates that it will not be hard to clone the signal head offset logic for use by signal masts.

Someone else will need to test the changes since I have no hardware.

There is a possible confusion factor.  A Light-It is a `signal head` device.  Adding the offset to signal masts, may lead some people to think that a Light-It can handle a signal mast regardless of the number of heads on the mast.  The current limitation forces them to use heads for the Light-It and one or more heads for the mast.  With the offset option on the mast, they can create single head masts that communicate directly to the Light-It, but still need to use head based masts for masts with multiple heads.

dheap, 2021-09-30:
@dsand47 A Light-It is a DCC Signal Decoder with three LEDs and able to display multiple aspects which may use one or more of the LEDs, possibly with some flashing. Is that effectively a signal mast? Or is it just a 3 lamp signal head?

dsand47, 2021-09-30:
> @dsand47 A Light-It is a DCC Signal Decoder with three LEDs and able to display multiple aspects which may use one or more of the LEDs, possibly with some flashing. Is that effectively a signal mast? Or is it just a 3 lamp signal head?

Yes.

What the Light-It documentation calls an "aspect" is an "appearance" in JMRI terms, which is the color(s) that a signal head can display.  These are red, yellow, green plus flashing for each.

The "aspect" displayed by a signal mast is based on the "appearance" of the signal head or heads on a mast as defined by the rules of the signaling system, such as AAR-1946, BR-2003, etc.

A Light-It can be used for one signal head or a single head mast (after the offset fix).  A multiple head mast will require multiple signal heads, each addressed for a separate Light-It.

dheap, 2021-09-30:
> @dsand47 A preliminary investigation indicates that it will not be hard to clone the signal head offset logic for use by signal masts.
> 
> Someone else will need to test the changes since I have no hardware.

I have hardware, both Light-Its and a DCC Packet Analyser to check the actual DCC packers generated by JMRI.

dheap, 2021-09-30:
From the NCE Light-It Manual:
Using the Light-It as an NMRA conforming signal decoder
The Light-It will accept NMRA signal addresses and display up to 18 different lighting effects 
on one 3 LED signal head. The decoder responds to aspects 0-15 plus 30 and 31 which al-
lows all 18 effects to be available for your signal heads. Only Aspect 0 (absolute stop) is de-
fined by the NMRA standards. Table 1 below describes each lighting effect and its mapping to
an aspect number. It is possible to change the aspect mapping (see Description of configura-
tion variables) for aspects 0-15. Aspects 30 and 31 are 'hard coded' into the decoder as they 
are not the kind of lighting effect normally displayed on a real signal. 

**Table 1:**

| Aspect | Effect Number | Lighting effect | 
| --- | --- | --- | 
| 0 | 0 | Red | 
| 1 | 1 | Yellow
| 2 | 2 | Green | 
| 3 | 3 | flash red | 
| 4 | 4 | flash yellow | 
| 5 | 5 | flash green | 
| 6 | 6 | red+yellow
| 7 | 7 | flash red+yellow | 
| 8 | 8 | red+flash yellow | 
| 9 | 9 | red+green | 
| 10 | 10 | flash red+green | 
| 11 | 11 | red+flash green | 
| 12 | 12 | yellow+green | 
| 13 | 13 | flash yellow+green | 
| 14 | 14 | yellow+flash green | 
| 15 | 15 | effect 15=all on | 
| 30 | -- | all flash | 
| 31 | -- | all off | 

Since we cannot control which LEDs (and therefore the brightness) manufacturers use in their 
signals we provide a CV to control the brightness of each LED. The brightness ranges from 0 
(off) to 255 (max) with 255 being the factory default for brightness. For the CV numbers see 
“Description of configuration variables” later in this manual.

dsand47, 2021-09-30:
@dheap 

PR #10149 has been created.  For testing, **Basic Model Signals** is a simple signaling system.  Use either of the of Single Searchlight mast types.

dheap, 2021-09-30:
@dsand47 Thanks. Will  test review when weather (thunderstorms) permit.

closed 2021-10-02 by danielb987