JMRI/JMRI #5894: Some Programmers intolerant of null ProgListener

bobjacobsen, 2018-09-28:
From a jmriusers post:
> I tried out this script for POM writing a CV value to a mobile decoder using my Roco Z21 command station (CV 3=30 and decoder address 12 in the example).
>
>import jmri
>
>programmer = jmri.InstanceManager.getDefault(jmri.AddressedProgrammerManager).getAddressedProgrammer(False, 12)
>programmer.writeCV("3", 30, None)
>
>This is OpsProgExample.py in the Jython directory, keeping only its relevant part.
>
>PanelPro/Edit/Preferences/Defaults offers for Ops Programming the following default connections: Z21, Z21XpressNet, Z21LocoNet and Internal. Selecting either of the first two, the script works but, surprisingly, session.log reports an error. There is no error message for the third choice but the script does not do anything. The session.log file containing the error is uploaded to Files/ProblemsBeingWorkedOn under the title POM Write Error.

The error message was:
```
2018-09-27 18:19:11,198 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
       at jmri.jmrix.roco.z21.Z21XNetOpsModeProgrammer.message(Z21XNetOpsModeProgrammer.java:203)
       at jmri.jmrix.loconet.LnTrafficController.notify(LnTrafficController.java:114)
       at jmri.jmrix.loconet.LnPacketizer$RcvMemo.run(LnPacketizer.java:365)
       at java.awt.event.InvocationEvent.dispatch(Unknown Source)
```

This is due to the 3rd `ProgListener` argument being `None` in the `writeCV` call, but that's always been allowed.

The NPE was from an unprotected reference in `jmri.jmrix.roco.z21.Z21XNetOpsModeProgrammer`:
```
            progListener.programmingOpReply(val, code);
```

Only a few systems do exactly that:
```
java/src//jmri/jmrix/roco/z21/Z21XNetOpsModeProgrammer.java:                progListener.programmingOpReply(value, jmri.ProgListener.OK);
java/src//jmri/jmrix/roco/z21/Z21XNetOpsModeProgrammer.java:                progListener.programmingOpReply(value, jmri.ProgListener.OK);
java/src//jmri/jmrix/roco/z21/Z21XNetOpsModeProgrammer.java:                    progListener.programmingOpReply(value, jmri.ProgListener.NoLocoDetected);
java/src//jmri/jmrix/roco/z21/Z21XNetOpsModeProgrammer.java:                    progListener.programmingOpReply(value, jmri.ProgListener.NotImplemented);
java/src//jmri/jmrix/roco/z21/Z21XNetOpsModeProgrammer.java:                    progListener.programmingOpReply(value, jmri.ProgListener.UnknownError);
java/src//jmri/jmrix/roco/z21/Z21XNetOpsModeProgrammer.java:            progListener.programmingOpReply(val, code);
java/src//jmri/jmrix/lenz/XNetOpsModeProgrammer.java:                progListener.programmingOpReply(value, jmri.ProgListener.OK);
java/src//jmri/jmrix/lenz/XNetOpsModeProgrammer.java:                    progListener.programmingOpReply(value, jmri.ProgListener.NotImplemented);
java/src//jmri/jmrix/lenz/XNetOpsModeProgrammer.java:                    progListener.programmingOpReply(value, jmri.ProgListener.UnknownError);
java/src//jmri/jmrix/lenz/XNetOpsModeProgrammer.java:        progListener.programmingOpReply(value, jmri.ProgListener.FailedTimeout);
java/src//jmri/jmrix/dccpp/DCCppOpsModeProgrammer.java:        progListener.programmingOpReply(value, jmri.ProgListener.OK);
java/src//jmri/jmrix/dccpp/DCCppOpsModeProgrammer.java: progListener.programmingOpReply(value, jmri.ProgListener.NotImplemented);     
java/src//jmri/jmrix/dccpp/DCCppOpsModeProgrammer.java:                progListener.programmingOpReply(value, jmri.ProgListener.OK);
java/src//jmri/jmrix/dccpp/DCCppOpsModeProgrammer.java:        progListener.programmingOpReply(value, jmri.ProgListener.FailedTimeout);
java/src//jmri/jmrix/dcc4pc/Dcc4PcOpsModeProgrammer.java:        progListener.programmingOpReply(cv, jmri.ProgListener.OK);
java/src//jmri/jmrix/dcc4pc/Dcc4PcOpsModeProgrammer.java:             progListener.programmingOpReply(0, status);
java/src//jmri/jmrix/dcc4pc/Dcc4PcOpsModeProgrammer.java:        progListener.programmingOpReply(0, jmri.ProgListener.FailedTimeout);
java/src//jmri/jmrix/dcc4pc/Dcc4PcOpsModeProgrammer.java:                    progListener.programmingOpReply(newValue, ProgListener.OK);
```

Others seem to protect against the NPE by checking before sending.

bobjacobsen, 2018-09-28:
There are other programmers that exhibit the same failure, i.e. NceProgrammer.  So adding a test for operation with null to the base tests would be good.

pabender, 2018-09-28:
> There are other programmers that exhibit the same failure, i.e. NceProgrammer. So adding a test for operation with null to the base tests would be good.

I am actually trying to figure out how we can do this, and it appears that a lot of potentially breaking changes are going to be required.

While most of the concrete programmers do notify the listeners, they don’t all do this in the same ways under similar same condition and there isn’t currently a requirement that any of the programmers implement a method to handle the notification.

I propose the following solution:

1. Add to the programmer interface a method to notify listeners.   My thought is to use the method name 
2. Add concrete implementations of these to the abstractProgrammer and AbstractOpsModeProgrammer that call the appropriate reply method.
3. Refactor the programmers to call the new method instead of directly calling the reply methods in the listeners.

pabender, 2018-09-29:
a test has been added as part of #5896 as a means to try to flag where this is occurring.  It may not be entirely successful, and I haven't run it against all the programmers, but it did catch a few I knew were going to cause an issue.

In a bit of test driven development, that PR will be merged when the new test passes.

PR #5896 (2018-10-02): programmer test update minor XBee update
The main contribution of this pr is a test showing how much work there is to fix #5894.  As of the initial commit, this test is known to not pass.
closed 2018-10-02 by pabender