JMRI/JMRI #5925: Followup:  JMRI 4.13.03 - Signal Mast Logic - Destination Mast Missing - Parsing Error

dsand47, 2018-10-03:
When a signal mast is renamed and it is a destination mast in signal mast logic, errors occur which prevent the destination mast being included in the panel xml file.  This results in the the parsing error.

The main component involved is **destList** in jmri.implementation.DefaultSignalMastLogic:
```Hashtable<SignalMast, DestinationMast> destList = new Hashtable<SignalMast, DestinationMast>();```

Most of the access in the class and xml class use a key list to iterate through the list.  After a user name change, the affected key returns null instead of the related value.

The following output uses a forEach to get the keys and values and then verifies access by key.
```
INFO  - >> name = dmast
INFO  -    key = IF$vsm:AAR-1946:PL-2-high($0002), value = jmri.implementation.DefaultSignalMastLogic$DestinationMast@40bc3b67, key hash = -1650181794
INFO  -    get(k): jmri.implementation.DefaultSignalMastLogic$DestinationMast@40bc3b67

INFO  - >> name = xx
INFO  -    key = IF$vsm:AAR-1946:PL-2-high($0002), value = jmri.implementation.DefaultSignalMastLogic$DestinationMast@40bc3b67, key hash = -1745874171
INFO  -    get(k): null
INFO  -    equals: true, ==: true
```
I included a before and after object comparison using both methods to see if that has an impact.  The only difference I see is that the hashCode() changed which is partially based on the user name.

The solution I see is to copy destList to a temporary list, clear destList and copy back.  

Are there better solutions?

bobjacobsen, 2018-10-03:
I don't know enough about the details of how SignalMastLogic is intended to be set up.

@dsand47 Could you take the existing test in java/test/jmri/SignalMastLogicTest.java and make a new one that shows the problem by renaming a mast at the appropriate point?  I thought it would be as simple as doing that at the end, but I didn't get a failure.  See testRename in my bobjacobsen/bean-equals branch.

bobjacobsen, 2018-10-03:
Not sure if it fixed the original issue here, but that equals fix passed CI _except_ 
```
LnTurnoutTest.testLnTurnoutExactFeedback:155 KnownState after AUX report THROWN is INCONSISTENT expected:<8> but was:<1>
```

Working on this.

bobjacobsen, 2018-10-03:
Does #5926 fix this?

dsand47, 2018-10-03:
#5926 does fix the user name change issue for DefaultSignalMastLogic.

I did have to rework testRename.  It was doing the setup but never actually running the logic which means it never created the failure scenario.

**testRename** probably needs to be changed to **testExecution** (or something meaningful) since it activates the logic, changes signal mast aspects to trigger SML events and includes the user name change.

The diff and updated java file are in the attached zip file.

[sml test.zip](https://github.com/JMRI/JMRI/files/2442852/sml.test.zip)

dsand47, 2021-03-17:
Problem no longer occurs.

closed 2021-03-17 by dsand47