JMRI/JMRI #5930: Headless (faceless?) not working

positron96, 2018-10-04:
Hello. I'm trying to start JMRI on headless beaglebone black but have problems with it.

1. I've installed `default-jre-headless` from debian repos which provides java 8.
2. I unpacked fresh tgz archive (v4.12/JMRI.4.12+Rb6a9bb1) and tried to run ./JMRIFaceless.
3. all sorts of errors relating to AWT and Swing. I'm trying to fix them one by one as they appear, but there are so many that I started to doubt that it will work.

What I've encountered so far:
 * jmri.util.Log4JUtil calls SystemConsole.create() which is a JTextArea. I've added GraphicsEnvironment.isHeadless() check.
 * jmri.beans.Bean.firePropertyChange uses jmri.util.ThreadingUtil.runOnGUIEventually which calls SwingUtilities.invokeLater. If headless, I make a direct call instead of invokeLater. This is probably a huge change but I did not get to a point where it fails. Other methods jmri.util.ThreadingUtil patched similarly.
 *  jmri.implementation.JmriConfigurationManager tries to instantiate apps.gui.GuiLafPreferencesManage which certainly fails. Here I don't know how to remove it from loading chain.

The last exception text is as below:
```
java.util.ServiceConfigurationError: jmri.spi.PreferencesManager: Provider apps.gui.GuiLafPreferencesManager could not be instantiated
	at java.util.ServiceLoader.fail(ServiceLoader.java:232)
	at java.util.ServiceLoader.access$100(ServiceLoader.java:185)
	at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:384)
	at java.util.ServiceLoader$LazyIterator.next(ServiceLoader.java:404)
	at java.util.ServiceLoader$1.next(ServiceLoader.java:480)
	at jmri.implementation.JmriConfigurationManager.<init>(JmriConfigurationManager.java:44)
	at apps.AppsBase.installConfigurationManager(AppsBase.java:201)
	at apps.AppsBase.<init>(AppsBase.java:97)
	at apps.JmriFaceless.<init>(JmriFaceless.java:30)
	at apps.JmriFaceless.main(JmriFaceless.java:44)
Caused by: java.awt.AWTError: Assistive Technology not found: org.GNOME.Accessibility.AtkWrapper
	at java.awt.Toolkit.loadAssistiveTechnologies(Toolkit.java:807)
	at java.awt.Toolkit.getDefaultToolkit(Toolkit.java:886)
	at sun.swing.SwingUtilities2.getSystemMnemonicKeyMask(SwingUtilities2.java:2020)
	at javax.swing.plaf.basic.BasicLookAndFeel.initComponentDefaults(BasicLookAndFeel.java:1158)
	at javax.swing.plaf.metal.MetalLookAndFeel.initComponentDefaults(MetalLookAndFeel.java:431)
	at javax.swing.plaf.basic.BasicLookAndFeel.getDefaults(BasicLookAndFeel.java:148)
	at javax.swing.plaf.metal.MetalLookAndFeel.getDefaults(MetalLookAndFeel.java:1577)
	at javax.swing.UIManager.setLookAndFeel(UIManager.java:539)
	at javax.swing.UIManager.setLookAndFeel(UIManager.java:579)
	at javax.swing.UIManager.initializeDefaultLAF(UIManager.java:1349)
	at javax.swing.UIManager.initialize(UIManager.java:1459)
	at javax.swing.UIManager.maybeInitialize(UIManager.java:1426)
	at javax.swing.UIManager.getLookAndFeel(UIManager.java:492)
	at apps.gui.GuiLafPreferencesManager.<init>(GuiLafPreferencesManager.java:70)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at java.lang.Class.newInstance(Class.java:442)
	at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:380)
	... 7 more
```

So the main questions are: 
What's the correct procedure of running JMRI on a headless machine? I hoped I could copy XML files from my PC to beaglebone, automatically start WiThrottle server and not bother with desktop PC at all. Is that even possible?

bobjacobsen, 2018-10-04:
I think you’re trying to do too much.  By installing the `default-jre-headless`, you’re providing a partial _Java_ environment.  

Instead, I recommend that you install a (regular) JRE, and then run it in headless mode. It won’t try to reference the screen, but it will still provide the class, timer, etc support that JMRI needs.

> Hello. I'm trying to start JMRI on headless beaglebone black but have problems with it.
> 
> 	• I've installed default-jre-headless from debian repos which provides java 8.
> 	• I unpacked fresh tgz archive (v4.12/JMRI.4.12+Rb6a9bb1) and tried to run ./JMRIFaceless.
> 	• all sorts of errors relating to AWT and Swing. I'm trying to fix them one by one as they appear, but there are so many that I started to doubt that it will work.
> What I've encountered so far:
> 
> 	• jmri.util.Log4JUtil calls SystemConsole.create() which is a JTextArea. I've added GraphicsEnvironment.isHeadless() check.
> 	• jmri.beans.Bean.firePropertyChange uses jmri.util.ThreadingUtil.runOnGUIEventually which calls SwingUtilities.invokeLater. If headless, I make a direct call instead of invokeLater. This is probably a huge change but I did not get to a point where it fails. Other methods jmri.util.ThreadingUtil patched similarly.
> 	• jmri.implementation.JmriConfigurationManager tries to instantiate apps.gui.GuiLafPreferencesManage which certainly fails. Here I don't know how to remove it from loading chain.
> 

> 
> So the main questions are:
> What's the correct procedure of running JMRI on a headless machine? I hoped I could copy XML files from my PC to beaglebone, automatically start WiThrottle server and not bother with desktop PC at all. Is that even possible?
> 
> 

--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

mstevetodd, 2018-10-04:
I tried it on my RaspberryPi just now and it worked when I passed an existing profile
`JMRI/JmriFaceless -p Digitrax_Simulator.3f66e423`
I did encounter the crash in apps.SystemConsole.getFrame() that you mentioned. That's been added since I last tested JmriFaceless. Please create a PR with the headless check to avoid that.

positron96, 2018-10-07:
Ok, I'll prepare the PR.

closed 2021-04-14 by stale[bot]