JMRI/JMRI #5940: Abstract Automation abend

KenC57, 2018-10-06:
Layout has an Android running Linux with a browser showing one of the panels on the layout computer, it displays to the public at the museum. After some period of time, usually hours to a few days, this message comes up and any further updates to the panels stop. The panel includes a real time clock and things like a lap counter and speed computation. Some of these freeze and stop getting updated on the main computer:

2018-10-03 14:36:00,890 automat.AbstractAutomaton             WARN  - Unexpected Exception ends AbstractAutomaton thread [Heartbeat]
java.lang.IllegalStateException: Blocking message pending 10000 for BLOCKING
	at org.eclipse.jetty.websocket.common.WebSocketRemoteEndpoint.lockMsg(WebSocketRemoteEndpoint.java:130)
	at org.eclipse.jetty.websocket.common.WebSocketRemoteEndpoint.sendString(WebSocketRemoteEndpoint.java:385)
	at jmri.jmris.JmriConnection.sendMessage(JmriConnection.java:85)
	at jmri.server.json.JsonConnection.sendMessage(JsonConnection.java:67)
	at jmri.server.json.memory.JsonMemorySocketService$MemoryListener.propertyChange(JsonMemorySocketService.java:100)
	at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:327)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:263)
	at jmri.implementation.AbstractNamedBean.firePropertyChange(AbstractNamedBean.java:243)
	at jmri.implementation.AbstractMemory.setValue(AbstractMemory.java:42)
	at sun.reflect.GeneratedMethodAccessor28.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:186)
	at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:204)
	at org.python.core.PyObject.__call__(PyObject.java:478)
	at org.python.core.PyObject.__call__(PyObject.java:482)
	at org.python.core.PyMethod.__call__(PyMethod.java:141)
	at org.python.pycode._pyx0.handle$11(<script>:392)
	at org.python.pycode._pyx0.call_function(<script>)
	at org.python.core.PyTableCode.call(PyTableCode.java:167)
	at org.python.core.PyBaseCode.call(PyBaseCode.java:307)
	at org.python.core.PyBaseCode.call(PyBaseCode.java:198)
	at org.python.core.PyFunction.__call__(PyFunction.java:482)
	at org.python.core.PyMethod.instancemethod___call__(PyMethod.java:237)
	at org.python.core.PyMethod.__call__(PyMethod.java:228)
	at org.python.core.PyMethod.__call__(PyMethod.java:218)
	at org.python.core.PyMethod.__call__(PyMethod.java:213)
	at org.python.core.PyObject._jcallexc(PyObject.java:3626)
	at org.python.core.PyObject._jcall(PyObject.java:3658)
	at org.python.proxies.__builtin__$Heartbeat$3.handle(Unknown Source)
	at jmri.jmrit.automat.AbstractAutomaton.run(AbstractAutomaton.java:150)
	at java.lang.Thread.run(Thread.java:748)

bobjacobsen, 2018-10-07:
For some possible background on this:

https://bugs.eclipse.org/bugs/show_bug.cgi?id=474488

https://stackoverflow.com/questions/36305830/blocking-message-pending-10000-for-blocking-using-spring-websockets

KenC57, 2018-10-08:
I'm seeing two parts to this problem. One is the web side of things and how it seems we might have some issues in the web serving part or how the remote HTML for a panel is asking for status. This is hinted at by the articles supplied by Bob J, something in the threading of either the web service or the remote panel is messing up the threads.

The second side, one I find more bothersome, is how a fault in the web service is back feeding into the rest of the automatons and locking up script driven panel (memory) updates. In one case a very simple automation that is formatting time and putting it in the memory for display. But  many (if not all) other script driven threads doing updates see to either die or lock up. Next failure I will inspect the threads and see what status some of the ones that seem to freeze are in or gone.

bobjacobsen, 2018-10-21:
Are you running your own script for this, or is it something built into JMRI?

closed 2021-04-14 by stale[bot]