JMRI/JMRI #5943: Enhancement Request: Add graphical icons to description of function keys

DrCodeWizzard, 2018-10-06:
As and end user I think it would be awesome if jrmi supported assigning graphical icons as descriptions of decoder function buttons (similarly to what is done in esu ecos 2, Z21 app, and esu mobile control 2). It would be really great if these icons were synced automatically over WiFi to whatever throttle software users are using (WiThrottle or EngineDriver). In case of EngineDriver that would solve the problem where the buttons truncate function descriptions.

pabender, 2018-10-06:
I am not positive about sending the icons to WiThrottle or engine driver, but we already support assigning icons to buttons in throttles.

See: 
http://jmri.org/help/en/manual/DecoderPro/Main_Throttle_functionPanel.shtml

KenC57, 2018-10-06:
The roster supports icons for function buttons. The JMRI throttles use them.
As I understand it, the format requirements for the phones may be trickier.
But last I heard, neither phone app supports icons. Granted I don't know if
the WiThrottle server knows how to offer the icons.
-ken c

mstevetodd, 2018-10-09:
@DrCodeWizzard EngineDriver currently ignores the function icons.
It could be enhanced to do so, retrieving the images via the web server just as is done with the loco icons.
However, when I looked into this a while back, I could not find suitable set of icons beyond the light, horn and bell. Most of the functions on my locos do not lend themselves to easy iconization. 

Do you have a set of icons that you use? I'd like to know that a useful set existed.

DrCodeWizzard, 2018-10-09:
@mstevetodd I don't have a personal useful set to which I own graphical copyrights. All I can do is point to either ESU or Z21 documentation for examples. They have an exhaustive set of icons that seem to cover majority (if not all sound functions I've encountered so far). At the moment I use the Z21 app for controlling trains, but as I want to add signaling to my layout I'm looking at moving to JRMI (I already use DecoderPro to store my locomotive roster. A benefit of switching to JRMI is that it would allow me to use ESU throttles as well :-) irrespective of what base DCC system I use to run trains). 

Just to provide a bit of a feedback of why I'm looking into this:
I model Polish State Railways in N scale. Unlike US prototypes the loco numbers are really small so the standard way of just selecting the loco number as a DCC address does not work, especially when standing and looking at the layout from above. So I fallback on using railcom for returning the address of locomotive on a given track. Currently I can display that address in JRMI panels as well as the Z21 app. Many of my locomotives also have sound (and go figure every manufacturer maps a sound function to a different function key) so something like JRMI or Z21 or esu app are useful for knowing what sound is what function button.

In ideal world I'd love to be able to display icons for sound (so the labels don't get truncated), and also see the web panel on the throttle or tablet showing me the loco number to select. 

The benefit of using JRMI over relying on just the Z21 app is that it is a central storage point of all locomotives, and there is little chance of the repository going out of sync ... (I've already de-synchronized my Z21 app and I only have 3 devices with the app on it...)... I guess I'm trying to get the best possible DCC system??

dheap, 2018-10-09:
@mstevetodd As @DrCodeWizzard has said, there is an ESU icon set. I'll look for the link to the document.

The ESU decoders have provision for storing the Icon Number against F key in decoder CVs and on my if-I-live-long-enough-plans to be able to link these (and the also-stored momentary status, etc.) to  the JMRI Function Labels panel.

I can provide more information.

dheap, 2018-10-09:
Function symbols (ECoS & CS "Reloaded") document from http://www.esu.eu/en/downloads/instruction-manuals/digital-systems/

DrCodeWizzard, 2018-10-09:
And here is the document link to the Z21 symbols for functions:
http://www.z21.eu/en/Z21/Z21-App/Function-icons

bobjacobsen, 2018-10-09:
The ESU and Z21 examples are interesting,  but please don’t make copies of them (or develop new ones starting _from_ them)  for inclusion in JMRI unless you get permission from their author(s). The terms and conditions on both sets make it clear that their authors do not want that.

dheap, 2018-10-09:
In the case of the ESU Function Labels and icons the Index(icon number), Momentary, Inverted & Category column items are stored in decoder CVs readable by JMRI. Do we need to get permission to map the stored Index number to an icon set of our own making, but with a similar meaning? We map CV values to word/phrase lists in enums. Does the situation differ with images?

[image]

DrCodeWizzard, 2018-10-09:
I was going to ask the same question actually... I agree that JRMI should not copy anyones images, but I think it would be fair to use images used elsewhere as an inspiration... In my mind, there are only so many ways of drawing a turtle for slow/switching speed or steam out of chimney for either smoke generator or sound generator, or light bulb for lights... 

Also, if the capability is there to add user images, and send them to throttles, then individuals could develop their own sets for posting, sharing and eventual incorporation into the project, etc... 

Also perhaps we could ask ESU guys if their graphics could be used by JRMI...? (Though I'm a developer by day, I have not contributed to JRMI ,so I do not know what are the policies around this in the project)

DrCodeWizzard, 2018-10-09:
@bobjacobsen thanks for clarifying

DrCodeWizzard, 2018-10-12:
Ok. So I can volunteer to create an icon set. Is 64*64 pixels ok? I'm not a graphics designer but I think I can get ball rolling if that will help getting icons for buttons in Android devices. 

My preliminary list is:
Lights
Bell
Whistle short
Whistle long
Whistle
Horn
Horn low
Horn high
Shunting speed
Momentum
Breaking curve
Acceleration curve
Cab lights
Cab talk
Signal board lights
Walkway lights
Car lights
Smokebox flicker
Smoke
Sound 
Compressor
Break squeal
Air pump
Doors
Station announcements
Shovel
Steam blow down
Track sound
Curve squeal
Sand to track
Coupler sound
Rpm
volume

Any others?

DrCodeWizzard, 2018-10-12:
Ok. Here's what I got so far...
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

DrCodeWizzard, 2018-10-13:
And here's the rest from my list:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

DrCodeWizzard, 2018-10-13:
[image]

bobjacobsen, 2018-10-13:
These are great.  Happy to include them.  A couple of things:

- Where should they go?  Perhaps as a new "functionlabels" directory in [resources/icons](http://jmri.org/resources/icons/)?  Or in the already-existing (and with lots of contents, including some a few related icons) [resources/icons/throttles](http://jmri.org/resources/icons/throttles/) directory?

- Would you mind attaching a .zip file of these?  That'll make it more likely that I'll get them all merged intact.

Thanks!

DrCodeWizzard, 2018-10-13:
Will do. I will add a few extra ones. Going through my steam engine sounds. I will need to go through diesel sounds on my bli csx add some diesel sounds icons. As well as fans.  I will add the gimp xcf files so people can modify the source files if need be. Any icons that strike anyone as missing?

DrCodeWizzard, 2018-10-13:
Here's all the icons in a single zip file. Hopefully that will make a complete set that can be used in EngineDriver as well.

[JRMI Icons.zip](https://github.com/JMRI/JMRI/files/2475837/JRMI.Icons.zip)

DrCodeWizzard, 2018-10-13:
@bobjacobsen are png images ok as well as gifs?

DrCodeWizzard, 2018-10-13:
another question... is transparent background ok? If not, do you have a preference? Looks like web throttle is having problems displaying transparency in buttons.

DrCodeWizzard, 2018-10-13:
[image]

DrCodeWizzard, 2018-10-13:
Ok. Tracked it down to gif transparency channel. I will re-save everything as png. That appears to fix the problem.

DrCodeWizzard, 2018-10-13:
@pabender  @mstevetodd  @dheap  @bobjacobsen All right here is the status: 
If I save as PNG file with transparency then windows 10 displays all icons with black background but JRMI shows them correct.
If I save as GIF file with transparency then JRMI shows all icons with black background but Windows shows them correct.
Given this state I am thinking of either using a gray or white background for all icons. Any preferences?

DrCodeWizzard, 2018-10-13:
Well this is interesting... windows shows png images with incorrect background if the images are on my onedrive drive... but showes them correctly if i copy them wholesale someplace else. Still not sure why JMRI is not showing gifs correctly...

pabender, 2018-10-14:
Sent from my iPhone

> On Oct 13, 2018, at 7:40 PM, Konrad Malkowski <notifications@github.com> wrote:
> Given this state I am thinking of either using a gray or white background for all icons. Any preferences?
> 
My preference is a transparent background.   Jmri may not always put these on a background of a specific color, and there may be cases where these icons are used on other background colors.  ( someone may want to use one of these icons on a panel, for example ).

bobjacobsen, 2018-10-14:
Perhaps the code isn’t setting a background color for these images. If it’s not, what’s presented for transparent regions is not well defined by Java. We should make sure it’s set by JMRI. Not sure what to do for the apps.

DrCodeWizzard, 2018-10-14:
I'm in the process on putting all the gifs on a white background. There seems to be no problem with PNGs... I will deliver all of the files in three subdirectories:
white_background, gray_background and transparent
That way best possible choice can be made by end user/app designer.

DrCodeWizzard, 2018-10-14:
Ok here is the set of 193 icons in transparent background and 193 corresponding icons on white background. Please place it in whatever directory is appropriate :-)
[JRMI Icons final.zip](https://github.com/JMRI/JMRI/files/2476673/JRMI.Icons.final.zip)

bobjacobsen, 2018-10-14:
These are in PR #5998 now.  Thanks!

Bob

closed 2021-04-14 by stale[bot]