JMRI/JMRI #5949: Is an interface in package jmri allowed to return a JmriPanel?

danielb987, 2018-10-07:
I'm thinking of maybe adding a new feature to JMRI and I'm thinking of how to structure the code.

I would like to have some interfaces, which should probably be located in package jmri, there each of them need to be able to supply a panel to configure itself. So I'm thinking of a parent interface like:
```
interface Configurable {
    JmriPanel getPanel();
}
```
which the other interfaces then inherits. But if I read [Introduction to JMRI Library Structure](http://jmri.sourceforge.net/help/en/html/doc/Technical/IntroStructure.shtml) correctly, it's wrong to use swing in package jmri.

A simple solution is to return an Object instead of a JmriPanel, but I'm not sure if that's better.

pabender, 2018-10-07:
Two questions:
1) can you tell us more about what you are trying to do.
2) how do you envision this feature working in a headless environment?

danielb987, 2018-10-07:
It's about Logix. In PR #5939, I have changed integer constants to enums in Conditional. That has got me thinking about maybe refactor Conditonal. But I don't know where to start and how to make the code better and it's difficult to refactor code without breaking user configuration.

So I'm thinking it might be better to create a complete new Logix, a NewLogix. (Better name is welcome).

My idea is to have Expression and Action, where Expression is anything that can answer True or False, and Action is anything that can do something. Each NewLogix would then have one Expression and one Action. Then I would create concrete classes for Expression, like asking if a turnout has a specifik state, or an expression which is AND of several other expressions. And I would create concrete classes for Action, for example throwing a switch or playing a sound, but also actions that do several other actions.

I'm thinking of something like this:
```
interface Configurable {
    JmriPanel getPanel();
}

interface Expression inherits Configurable {
    boolean evaluate();
}

interface Action inherits Configurable {
    void run();
}

class ExpressionTurnoutStatus implements Expression ...
class ExpressionAnd implements Expression ...
class ActionSetTurnout implements Action ...
```
I would like to create an Editor that can configure both expressions and actions, but without knowing anything about the specific expressions and actions. In the current Conditional editors, there is several switch statements with case statements for everything the Conditional supports. It makes it difficult to read the code and it makes it difficult to add new things.

The panel is requested in a non headless environment. You cannot open an Editor if you are running headless. And in Issue #5930 @bobjacobsen recommends a user to install full Java JRE despite the user is running headless. And therefore I think it would not be any problem with headless since the recommendation is to run JMRI on a full JRE even then running headless.

If the user is running headless, the parts that requires a GUI, like the editor, would be disabled. But everything else working as usual.

pabender, 2018-10-07:
If you were to build a new replacement for logix, it would go into the jmri.jmrit tree, just like the existing one.  It is a tool that uses items in the jmri.jmrix tree.  

The interfaces in jmri provide the link between those two.

pabender, 2018-10-07:
One other comment about this:

> It's about Logix. In PR #5939, I have changed integer constants to enums in Conditional. That has got me thinking about maybe refactor Conditonal. But I don't know where to start and how to make the code better and it's difficult to refactor code without breaking user configuration.

Sometimes we break things.   ( the saying goes you have to break a few eggs to make an omelet).  We try not to, but it happens.  When we do, we fix it and move on.  

This is also where testing comes into play.  If you are worried about breaking something, make sure there is a test that captures the current behavior and make sure you don’t break the test with your new code.

pabender, 2018-10-07:
> Yes. But my question is about the interfaces Configurable, Expression and Action. The corresponding interfaces in Logix, which is Conditional, ConditionalAction and ConditionalVariable lies in package jmri. And I would like to have a method that returns a JmriPanel in the interface Configurable.
> 

I am actually not sure the conditional interfaces you mention really belong there, but that is water under the bridge.

> But I could solve it by letting the method in the interface return an Object instead of a JmriPanel, if that is better.

Being more precise is probably better.

bobjacobsen, 2018-10-07:
Coming to this a bit late:

In the long run, we don't want any Swing in `jmri` or even in the main functional packages i.e. `jmri.jmrit.roster` etc.  There's a [web page](http://jmri.org/help/en/html/doc/Technical/Swing.shtml) that talks about this briefly, and should be beefed up. 

The idea is to have JMRI function _without_ Swing, either headless or on Android or somewhere else.  That's a long way from here, there's a lot to do and we might never get there.  

The preferred structure is to separate the GUI configuration from the object, by having a separate class that handles the GUI. Although there are more classes (by a factor of two or so) when you do this, each one is _much_ simpler:  You're guaranteed to be able to test the Functional class without going through a GUI, and you're guaranteed to be able to test the GUI class with a mock Functional because all the GUI class is allowed to do is to set and get state and make specific calls.  Once the class structure is set up, the resulting code is pretty much cleaner.

For example:

[image]

(Sorry the lines are all tangled, haven't had time to sort that out)

So if you're programming a button that says "open the configuration tool for a NewFunctionTypeB", it creates a `NewFunctionTypeBPane` and uses it. That might be the actual pane (i.e. to use it, put it in a frame) or it might be a factory with a `getPane()` call.

The one piece of complexity is when you have a `NewFunction` object and want to create a GUI that knows about the specific type.  It's true that if there was a `jmri.NewFunction#getPane()` method, that could be reflected down to the specific subtypes and come back with a concrete implementation that knows about the type.  `jmri.swing.NewFunctionPane` can't do that as it. It's not that it doesn't know the concrete type of the `NewFunction` object it's configuring: that's easy in Java.  It's that it doesn't know a-priori the proper class to use.

Randall would argue that this is a place to use services:  Have each `NewFunctionPane` register for what it can configure, and use that to look up what to use.  I don't know whether that's the best approach or not.

Another way, which I think is conceptually simpler, is to have a convention:  Any `foo.Bar` class that descends from `jmri.NewFunction` that has a class that will handle its GUI configuration must call it `foo.swing.BarPane`. You can then look that class up quickly and efficiently via reflection.  This is how ConfigureXML works quite successfully.  

None of this is necessary if there's just one NewFunction interface and _everything_ can be configured through it, because then you only need the one configure pane class.

danielb987, 2018-10-07:
> Coming to this a bit late:

No problem. Thanks for your answer. I will look into it. Having a convention that jmri.jmrit.newlogix.ExpressionAnd and jmri.jmrit.newlogix.swing.ExpressionAndPane seems to be a good solution.

With that problem probably solved comes the question there the top interfaces Expression and Action should go. In package jmri, jmri.jmrit.newlogix or somewhere else?

bobjacobsen, 2018-10-07:
>  In the current Conditional editors, there is several switch statements with case statements for everything the Conditional supports. It makes it difficult to read the code and it makes it difficult to add new things.

The way to fix this is first to break Conditional itself into specific subparts, where subtypes are used to implement _just_ _one_ kind of functionality:

[image]

Then you have no `switch` statements:  You have a list of `ConditionalOperator` objects, and you go down the list telling each one to `doYourThing()` and it does.

Since those specific types are now in `jmri.implementation` (or perhaps some other low-level package) I don't care so much if they have a `JPanel getPane()` method in them.  That'll require loading Swing to run (not just configure) logix, but that's already true.  And if all the Swing stuff is kept in a local `extends JPanel` local class, we can factor that out later pretty easily.

devel-bobm, 2018-10-07:
> Each NewLogix would then have one Expression and one Action. 

It is too restrictive to allow only one Action for each Expression.

I have implemented a small CTC machine (5 levers) in Logix, plus one non-CTC interlocking.  I found that it was not particularly common to have a single Logix Conditional with only one Action.  A quick count from that design shows that the panel .XML has 325 occurrences of "<Conditional " and 628 "<conditionalAction ".  I can think of some pieces of the "interlocking" where a single conditional had as many as 8 actions.

danielb987, 2018-10-07:
> > Each NewLogix would then have one Expression and one Action.
> 
> It is too restrictive to allow only one Action for each Expression.

Instead of having the Logix itself having a number of Expressions and Actions, I would have an ExpressionAnd that has a number of expressions and evaluate these and do boolean AND. And I would have an ExpressionMixed which also takes a number of Expressions and do a mixed evaluation, like "R1 and not (R2 or R2)".

Same with Actions. I would have an ActionMany that has many other actions which it executes.

Let say that you want to set a route:
```
1) Set turnouts 12, 23 and 25 to desired states.
2) Wait until the turnouts has got to their states.
3) Set signal 52 and signal 48 to Green.
```
My idea is that this Logix has this Action tree:
```
* ActionManySortedOrder - Do things in a predefined order.
    * ActionSetTurnout - Set turnout 12
    * ActionSetTurnout - Set turnout 23
    * ActionSetTurnout - Set turnout 25
    * ActionDoWhen - Do something when a condition is true
        ? ExpressionAnd - Boolean AND condition on many expressions
            ExpressionTurnoutState - Is turnout 12 in desired state?
            ExpressionTurnoutState - Is turnout 23 in desired state?
            ExpressionTurnoutState - Is turnout 25 in desired state?
        * ActionDoMany
            * ActionSetSignal - Set signal 52 to Green
            * ActionSetSignal - Set signal 48 to Green
```

danielb987, 2018-10-07:
Interesting. But I'm not too surprised. In the last weeks, I was hit by an if statement which evaluated two functions and there both needed to be called, even if the first returned false. I therefore needed to call the functions first and assign their return values to variables, and then do the if statement.

If someone remembers the difference between one Logix with two Conditionals, and two Logix with each with one of those two Conditionals, it would be interesting to hear about it.

devel-bobm, 2018-10-07:
The existing Logix implementation has some real nuances with respect to order-of-operation - ones I never really understood.  I had a dickens of a time getting some of my Logix to execute correctly.  I occasionally found that stacking the "Conditional"s within a given logix in a different order would have a significant influence on the behavior of my implementation.

danielb987, 2018-10-07:
I want to achieve two things in this regard:

* By having each kind of Expression in its own class, it will be much easier to see what that Expression actually does, by looking at the source code for that class.

* If the current set of Expressions doesn't do what is desired, a new type of Expression could easily be created. If, for exampe, the evaluation of the expressions take some time, someone could create an ExpressionAndMultithreading that uses a thread pool to launch a thread for each expression and thereby evaluates all the expressions in parallel.

I'm interested to hear if you have more thoughts about that works and what doesn't work in Logix. My ideas might be off topic.

bobjacobsen, 2018-10-07:
I've taken a quick pass through my notes on this. Not clear they actually describe what's happening now, nor that I understand them, but here they are:

- When a state change triggers recalculation, that's handled at the level of the Logix.  It recalculates the Conditionals in order, and they re-evaluate their state variables in order.

- The "Trigger Calculation" specifies whether that _particular_ Variable (in that conditional) will cause the _Logix_ to recalculate when it fires.  Note that this can be inconsistent: A variable in one conditional can say to recalculate, while another says not to. I don't know what happens in that case. Also note that it appears to be "calculate on listener fired", not "calculate on listener says there's a change in the relevant state". 
- By having the Logix handle all the invocation, you can be sure that Conditional execution doesn't overlap:  An action in Conditional A won't result in Conditional B firing before all the actions in Conditional A have completed.

This ordered structure lets you do procedural programming, which some people find easier than state-based programming.

If you're going to make a NewLogix, it might be good to decide in advance what the execution model.

dsand47, 2018-10-09:
Some additional comments:

I have a **_medium_** size Logix setup:  56 Logixs, 431 conditionals, 1456 state variables and 2040 actions.

The conditional state is always recalculated each time one of the state variables change.  If a trigger variable changes, then the action section might be invoked.  If the trigger type is **_when triggered_**, the actions are processed.  If the type is **_on change_**, then the actions are only processed if the conditional state itself changed.  Additionally, each action can be invoked on conditional **_true_**, **_false_**, or **_always_**.

The run time logic is in jmri, jmri.managers, and jmri.implmentation.  The GUI components are in jmri.jmrit.beantable and jmri.jmrit.conditional.  Last year the Conditional maintenance GUI was moved out of jmri.jmrit.beantable.LogixTableAction to jmri.jmrit.conditional.  I have always wondered why jmri.ConditionalVariable is a class instead of an interface with a default class in jmri.implementation.

Ignore jmri.jmrit.logix.  It has nothing to do with Logix.  It is the CPE Warrant/OBlock/OPath/Portal feature.  Another historical curiosity.

danielb987, 2018-10-09:
Thanks.

What is "Delayed Set Turnout" and "Delayed Reset Turnout" used for? I understand what they do, but not why they are used.

danielb987, 2018-10-10:
What is the procedure to get a new device type? I'm working on a demo for NewLogix and I will need system names for it.

Letter Q seems to be free and I suppose my system names should start with letter I, so my idea is to have system names beginning with 'IQ'.

danielb987, 2018-10-11:
With device type letter, I refer to [this page](http://jmri.sourceforge.net/help/en/html/doc/Technical/Names.shtml), there A is audio, B is block and so on.

I read about normalization of system names. Would the following naming convention for system names be ok?
`IQM10` - user entered NewLogix
`IQA20` - program generated NewLogix
`IQM3:E54` - user entered Expression
`IQA55:A321` - program generated Action

`IQM` is then system names the program generates and `IQA` is system names the user chooses. Or is it a disadvantage to be so specific as I am above? One benefit is that the method `manager.normalizeSystemName()` can be very specific about what to approve.

danielb987, 2018-10-11:
> What is the procedure to get a new device type? I'm working on a demo for NewLogix and I will need system names for it.

I [found](http://jmri.sourceforge.net/help/en/html/doc/Technical/NewType.shtml) it.

closed 2021-04-14 by stale[bot]