JMRI/JMRI #5951: Cannot launch JMRI Demo 4.12 - tested on 2 PCs

jba3, 2018-10-07:
Unable to launch the main JMRI program on either of two PCs tested. Both using PR4 and Digitrax DCS210. Both worked with JMRI 4.10.

The "test" that gets installed with JMRI runs without errors and opens windows as expected. DecoderPro will also open fine if launched directly. But I can't get the main program to start. Here is a dump of the /debug flag:

```
CommandLine: "C:\Program Files (x86)\JMRI\LaunchJMRI.exe" /debug /noisy apps.JmriDemo.JMRIdemo
Default options: 
AppName: JMRIdemo
Class: apps.JmriDemo.JMRIdemo
Parameters: 
Noisy: 1
Force32bit: 0
Profile: 
Testing for x64...
Result: 1
Setting x64 registry view...
Setting x86 registry view...
JavaPath: C:\Program Files (x86)\Java\jre1.8.0_181\bin\java.exe
Delete file: C:\Users\jamie\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe
Delete file: C:\Users\jamie\AppData\Local\Temp\LaunchJMRI\JMRIdemo.exe
Remove folder: C:\Users\jamie\AppData\Local\Temp\LaunchJMRI\
Create folder: C:\Users\jamie\AppData\Local\Temp\LaunchJMRI
CopyFiles: C:\Program Files (x86)\Java\jre1.8.0_181\bin\java.exe C:\Users\jamie\AppData\Local\Temp\LaunchJMRI\JMRIdemo.exe
Result: 0
JExePath: C:\Users\jamie\AppData\Local\Temp\LaunchJMRI\JMRIdemo.exe
PhysicalMemory: 8113m
Check if more than 4GB memory
More than 4GB
InitHeap: 96m
MinMemory: 192m
MaxMemory: 2028m
Checking maximum heap size...
Running x86 JVM
Adjusted MaxMemory: 1024m
...finished
Checking user.home...
Reading Desktop Shell Folder registry key...
...read: C:\Users\jamie\Desktop
Checking if equal to %USERPROFILE%...
Comparing: C:\Users\jamie
to: C:\Users\jamie
user.home OK
Options:   -noverify -Dsun.java2d.d3d=false -Djava.security.policy=security.policy -Djogamp.gluegen.UseTempJarCache=false -Djinput.plugins=net.bobis.jinput.hidraw.HidRawEnvironmentPlugin -Dswing.defaultlaf=com.sun.java.swing.plaf.windows.WindowsLookAndFeel -Djava.library.path=.;lib\x86;lib -Djava.rmi.server.codebase=file:java/classes/ -Dsun.java2d.noddraw -Xms96m -Xmx1024m -Dfile.encoding=UTF-8 -Dpython.home="C:\Users\jamie\JMRI\systemfiles" -Djmri.log.path="C:\Users\jamie\JMRI\systemfiles\\"
ClassPath: .;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\commons-io-2.6.jar;lib\commons-lang3-3.7.jar;lib\commons-text-1.2.jar;lib\gluegen-rt.jar;lib\hid4java-0.5.0.jar;lib\jackson-annotations-2.8.11.jar;lib\jackson-core-2.8.11.jar;lib\jackson-databind-2.8.11.jar;lib\jakarta-regexp-1.5.jar;lib\javacsv.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom-2.0.5.jar;lib\jdom.jar;lib\jetty-http-9.4.7.v20170914.jar;lib\jetty-io-9.4.7.v20170914.jar;lib\jetty-security-9.4.7.v20170914.jar;lib\jetty-server-9.4.7.v20170914.jar;lib\jetty-servlet-9.4.7.v20170914.jar;lib\jetty-util-9.4.7.v20170914.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jlfgr-1_0.jar;lib\jmdns-3.5.1.jar;lib\jna-4.4.0.jar;lib\jna-platform-4.4.0.jar;lib\joal.jar;lib\json-schema-validator-0.1.19.jar;lib\jul-to-slf4j-1.7.25.jar;lib\jython-standalone-2.7.0.jar;lib\libusb4java-1.2.0-linux-arm.jar;lib\libusb4java-1.2.0-linux-x86.jar;lib\libusb4java-1.2.0-linux-x86_64.jar;lib\libusb4java-1.2.0-osx-x86.jar;lib\libusb4java-1.2.0-osx-x86_64.jar;lib\libusb4java-1.2.0-windows-x86.jar;lib\libusb4java-1.2.0-windows-x86_64.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\mqtt-client-0.4.0.jar;lib\openlcb.jar;lib\pi4j-core.jar;lib\pi4j-device.jar;lib\pi4j-gpio-extension.jar;lib\pi4j-service.jar;lib\purejavacomm-1.0.1.jar;lib\slf4j-api-1.7.25.jar;lib\slf4j-log4j12-1.7.25.jar;lib\smtp.jar;lib\usb-api-1.0.2.jar;lib\usb4java-1.2.0.jar;lib\usb4java-javax-1.2.0.jar;lib\vecmath-1.5.2.jar;lib\websocket-api-9.4.7.v20170914.jar;lib\websocket-client-9.4.7.v20170914.jar;lib\websocket-common-9.4.7.v20170914.jar;lib\websocket-server-9.4.7.v20170914.jar;lib\websocket-servlet-9.4.7.v20170914.jar;lib\xAPlib.jar;lib\xbee-java-library-1.2.0.jar;lib\xercesImpl.jar
Check for any prepended/appended classpath entries
Final ClassPath: .;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\commons-io-2.6.jar;lib\commons-lang3-3.7.jar;lib\commons-text-1.2.jar;lib\gluegen-rt.jar;lib\hid4java-0.5.0.jar;lib\jackson-annotations-2.8.11.jar;lib\jackson-core-2.8.11.jar;lib\jackson-databind-2.8.11.jar;lib\jakarta-regexp-1.5.jar;lib\javacsv.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom-2.0.5.jar;lib\jdom.jar;lib\jetty-http-9.4.7.v20170914.jar;lib\jetty-io-9.4.7.v20170914.jar;lib\jetty-security-9.4.7.v20170914.jar;lib\jetty-server-9.4.7.v20170914.jar;lib\jetty-servlet-9.4.7.v20170914.jar;lib\jetty-util-9.4.7.v20170914.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jlfgr-1_0.jar;lib\jmdns-3.5.1.jar;lib\jna-4.4.0.jar;lib\jna-platform-4.4.0.jar;lib\joal.jar;lib\json-schema-validator-0.1.19.jar;lib\jul-to-slf4j-1.7.25.jar;lib\jython-standalone-2.7.0.jar;lib\libusb4java-1.2.0-linux-arm.jar;lib\libusb4java-1.2.0-linux-x86.jar;lib\libusb4java-1.2.0-linux-x86_64.jar;lib\libusb4java-1.2.0-osx-x86.jar;lib\libusb4java-1.2.0-osx-x86_64.jar;lib\libusb4java-1.2.0-windows-x86.jar;lib\libusb4java-1.2.0-windows-x86_64.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\mqtt-client-0.4.0.jar;lib\openlcb.jar;lib\pi4j-core.jar;lib\pi4j-device.jar;lib\pi4j-gpio-extension.jar;lib\pi4j-service.jar;lib\purejavacomm-1.0.1.jar;lib\slf4j-api-1.7.25.jar;lib\slf4j-log4j12-1.7.25.jar;lib\smtp.jar;lib\usb-api-1.0.2.jar;lib\usb4java-1.2.0.jar;lib\usb4java-javax-1.2.0.jar;lib\vecmath-1.5.2.jar;lib\websocket-api-9.4.7.v20170914.jar;lib\websocket-client-9.4.7.v20170914.jar;lib\websocket-common-9.4.7.v20170914.jar;lib\websocket-server-9.4.7.v20170914.jar;lib\websocket-servlet-9.4.7.v20170914.jar;lib\xAPlib.jar;lib\xbee-java-library-1.2.0.jar;lib\xercesImpl.jar
MaxLen: 8192
ExeString: "C:\Users\jamie\AppData\Local\Temp\LaunchJMRI\JMRIdemo.exe"   -noverify -Dsun.java2d.d3d=false -Djava.security.policy=security.policy -Djogamp.gluegen.UseTempJarCache=false -Djinput.plugins=net.bobis.jinput.hidraw.HidRawEnvironmentPlugin -Dswing.defaultlaf=com.sun.java.swing.plaf.windows.WindowsLookAndFeel -Djava.library.path=.;lib\x86;lib -Djava.rmi.server.codebase=file:java/classes/ -Dsun.java2d.noddraw -Xms96m -Xmx1024m -Dfile.encoding=UTF-8 -Dpython.home="C:\Users\jamie\JMRI\systemfiles" -Djmri.log.path="C:\Users\jamie\JMRI\systemfiles\\" -Djava.class.path=".;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\commons-io-2.6.jar;lib\commons-lang3-3.7.jar;lib\commons-text-1.2.jar;lib\gluegen-rt.jar;lib\hid4java-0.5.0.jar;lib\jackson-annotations-2.8.11.jar;lib\jackson-core-2.8.11.jar;lib\jackson-databind-2.8.11.jar;lib\jakarta-regexp-1.5.jar;lib\javacsv.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom-2.0.5.jar;lib\jdom.jar;lib\jetty-http-9.4.7.v20170914.jar;lib\jetty-io-9.4.7.v20170914.jar;lib\jetty-security-9.4.7.v20170914.jar;lib\jetty-server-9.4.7.v20170914.jar;lib\jetty-servlet-9.4.7.v20170914.jar;lib\jetty-util-9.4.7.v20170914.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jlfgr-1_0.jar;lib\jmdns-3.5.1.jar;lib\jna-4.4.0.jar;lib\jna-platform-4.4.0.jar;lib\joal.jar;lib\json-schema-validator-0.1.19.jar;lib\jul-to-slf4j-1.7.25.jar;lib\jython-standalone-2.7.0.jar;lib\libusb4java-1.2.0-linux-arm.jar;lib\libusb4java-1.2.0-linux-x86.jar;lib\libusb4java-1.2.0-linux-x86_64.jar;lib\libusb4java-1.2.0-osx-x86.jar;lib\libusb4java-1.2.0-osx-x86_64.jar;lib\libusb4java-1.2.0-windows-x86.jar;lib\libusb4java-1.2.0-windows-x86_64.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\mqtt-client-0.4.0.jar;lib\openlcb.jar;lib\pi4j-core.jar;lib\pi4j-device.jar;lib\pi4j-gpio-extension.jar;lib\pi4j-service.jar;lib\purejavacomm-1.0.1.jar;lib\slf4j-api-1.7.25.jar;lib\slf4j-log4j12-1.7.25.jar;lib\smtp.jar;lib\usb-api-1.0.2.jar;lib\usb4java-1.2.0.jar;lib\usb4java-javax-1.2.0.jar;lib\vecmath-1.5.2.jar;lib\websocket-api-9.4.7.v20170914.jar;lib\websocket-client-9.4.7.v20170914.jar;lib\websocket-common-9.4.7.v20170914.jar;lib\websocket-server-9.4.7.v20170914.jar;lib\websocket-servlet-9.4.7.v20170914.jar;lib\xAPlib.jar;lib\xbee-java-library-1.2.0.jar;lib\xercesImpl.jar" apps.JmriDemo.JMRIdemo 
Output folder: C:\Program Files (x86)\JMRI
Launching JMRI
Execute: "C:\Users\jamie\AppData\Local\Temp\LaunchJMRI\JMRIdemo.exe"   -noverify -Dsun.java2d.d3d=false -Djava.security.policy=security.policy -Djogamp.gluegen.UseTempJarCache=false -Djinput.plugins=net.bobis.jinput.hid [... cut ...] .4.7.v20170914.jar;lib\xAPlib.jar;lib\xbee-java-library-1.2.0.jar;lib\xercesImpl.jar" apps.JmriDemo.JMRIdemo 
**Return code from process: 1**
To copy this text to the clipboard, right click then choose
  'Copy Details To Clipboard'
Completed

```

I'm sure the return code 1 is what makes it not open, but there's no info as to what is causing the code 1.

Would appreciate any help! I'd like to get 4.12 running so I can add my command station in the roster and read/set flags from there. Thank you!

pabender, 2018-10-07:
JMRIDemo hasn’t worked for a long time because we changed the way menu items are constructed so we can have multiple connections of each type.  I didn’t think we were still shipping the launcher for it because it no longer works.

You should switch to using one of the other launchers ( DecoderPro, PanelPro, etc ).

bobjacobsen, 2018-10-07:
> Unable to launch the main JMRI program on either of two PCs tested. Both using PR4 and Digitrax DCS210. Both worked with JMRI 4.10.
>
> The "test" that gets installed with JMRI runs without errors and [...]

Note the last line. It’s trying to launch the “JMRIdemo” program, which no longer exists.  That should say apps.DecoderPro.DecoerPro instead. 

What command are you typing?

PUlvestad, 2018-10-07:
JMRIdemo still shows in the list of available startups in Windows and should be removed.

jba3, 2018-10-07:
@pabender @bobjacobsen it's the regular menu and desktop link that says "JMRI Demo" that I always used to open the app in v4.10

I upgraded both machines from 4.10 to 4.12 so it's possible the JMRI Demo icon was a holdover from previous version. PUIvestad, did you have a fresh install or upgrade too?

I confirmed everything worked fine after launching through Decoder Pro. I had issues getting multiple locos running under 4.10, hence the upgrade.

PUlvestad, 2018-10-07:
Fresh install of 4.13.4ish from Jenkins a couple hours ago.

bobjacobsen, 2018-10-07:
SoundPro should still be supported.

Looks like the JMRIdemo shortcut is still being created via scripts/WinInstallFiles/InstallJMRI.nsi  but I'm far from an expert in that control file's language.

rhwood, 2020-05-12:
I think this has been resolved.

PR #7709 (2019-12-16): Windows Installer remove JMRI Demo Shortcut
After next production release.

Remove JMRIDemo link
Rename "Tools and Demos" Folder to "Tools"
https link for Java download and JMRI Website
Closes #5951
closed 2020-05-12 by rhwood