JMRI/JMRI #5953: C/MRI monitor can't be launched via startup

bobjacobsen, 2018-10-07:
(From JMRIusers_

Using JMRI 4.13.2+R8a2b21d, JAVA version 1.8.0_181 on Windows 7 PC.

Put a start-up action of opening C/MRI Monitor in my prefs (just after opening the JMRI System Console). C/MRI device (arduino) connected to COM7. Got this error:

java.lang.InstantiationException: jmri.jmrix.cmri.serial.serialmon.SerialMonAction

Was then able to open console manually.

System console:

2018-10-07 11:36:51,360 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "C/MRI" [main]
2018-10-07 11:36:51,890 serialdriver.SerialDriverAdapter      INFO  - COM7 port opened at 19200 baud with DTR: true RTS: true DSR: false CTS: false  CD: false [main]
2018-10-07 11:36:52,967 startup.AbstractActionModel           ERROR - Could not instantiate specified class: jmri.jmrix.cmri.serial.serialmon.SerialMonAction [main]
java.lang.InstantiationException: jmri.jmrix.cmri.serial.serialmon.SerialMonAction
at java.lang.Class.newInstance(Unknown Source)
at apps.startup.AbstractActionModel.performAction(AbstractActionModel.java:111)
at apps.StartupActionsManager.lambda$1(StartupActionsManager.java:120)
at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(Unknown Source)
at java.util.stream.ReferencePipeline$2$1.accept(Unknown Source)
at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(Unknown Source)
at java.util.stream.AbstractPipeline.copyInto(Unknown Source)
at java.util.stream.AbstractPipeline.wrapAndCopyInto(Unknown Source)
at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(Unknown Source)
at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(Unknown Source)
at java.util.stream.AbstractPipeline.evaluate(Unknown Source)
at java.util.stream.ReferencePipeline.forEachOrdered(Unknown Source)
at apps.StartupActionsManager.initialize(StartupActionsManager.java:118)
at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:264)
at jmri.implementation.JmriConfigurationManager.lambda$1(JmriConfigurationManager.java:183)
at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(Unknown Source)
at java.util.stream.ReferencePipeline$Head.forEach(Unknown Source)
at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:182)
at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:170)
at apps.Apps.(Apps.java:278)
at apps.PanelPro.PanelPro.(PanelPro.java:40)
at apps.PanelPro.PanelPro.main(PanelPro.java:120)
Caused by: java.lang.NoSuchMethodException: jmri.jmrix.cmri.serial.serialmon.SerialMonAction.()
at java.lang.Class.getConstructor0(Unknown Source)
.. 22 more
2018-10-07 11:36:53,996 plementation.JmriConfigurationManager ERROR - Exception initializing apps.StartupActionsManager: jmri.JmriException: java.lang.InstantiationException: jmri.jmrix.cmri.serial.serialmon.SerialMonAction [main]
2018-10-07 11:37:09,756 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2018-10-07 11:37:09,756 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Jerry\JMRI\CMRI_Config\ [main]
2018-10-07 11:37:09,756 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Jerry\JMRI\CMRI_Config\ [main]
2018-10-07 11:37:09,756 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Jerry\JMRI\ [main]
2018-10-07 11:37:09,756 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Jerry\ [main]
2018-10-07 11:37:09,756 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
2018-10-07 11:37:12,232 gui3.TabbedPreferencesAction          INFO  - We are already waiting for the preferences to be displayed [main]
2018-10-07 11:37:12,497 PanelPro.PanelPro                     INFO  - Main initialization done [main]

bobjacobsen, 2018-10-07:
The C/MRI monitor seems to still have the older, obsolete Swing structure.  C.f. for the more modern approach java/src/jmri/jmrix/nce/ncemon/NceMonPanel.java

jerryg2003, 2021-03-15:
Working in 4.23.2 (and perhaps earlier).

closed 2021-03-15 by jerryg2003