JMRI/JMRI #5962: All internal defaults check doesn't consider systems that don't support any default

pabender, 2018-10-08:
When your system's only configured connection is \a system such as C/MRI or XBee that doesn't provide any of the defaults, the check for all internal provides a false positive.  In other words, it correctly says everything is set to internal, but doesn't realize it isn't possible for something other than internal to be selected.

KenC57, 2018-10-08:
Same issue as #5381 but with a more general description of systems effected.

pabender, 2018-10-08:
@KenC57 yes and no.

OpenLCB has an implementation of throttles and a programmer, so those be selectable as defaults.  I will see if I get the OpenLCB hardware setup tonight to verify that.

XBee doesn’t support throttles at all.  C/MRI technically supports at least analog throttles, but we don’t have an implementation of that.  Neither XBee or C/MRI support programming.

KenC57, 2018-10-08:
@pabender  I see what you mean. Well, I think there is a connection between the two issues. Likely the one fix will solve both. In theory, we may have a day where the decoder/roster pages are doing programming/configuring of the LCC nodes too. I suspect the key right now is that these systems are written to never pick any defaults and that is what confuses the normal logic.

pabender, 2018-10-08:
The fix for OpenLCB is to make sure it offers the defaults.  We can’t do more than that.