JMRI/JMRI #5992: Bachrus Speedometer - Invalid Reader Type

toddthetrainnut, 2018-10-13:
When running from the latest source code (downloaded 10/12/18) and with no changes, the Bachrus Speedometer is not working. As soon as the speedometer moves, JMRI throws Invalid reader type [AWT-EventQueue-0] errors from the SpeedoConsoleFrame as long as it is moving. The speedo still works correctly in the latest official release of JMRI. 

I worked on the Bachrus speedo a few months ago to build my own auto speed matcher. I got back into the code to try to get everything up to date to finally submit a pull request, but I found that the speedo had this problem after the merge. I've since checked out a clean repository on another machine and am having the same problem. I can try to dig into the problem myself, but I thought I'd ask if anyone knew about and was already fixing the problem.

Thanks!

bobjacobsen, 2018-10-13:
> As soon as the speedometer moves, JMRI throws Invalid reader type [AWT-EventQueue-0] errors from the SpeedoConsoleFrame as long as it is moving. 

Could you provide the complete trace back please?

Pugwash1, 2018-10-15:
See #6005

toddthetrainnut, 2018-10-15:
Sorry guys, had a busy weekend. Still need that stack trace?

Pugwash1, 2018-10-16:
@toddthetrainnut No. Easy to duplicate. Thx.

closed 2019-07-13 by toddthetrainnut