JMRI/JMRI #5995: DecoderPro reset to default and won't use my old roster

regularears, 2018-10-14:
Hi,

Hoping someone can help.

I'm using 4.12 on a Raspberry Pi with a SPROG.

When I opened DecoderPro today it gave me an error message about an invalid config. When I opened it up, all settings had returned to default and my roster disappeared.

I had been using a custom profile which I can still see in file explorer but when I click 'add existing' I can't browse to it because it's in the '.pi' folder which the open dialog can't see.

I copied the profile folder into my Documents and clicked 'Add Existing and browsed to it but just get an error 'unable to open profile' when I try.

I tried to import the roster into the existing blank profile by browsing to the roster.xml but nothing happens, no error message.

Please could someone give me some guidance on how I get my roster back?

regularears, 2018-10-14:
Hi,

A bit more info - this seems to be the issue according to log files

2018-10-14 09:59:05,341 profile.ProfileManager                ERROR - Error attempting to read Profile at /home/pi/.jmri/Train_set [main]
java.io.FileNotFoundException: /home/pi/.jmri/Train_set/profile.properties (No such file or directory)

The file does exist but appears to be empty. I've tried copying the contents of what appear to be backup files of that file in the same folder but it just seems to get emptied when I open DecoderPro and the errors persist.

regularears, 2018-10-14:
Ah actually copying the contents of the profile.properties file DID work - I was just copying it into the copied profile in my Documents folder, while DecoderPro was still trying to use the one in the hidden .pi folder.

Once I copied the file contents into the profile.properties in the .jmri folder it started up OK.

I don't know why or how this file got blanked I the first place though.

dheap, 2018-10-14:
As an aside, the "roster.xml" file doesn't contain your roster. It is simply a rebuildable index of the contents of the "roster" folder at the same level, which contains all your individual roster entry files.

> On 14 Oct 2018, at 8:41 PM, regularears <notifications@github.com> wrote:
> 
> I tried to import the roster into the existing blank profile by browsing to the roster.xml but nothing happens, no error message.

regularears, 2018-10-14:
Thanks. But would this not be the file to select when in the 'import roster' dialog? Or would I choose the roster folder?

bobjacobsen, 2018-10-14:
“import roster” is to bring in a file you created when you did an “export roster”. You didn’t do that, so it’s not a useful option.

If the individual roster files (the ones in the roster/ directory) are there, then you need to recreate the roster index.  That’s under the “Actions” menu in DecoderPro.

Bob

> On Oct 14, 2018, at 4:47 AM, regularears <notifications@github.com> wrote:
> 
> Thanks. But would this not be the file to select when in the 'import roster' dialog? Or would I choose the roster folder?
> 
> 

--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

closed 2021-04-14 by stale[bot]