JMRI/JMRI #6015: Roster headings changed in 4.13.3

brianjackson1, 2018-10-17:
up to version 4.13.2  the roster headings have been the same. 
 In 4.13.3 two of the headings have changed.

'Number' has become 'Road Number' and 'Operator' has become 'Road Name'

bobjacobsen, 2018-10-17:
I think "Operator" comes from using the en_GB locale.  Could you check that's still selected? It might have gotten lost in the update.

brianjackson1, 2018-10-18:
Just checked and yes I found it had changed to en_USA.  so have changed back to em_GB Should have looked there!
Not sure why it changed, but to be honest I did have a few issues trying to update.

When I updated and ran Decoder pro I clicked on New loco  but nothing happened.  The a fatal error message came up.  Unfortunately I didn't keep the Log I did notice JMRI launcher kept running in the background and I had to use Task manager to shut it down.

I uninstalled JMRI and ensured the folder had been deleted and still got the same issue.  I then deleted the JMRI in my users folder and did a clean install and all worked but the heading had changed.

It was after this that it changed.  I also think it was there with 4.13.2 but can't be certain.  
I did a completely clean install last night all seemed well so may have just been a glitch.