JMRI/JMRI #6026: Parallax RFID serial port incorrect baud

spassmo, 2018-10-19:
Configuring an RFID connection as Serial using Parallax protocol opens serial port at 9600 bps when it should be 2400 bps

[image]

closed 2018-10-20 by mattharris

mattharris, 2018-10-20:
@spassmo This has now been fixed by #6028 and should be part of the next development release of JMRI.

If you want to try a pre-release build, you can download one from the build server at http://builds.jmri.org/jenkins/job/Development/job/Packages/

Right now, this change is not yet available in current builds, but build 2860 or later should incorporate the change and be available within the next hour or so.

spassmo, 2018-10-22:
@mattharris  It's working now.  Thank you for fixing it so quickly.

mattharris, 2018-10-23:
@spassmo No problem & thanks for confirming.