JMRI/JMRI #6027: Jython BlockLister.py exception

icklesteve, 2018-10-20:
On first block lookup most of the block data is output to the window, then exception with

```
  File "<script>", line 260, in displaySegmentData
AttributeError: 'jmri.jmrit.display.layoutEditor.LayoutEditor' object has no attribute 'trackList'
```

Fails silently on any further valid Block Lookups.
Would have a go at fixing myself but not great with Jython ;-)

Steve.

dsand47, 2018-10-20:
The LayoutEditor code has had significant changes during the last couple of releases.  These are the required changes to fix the script:

```
     def displaySegmentData(self, b) :
         PanelMenu = jmri.jmrit.display.PanelMenu.instance()
         layout = PanelMenu.getLayoutEditorPanelList()
-        for s in layout[0].trackList.toArray() :
+        for s in layout[0].getTrackSegments():
             #self.msgText("Segment: " + s.getID() + " block: " + s.getBlockName() + "\n")
             if ((s.getBlockName() == b.getSystemName()) or (s.getBlockName() == b.getUserName())) :
                 if (s.getLayoutBlock() != None) :
-                    self.msgText("Segment: " + s.getID() + " LayoutBlock: " + s.getLayoutBlock().getID() + "\n")
-                if (s.getHidden()) :
-                    self.msgText("Segment: " + s.getID() + " is hidden.\n")
-                if (s.getMainline()) :
-                    self.msgText("Segment: " + s.getID() + " is Mainline.\n")
-                if (s.getDashed()) :
-                    self.msgText("Segment: " + s.getID() + " is dashed.\n")
+                    self.msgText("Segment: " + s.getId() + " LayoutBlock: " + s.getLayoutBlock().getId() + "\n")
+                if (s.isHidden()) :
+                    self.msgText("Segment: " + s.getId() + " is hidden.\n")
+                if (s.isMainline()) :
+                    self.msgText("Segment: " + s.getId() + " is Mainline.\n")
+                if (s.isDashed()) :
+                    self.msgText("Segment: " + s.getId() + " is dashed.\n")
         return
```

The changes are using a getter instead of getting the data directly (which is now **private** and has a different name), fixing **_getId_** case and replacing the attributes **getXyz()** with **isXyz()**.

I will let you make the change, otherwise I can do it.

icklesteve, 2018-10-20:
Thanks Dave,
I'll make the change and submit in a few hrs, will help my Jython understanding,
Steve

PR #6032 (2018-10-21): BlockLister update
Updates BlockLister.py to use TrackSegments from depreciated LE method.
Full credit to @dsand47 for providing the fix.
Resolves #6027
closed 2018-10-21 by bobjacobsen