JMRI/JMRI #6050: Create Test Release 4.13.5

bobjacobsen, 2018-10-23:
This is the next release in the 4.14 cycle. It's intended to be released in early November from HEAD of master. This is the last regularly-scheduled test release of this cycle.

bobjacobsen, 2018-11-21:
JMRI 4.13.5 has been released. Files are available in the GitHub release section.

closed 2018-11-22 by bobjacobsen

mattharris, 2018-11-22:
@bobjacobsen I don't recall seeing the release message on the jmriusers list (unless I missed it)

bobjacobsen, 2018-11-23:
I sent it, but apparently it didn’t get posted.  When I went to sign in, I got a message that “your account is bouncing” and a button to unbounce.  Looks like I’ve missed about 2 days of messages…..

Thanks for pointing it out.

Bob

> On Nov 22, 2018, at 11:48 AM, Matthew Harris <notifications@github.com> wrote:
> 
> @bobjacobsen I don't recall seeing the release message on the jmriusers list (unless I missed it)
>

bobjacobsen, 2018-11-28:
[image]

Ended early due to a deadlock:
```
    [java] Found one Java-level deadlock:
     [java] =============================
     [java] "Test Import Cars":
     [java]   waiting to lock monitor 0x00007fb316345228 (object 0x00000007af609a48, a java.lang.Object),
     [java]   which is held by "AWT-EventQueue-0"
     [java] "AWT-EventQueue-0":
     [java]   waiting to lock monitor 0x00007fb319133da8 (object 0x00000006c14eb250, a java.awt.Component$AWTTreeLock),
     [java]   which is held by "Test Import Cars"
     [java] 
     [java] Java stack information for the threads listed above:
     [java] ===================================================
     [java] "Test Import Cars":
     [java] 	at com.apple.laf.AquaFileSystemModel.getRowCount(AquaFileSystemModel.java:193)
     [java] 	- waiting to lock <0x00000007af609a48> (a java.lang.Object)
     [java] 	at javax.swing.JTable.getRowCount(JTable.java:2662)
     [java] 	at javax.swing.plaf.basic.BasicTableUI.createTableSize(BasicTableUI.java:1692)
     [java] 	at javax.swing.plaf.basic.BasicTableUI.getPreferredSize(BasicTableUI.java:1733)
     [java] 	at javax.swing.JComponent.getPreferredSize(JComponent.java:1659)
     [java] 	at javax.swing.ScrollPaneLayout.preferredLayoutSize(ScrollPaneLayout.java:495)
     [java] 	at java.awt.Container.preferredSize(Container.java:1794)
     [java] 	- locked <0x00000006c14eb250> (a java.awt.Component$AWTTreeLock)
     [java] 	at java.awt.Container.getPreferredSize(Container.java:1778)
     [java] 	at javax.swing.JComponent.getPreferredSize(JComponent.java:1661)
     [java] 	at java.awt.BorderLayout.preferredLayoutSize(BorderLayout.java:719)
     [java] 	- locked <0x00000006c14eb250> (a java.awt.Component$AWTTreeLock)
     [java] 	at java.awt.Container.preferredSize(Container.java:1794)
     [java] 	- locked <0x00000006c14eb250> (a java.awt.Component$AWTTreeLock)
     [java] 	at java.awt.Container.getPreferredSize(Container.java:1778)
     [java] 	at javax.swing.JComponent.getPreferredSize(JComponent.java:1661)
     [java] 	at java.awt.BorderLayout.preferredLayoutSize(BorderLayout.java:719)
     [java] 	- locked <0x00000006c14eb250> (a java.awt.Component$AWTTreeLock)
     [java] 	at java.awt.Container.preferredSize(Container.java:1794)
     [java] 	- locked <0x00000006c14eb250> (a java.awt.Component$AWTTreeLock)
     [java] 	at java.awt.Container.getPreferredSize(Container.java:1778)
     [java] 	at javax.swing.JComponent.getPreferredSize(JComponent.java:1661)
     [java] 	at javax.swing.BoxLayout.checkRequests(BoxLayout.java:484)
     [java] 	at javax.swing.BoxLayout.layoutContainer(BoxLayout.java:425)
     [java] 	- locked <0x00000007af67ac10> (a javax.swing.BoxLayout)
     [java] 	at java.awt.Container.layout(Container.java:1508)
     [java] 	at java.awt.Container.doLayout(Container.java:1497)
     [java] 	at java.awt.Container.validateTree(Container.java:1693)
     [java] 	at java.awt.Container.validateTree(Container.java:1702)
     [java] 	at java.awt.Container.validateTree(Container.java:1702)
     [java] 	at java.awt.Container.validateTree(Container.java:1702)
     [java] 	at java.awt.Container.validateTree(Container.java:1702)
     [java] 	at java.awt.Container.validate(Container.java:1628)
     [java] 	- locked <0x00000006c14eb250> (a java.awt.Component$AWTTreeLock)
     [java] 	at java.awt.Container.validateUnconditionally(Container.java:1665)
     [java] 	- locked <0x00000006c14eb250> (a java.awt.Component$AWTTreeLock)
     [java] 	at java.awt.Window.pack(Window.java:818)
     [java] 	at javax.swing.JFileChooser.createDialog(JFileChooser.java:821)
     [java] 	at javax.swing.JFileChooser.showDialog(JFileChooser.java:749)
     [java] 	at javax.swing.JFileChooser.showOpenDialog(JFileChooser.java:656)
     [java] 	at jmri.jmrit.operations.rollingstock.ImportRollingStock.getFile(ImportRollingStock.java:40)
     [java] 	at jmri.jmrit.operations.rollingstock.cars.tools.ImportCars.run(ImportCars.java:91)
     [java] "AWT-EventQueue-0":
     [java] 	at java.awt.Component.invalidate(Component.java:2921)
     [java] 	- waiting to lock <0x00000006c14eb250> (a java.awt.Component$AWTTreeLock)
     [java] 	at java.awt.Container.invalidate(Container.java:1585)
     [java] 	at javax.swing.JComponent.revalidate(JComponent.java:4853)
     [java] 	at javax.swing.JTable.tableRowsInserted(JTable.java:4482)
     [java] 	at javax.swing.JTable.tableChanged(JTable.java:4407)
     [java] 	at javax.swing.table.AbstractTableModel.fireTableChanged(AbstractTableModel.java:296)
     [java] 	at javax.swing.table.AbstractTableModel.fireTableRowsInserted(AbstractTableModel.java:231)
     [java] 	at com.apple.laf.AquaFileSystemModel$DoChangeContents.run(AquaFileSystemModel.java:458)
     [java] 	- locked <0x00000007af609a48> (a java.lang.Object)
     [java] 	- locked <0x00000007b7998060> (a java.lang.Object)
     [java] 	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
     [java] 	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:744)
     [java] 	at java.awt.EventQueue.access$400(EventQueue.java:97)
     [java] 	at java.awt.EventQueue$3.run(EventQueue.java:697)
     [java] 	at java.awt.EventQueue$3.run(EventQueue.java:691)
     [java] 	at java.security.AccessController.doPrivileged(Native Method)
     [java] 	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
     [java] 	at java.awt.EventQueue.dispatchEvent(EventQueue.java:714)
[... cut ...]
     [java] 	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
     [java] 	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
     [java] 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
     [java] 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
     [java] 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
     [java] 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
     [java] 
     [java] Found 1 deadlock.
```