JMRI/JMRI #6060: JMRI StartUp Wizard sets (almost) all connections to Internal.

dheap, 2018-10-24:
The JMRI StartUp Wizard creates a connection which works fine the first time you run it.

However, next time JMRI starts:
-  Everything except (possibly) Command Station has been set to Internal.
-  The new user is presented with the "Error initialising DecoderPro" window and is thrown into the Preferences->Defaults window, likely incomprehensible to a novice.

I tried this with various setups (NCE Power Pro, NCE Simulator, SPROG, SPROG Command Station).

Not a great new-user introduction to JMRI...

Marcfour, 2018-10-27:
Same results with Digitrax; PR3 or Locobuffer, DCS100, DCS50. Initially all is well but first restart, all on Internal (or all on Loconet depending on PR3 mode)  with a popup message to this effect.  Add to this the recurring  "The Defaults preferences are invalid " message one gets with the PR3 in standalone mode.

bobjacobsen, 2018-11-18:
C.f PR #6161

PR #6161 (2018-11-30): Update defaults and use of Internal system
This updates the creation and management of the automatically-defined Internal system and how it interacts with the default mechanism.

- The automatically-created Internal system still contains Lights, Reporters, Sensors, Turnouts but no longer contains Programmers, and Consist and Throttle managers. 
- If you want internals of those, add another system via the usual connections screen
- The automatic setting and changing of defaults while adding and removing systems has been simplified.
- Several changes have been made to how the DecoderPro startup  [... cut]
closed 2018-11-30 by bobjacobsen