JMRI/JMRI #6066: WiThrottle does not report current speed/direction or changes to speed/direction

pabender, 2018-10-25:
Two related issues:
1) The WiThrottle Server does not inform the clients of the current speed and direction of an address when acquiring an address.
2) When a WiThrottle device and another JMRI throttle both have control of the addresses, WiThrottle server ignores changes to speed and direction.

This appears to be intentional behavior, but it is also a bug because this SHOULD work, just like it does on any other JMRI throttle interface.

The relevant code for the Function buttons is: 
https://github.com/JMRI/JMRI/blob/e774ee2687efa3a0cd9846f57a9235d1ddaa05a9/java/src/jmri/jmrit/withrottle/ThrottleController.java#L220-L250

pabender, 2018-10-25:
Further research, MultiThrottleController appears to send both functions and the speedstep parameter:

https://github.com/JMRI/JMRI/blob/e774ee2687efa3a0cd9846f57a9235d1ddaa05a9/java/src/jmri/jmrit/withrottle/MultiThrottleController.java#L36-L99

It also has methods to send the current speed and current direction, but these don't appear to be called from anything in the class.

BHoffman351, 2018-10-26:
> On Oct 25, 2018, at 10:19 AM, Paul Bender <notifications@github.com> wrote:
> 
> Two related issues:
> 
> The WiThrottle Server does not inform the clients of the current speed and direction of an address when acquiring an address.
> When a WiThrottle device and another JMRI throttle both have control of the addresses, WiThrottle server ignores changes to speed and direction.
> This appears to be intentional behavior, but it is also a bug because this SHOULD work, just like it does on any other JMRI throttle interface.
> 
We may want to hold off on this.  I believe this was intentional for both Engine Driver and WiThrottle but I have to figure out why.

Regards,
Brett

pabender, 2018-10-30:
@BHoffman351 

The changes to forward the properties were already merged in.  That fixes the bug in the server, and WiThrottle at least appears to handle this correctly.

Among other things, not having the throttles receive the speed and direction properties makes it impossible for WiThrottle/EngineDriver to take over an already running locomotive because they don't know the speed or direction of that locomotive.

There really are only two potential issues here:
1) Rounding errors because the throttle back ends work in percent throttle and the front ends work in speed steps.
2) The throttle UI sending an update in response to a property change.

Because the WiThrottle server works in fixed 126 speed step mode, issue 1 basically is a non issue.

If 2 is an issue, that really needs to be fixed in WiThrottle and/or EngineDriver, as that is where the bug actually lies.

We had to deal with both of these issues in the JMRI throttle many moons ago.

Paul

BHoffman351, 2018-10-30:
> On Oct 29, 2018, at 9:40 PM, Paul Bender <notifications@github.com> wrote:
> Among other things, not having the throttles receive the speed and direction properties makes it impossible for WiThrottle/EngineDriver to take over an already running locomotive because they don't know the speed or direction of that locomotive.
> 
Upon acquiring a loco address, the WiThrottle Server sends state of all functions, speed, direction, and speed step mode.  This has been the case for many years and was happening up until at least 4.11.8.  It is in fact true that the server does not send updates to speed (and direction) as another throttle changes those.  Engine Driver does a periodic request for these values and handles it as Steve sees appropriate.
> There really are only two potential issues here:
> 
> Rounding errors because the throttle back ends work in percent throttle and the front ends work in speed steps.
> The throttle UI sending an update in response to a property change.
> Because the WiThrottle server works in fixed 126 speed step mode, issue 1 basically is a non issue.
> 
Agreed, it is handled correctly or very close where it has never been brought up as a flaw.
> If 2 is an issue, that really needs to be fixed in WiThrottle and/or EngineDriver, as that is where the bug actually lies.
> 
Number 2 is not an issue, both apps update speed and direction without sending it back to the server.

Issue number 3 is the one that we are looking into now.
> We had to deal with both of these issues in the JMRI throttle many moons ago.
>

BHoffman351, 2018-10-31:
> On Oct 25, 2018, at 10:19 AM, Paul Bender <notifications@github.com> wrote:
> 
> This appears to be intentional behavior, but it is also a bug because this SHOULD work, just like it does on any other JMRI throttle interface.

This would be correct that it was intentional behavior.  If JMRI sends the speed back to the device that is issuing the speed change, it causes the UI to jump.  (Not necessarily on all devices and configurations.)  The updated speed from JMRI comes after the user has passed that speed value.  As shown in the logging below, not only does the UI display jump, the speed control is erratic.  The log is a small sample of a deliberate user-created increase in speed.  OutStream is packets sent to JMRI, streamHandler shows the received.  Between are the UI display values.  
The sequence has become:	27, 28, 30, 27, 29, 39, 46, 50, 52, 46, 47, 54
The end speed step should have been approximately 72 for the distance that the touch moved on the iOS screen.

In my opinion, this behavior is better described as a “bug”.  Tracking a speed change created by another throttle I would consider an “enhancement”.

2018-10-30 21:48:09.887 WiTx.x[170:60b] -[InterfaceServerBrowser sendPacket:maxTries:] [Line 437] OutStream - MTAL1112<;>V27

2018-10-30 21:48:09.919 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 27, sliderValue: 70.750000

2018-10-30 21:48:09.939 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 28, sliderValue: 74.750000

2018-10-30 21:48:09.947 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 30, sliderValue: 78.750000

2018-10-30 21:48:09.967 WiTx.x[170:60b] -[InterfaceServerBrowser stream:handleEvent:] [Line 920] streamHandler: MTAL1112<;>V27
2018-10-30 21:48:10.016 WiTx.x[170:60b] -[VSlider adjustThrottleValue:] [Line 97] Value: 27
2018-10-30 21:48:10.017 WiTx.x[170:60b] -[VSlider adjustThrottleValue:] [Line 101] sliderValue: 68.343750, rosc: 2.531250
2018-10-30 21:48:10.019 WiTx.x[170:60b] -[VSlider drawHandle:] [Line 213] verticle: 27

2018-10-30 21:48:10.027 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 27, sliderValue: 72.343750

2018-10-30 21:48:10.097 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 29, sliderValue: 99.343750

2018-10-30 21:48:10.123 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 39, sliderValue: 116.343750

2018-10-30 21:48:10.145 WiTx.x[170:60b] -[InterfaceServerBrowser sendPacket:maxTries:] [Line 437] OutStream - MTAL1112<;>V46

2018-10-30 21:48:10.175 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 46, sliderValue: 127.343750

2018-10-30 21:48:10.196 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 50, sliderValue: 131.343750

2018-10-30 21:48:10.223 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 52, sliderValue: 135.343750

2018-10-30 21:48:10.263 WiTx.x[170:60b] -[InterfaceServerBrowser stream:handleEvent:] [Line 920] streamHandler: MTAL1112<;>V46
2018-10-30 21:48:10.297 WiTx.x[170:60b] -[VSlider adjustThrottleValue:] [Line 97] Value: 46
2018-10-30 21:48:10.298 WiTx.x[170:60b] -[VSlider adjustThrottleValue:] [Line 101] sliderValue: 116.437500, rosc: 2.531250
2018-10-30 21:48:10.302 WiTx.x[170:60b] -[VSlider drawHandle:] [Line 213] verticle: 46

2018-10-30 21:48:10.311 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 46, sliderValue: 119.437500

2018-10-30 21:48:10.374 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 47, sliderValue: 137.437500

2018-10-30 21:48:10.388 WiTx.x[170:60b] -[InterfaceServerBrowser sendPacket:maxTries:] [Line 437] OutStream - MTAL1112<;>V54

pabender, 2018-10-31:
> On Oct 25, 2018, at 10:19 AM, Paul Bender ***@***.***> wrote: 
> This would be correct that it was intentional behavior. If JMRI sends the speed back to the device that is issuing the speed change, it causes the UI to jump. (Not necessarily on all devices and configurations.) The updated speed from JMRI comes after the user has passed that speed value. As shown in the logging below, not only does the UI display jump, the speed control is erratic. The log is a small sample of a deliberate user-created increase in speed. OutStream is packets sent to JMRI, streamHandler shows the received. Between are the UI display values. The sequence has become:	27, 28, 30, 27, 29, 39, 46, 50, 52, 46, 47, 54 The end speed step should have been approximately 72 for the distance that the touch moved on the iOS screen.

Let me see if I can make something like this happen without the UI involved.

> In my opinion, this behavior is better described as a “bug”. Tracking a speed change created by another throttle I would consider an “enhancement”. 

The reason this tracking another throttle is a bug and not an enhancement is that’ll is inconsistent with other JMRI throttle interfaces.

pabender, 2018-10-31:
> On Oct 29, 2018, at 9:40 PM, Paul Bender ***@***.***> wrote: 
> Upon acquiring a loco address, the WiThrottle Server sends state of all functions, speed, direction, and speed step mode. This has been the case for many years and was happening up until at least 4.11.8. It is in fact true that the server does not send updates to speed (and direction) as another throttle changes those. Engine Driver does a periodic request for these values and handles it as Steve sees appropriate.

Well, that isn’t working, but I can explain why.

When a throttle is first acquired, it may be available before the command station tells JMRI what it thinks the current state of the throttle is.

So, when WiThrottle sends it’s update to the device it only correctly sends data  for a throttle which was already under JMRI’s control.  Newly created throttles get the defaults set by JMRI.

You may not see this on a LocoNet system, but the initial send isn’t sufficient on an XpressNet system.  The XpressNet initialization sequence takes several messages between the command station and throttle, but the throttle object itself handles that, hence the object is available before it is synchronized with the command station.  XpressNet is not unique in this respect.

pabender, 2018-10-31:
@BHoffman351 

I think I can explain the behavior you describe as "Jumping".

Both WiThrottle and Engine Driver appear to be sending speed updates for every adjustment of thier speed sliders.  I've been able to replicate this on both systems.

Ultimately, sending a message for each speed step causes JMRI to send some command to the command station for each one of those requests to change the speed.  On most systems we support, this means we end up queuing a new throttle bus command for each one of those changes when really all we really care about is sending the final speed.

So how do we fix that?  We either don't send all the speed steps to the throttle, or we ignore the property change information until we reach that last set speed.

As for why you don't end up with the same speed you think you should, there is a rounding error somewhere in the conversion from interface units to percent to speed steps.

It will take me a few days to work out the details, but I think I can fix that on the server side.

BHoffman351, 2018-10-31:
> On Oct 31, 2018, at 1:05 AM, Paul Bender <notifications@github.com> wrote:
> I think I can explain the behavior you describe as "Jumping”.
> 
Did the log I attached not explain the jump?  (That is a log from Xcode of a running WiThrottle app on a real device.)  Would a video be more helpful?
> Both WiThrottle and Engine Driver appear to be sending speed updates for every adjustment of thier speed sliders. I've been able to replicate this on both systems.
> 
NO. I will only speak for WiThrottle, but it does not, and that is also shown in the log that I attached.
> Ultimately, sending a message for each speed step causes JMRI to send some command to the command station for each one of those requests to change the speed. On most systems we support, this means [...]
See above.
> So how do we fix that? We either don't send all the speed steps to the throttle, or we ignore the property change information until we reach that last set speed.
> 
This is not meant as a nasty comment:
Is it that you do not want to hear from anyone else on this issue?  If so, let me know so I don’t waste my time.
> As for why you don't end up with the same speed you think you should, there is a rounding error somewhere in the conversion from interface units to percent to speed steps.
> 
Or we look at the log that shows the values getting out of sequence.  There is no rounding error.
> It will take me a few days to work out the details, but I think I can fix that on the server side.
> 
Or seven hours and you can try to patch this together again.

Perhaps rolling back the changes and figuring out a way that this will work not just for the server, but for the apps that connect to it as well.  The values that are currently shown in the logs would also cause an issue with a motor-driven potentiometer used as a speed control.

BHoffman351, 2018-10-31:
> On Oct 30, 2018, at 11:43 PM, Paul Bender <notifications@github.com> wrote:
> When a throttle is first acquired, it may be available before the command station tells JMRI what it thinks the current state of the throttle is.
> 
> So, when WiThrottle sends it’s update to the device it only correctly sends data for a throttle which was already under JMRI’s control. Newly created throttles get the defaults set by JMRI.
> 
> You may not see this on a LocoNet system, but the initial send isn’t sufficient on an XpressNet system. The XpressNet initialization sequence takes several messages between the command station and throttle, but the throttle object itself handles that, hence the object is available before it is synchronized with the command station. XpressNet is not unique in this respect.
> 
Okay, I see what you are referring to.  Thanks for pointing that out.  So it was not that the server did not send the speed and direction upon acquire, it was that the correct data was not available on all systems at that point in time.

Regards,
Brett

BHoffman351, 2018-10-31:
@pabender @mstevetodd 

> I would highly recommend that WiThrottle and Engine Driver be re-written to only send speed steps when the user pauses their motion on the slider, rather than basing this on how far the slider has traveled (whether this is every tick of the slider or every n ticks of the slider is irrelevant.) This is much more efficient from the point of view of both the network and interactions with the underlying system.

**Absolutely not.**

Eliminating control at the expense of tracking another throttle's changes is not acceptable.

Regards,
Brett

pabender, 2018-10-31:
> On Oct 31, 2018, at 1:05 AM, Paul Bender ***@***.***> wrote: I think I can explain the behavior you describe as "Jumping”.
> Did the log I attached not explain the jump? (That is a log from Xcode of a running WiThrottle app on a real device.) Would a video be more helpful?

I was able to replicate the issue.

The log you sent shows the speeds changing and I see multiple instances of the speed command being sent with different values. 

For me, it would have been better to have a A better log would have been a wireshark trace.

> This is not meant as a nasty comment: Is it that you do not want to hear from anyone else on this issue? If so, let me know so I don’t waste my time.

No, I want to hear from you and others, but I also want the bug fixed, and I am taking the time to do it.

> Or seven hours and you can try to patch this together again. Perhaps rolling back the changes and figuring out a way that this will work not just for the server, but for the apps that connect to it [...]

Try the changes I have in #6098.  I no longer see the jumping.

As I said earlier, we fixed these issues many moons ago on the JMRI throttle.  It didn’t show the jumping behavior but the ultimate cause was the same, fighting between the UI objects view of the current state and the underlying throttle objects view of the state as reported to that UI by the property change listener.

Ultimately, the UI ( or the server in this case ) needs to be smart enough to ignore property changes that it caused.  #6098 is the first step in making that happen.

pabender, 2018-10-31:
I'm attaching a pre #6098 wireshark conversation trace of an iPhone talking to the server to this message as a PDF.

I'll need to do this again because the  colors don't show, but it clearly shows that WiThrottle is sending multiple speed requests without waiting for the back end to catch up.  It also shows why the jumping occurs because property change events are intertwined with the throttle slider movements.

More when I have time to get back to this tonight.

[WiThrottleTraceBefore6098.pdf](https://github.com/JMRI/JMRI/files/2534659/WiThrottleTraceBefore6098.pdf)

bobjacobsen, 2018-10-31:
> Eliminating control at the expense of tracking another throttle's changes is not acceptable.

I hope the choice is not this binary. Best would be to have both tracking of changes _and_ find control when changing the speed.

But if only one is possible, I would also vote to prioritize the fine control via continuous movement of the throttle. That’s how I use it: I don’t lift off the throttle slider after every move, and I like the continuous response while making small adjustments.

pabender, 2018-10-31:
Ok, more digging while attempting to write a test to capture the behavior shown in the log I posted does show an issue still in #6098.

#6068 used the existing method to send the current speed of the throttle, and #6098 modified that method to ignore the property changes if we aren't looking at the last speed sent.  The ignore code needs to be moved up to the property change method itself, and we probably should be forwarding the property change value to the client instead of the current speed (which is true of other values as well).

pabender, 2018-10-31:
> But if only one is possible, I would also vote to prioritize the fine control via continuous movement of the throttle. That’s how I use it: I don’t lift off the throttle slider after every move, and I like the continuous response while making small adjustments.

My thought is that it should be possible to track the rate of the changes within WiThrottle or Engine Driver and adjust the number of speed steps sent accordingly.

In other words, If I zip from 0 to max I only want one speed change sent to the back end throttle.

If I creep from 0 to max, then more than one speed change should be acceptable, because the throttle back end has time to respond.

This should be true whether or not I lift my finger from the slider.

pabender, 2018-11-01:
@BHoffman351 @bobjacobsen @mstevetodd 

I think the current code in #6098 actually does what we want it to.  

In addition to the tests I put in place to make sure the observed sequence worked, I have also tested this with both Engine Driver and WiThrottle when connected to both a Lenz system and an XPressNet simulator.  I see no jumping throttle no matter how many speed updates are sent.

Watching what happens using Wireshark, there is a response generated by JMRI almost immediately after the speed is sent.  This response isn't explicitly generated by the protocol, but it is in response to the property changes generated by the back end.  I captured a few traces and can forward those if you like.

I would appreciate if one or more of you could pull the code in #6098 and verify it works for you before I merge it in.

Thanks!

pabender, 2018-11-02:
Ok, a further update.

While playing with a Roco Z21 system tonight, I was able to see this behavior (using Engine Driver tonight):
Move slider from 0 to 75.
Slider jumps back down to 0 and then steps back up to 75 (and it does end at 75).

So, this still isn't fixed, at least not for all systems.

However, I was also able to prove this isn't isolated to this code AND the cause is that we are sending multiple speed steps.  I was also able to replicate the behavior in the JMRI throttle.

The slider in the JMRI throttle has two modes of operation.  

In the default mode, a user drags the slider to where they want it, then releases the mouse button. On button release, a single speed command is sent.

In "Track Slider in Real Time" mode, the slider sends speed commands as the mouse moves.  In this mode, and this mode only, the JMRI throttle does the same thing that WiThrottle and Engine Driver do, and for exactly the same reason.

Why I see this with the Z21 and not the Lenz system I used yesterday is that the Z21 isn't as quick to respond when we make a change to the throttle.

BHoffman351, 2018-11-02:
> On Nov 2, 2018, at 1:30 AM, Paul Bender <notifications@github.com> wrote:
> 
> However, I was also able to prove this isn't isolated to this code AND the cause is that we are sending multiple speed steps.

The log trace shows that WiThrottle is not sending every speed step.  This is the same log that was included previously.  The only packets sent to JMRI are following the word “OutStream”.  The others are UI tracking or values received from JMRI.  There are also timestamps that indicate time.  This log shows that the iPhone:
sent V27 at 21:48:09.887
did not send 28, 30, 27, 29, 39
sent V46 at 21:48:10.145
did not send 50, 52, 46, 47
sent V54 at 21:48:10.388

The reason that the values are not sequential is because the changes made by you are sending a value back that WiThrottle is using to correctly update the UI.

I have made an attempt to fix this issue and am waiting to hear from Steve, when he gets a chance, to see how it works for him.

2018-10-30 21:48:09.887 WiTx.x[170:60b] -[InterfaceServerBrowser sendPacket:maxTries:] [Line 437] OutStream - MTAL1112<;>V27

2018-10-30 21:48:09.919 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 27, sliderValue: 70.750000

2018-10-30 21:48:09.939 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 28, sliderValue: 74.750000

2018-10-30 21:48:09.947 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 30, sliderValue: 78.750000

2018-10-30 21:48:09.967 WiTx.x[170:60b] -[InterfaceServerBrowser stream:handleEvent:] [Line 920] streamHandler: MTAL1112<;>V27
2018-10-30 21:48:10.016 WiTx.x[170:60b] -[VSlider adjustThrottleValue:] [Line 97] Value: 27
2018-10-30 21:48:10.017 WiTx.x[170:60b] -[VSlider adjustThrottleValue:] [Line 101] sliderValue: 68.343750, rosc: 2.531250
2018-10-30 21:48:10.019 WiTx.x[170:60b] -[VSlider drawHandle:] [Line 213] verticle: 27

2018-10-30 21:48:10.027 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 27, sliderValue: 72.343750

2018-10-30 21:48:10.097 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 29, sliderValue: 99.343750

2018-10-30 21:48:10.123 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 39, sliderValue: 116.343750

2018-10-30 21:48:10.145 WiTx.x[170:60b] -[InterfaceServerBrowser sendPacket:maxTries:] [Line 437] OutStream - MTAL1112<;>V46

2018-10-30 21:48:10.175 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 46, sliderValue: 127.343750

2018-10-30 21:48:10.196 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 50, sliderValue: 131.343750

2018-10-30 21:48:10.223 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 52, sliderValue: 135.343750

2018-10-30 21:48:10.263 WiTx.x[170:60b] -[InterfaceServerBrowser stream:handleEvent:] [Line 920] streamHandler: MTAL1112<;>V46
2018-10-30 21:48:10.297 WiTx.x[170:60b] -[VSlider adjustThrottleValue:] [Line 97] Value: 46
2018-10-30 21:48:10.298 WiTx.x[170:60b] -[VSlider adjustThrottleValue:] [Line 101] sliderValue: 116.437500, rosc: 2.531250
2018-10-30 21:48:10.302 WiTx.x[170:60b] -[VSlider drawHandle:] [Line 213] verticle: 46

2018-10-30 21:48:10.311 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 46, sliderValue: 119.437500

2018-10-30 21:48:10.374 WiTx.x[170:60b] -[VSlider drag:forEvent:] [Line 132] oldSpeedValue: 47, sliderValue: 137.437500

2018-10-30 21:48:10.388 WiTx.x[170:60b] -[InterfaceServerBrowser sendPacket:maxTries:] [Line 437] OutStream - MTAL1112<;>V54

pabender, 2018-11-02:
Brett,

Sent from my iPad

> On Nov 2, 2018, at 6:09 AM, Brett Hoffman <notifications@github.com> wrote:
> 
> 
> > On Nov 2, 2018, at 1:30 AM, Paul Bender <notifications@github.com> wrote:
> > 
> > However, I was also able to prove this isn't isolated to this code AND the cause is that we are sending multiple speed steps.
> 
> The log trace shows that WiThrottle is not sending every speed step.

I am not saying that WiThrottle sends every speedstep.  What I am saying is that when you move the slider between two points WiThrottle tends to send more than one speed change.  The exact same thing happens with the JMRI throttle when real time tracking is enabled.

> This is the same log that was included previously. The only packets sent to JMRI are following the word “OutStream”. The others are UI tracking or values received from JMRI. There are also timestamps that indicate time. This log shows that the iPhone:

Since this is the same log provided previously, it does not include the latest updates to #6098.  There is a test in this PR that proves the WiThrottle server sends no more property changes than speed changes sent in response to received speed changes from the devices.

> The reason that the values are not sequential is because the changes made by you are sending a value back that WiThrottle is using to correctly update the UI.

I can’t reproduce the extra values you are seeing.  Both with the current code in master and the code in #6098, I see a response to every speed request actually sent by the devices.

What I am seeing is that the slider jumps when the responses from the command station that generate the property changes are delayed, relative to when the slider was moved by the user, WiThrottle, Engine Driver, and the JMRI slider with real-time tracking enabled, jump in response to those property changes.  In all three cases, each jump corresponds to a speed setting actually 

If you are seeing different behavior, then the backend involved might be sending more property changes for some reason.

> I have made an attempt to fix this issue and am waiting to hear from Steve, when he gets a chance, to see how it works for him.

That’s fine, but in my message from last night my last message from last night indicated that the behavior in #6098 can be replicated on the JMRI throttle with real time tracking enabled.  Whatever fix we apply to the WiThrottle Server needs to be applicable in the JMRI throttle as well.

Paul

mstevetodd, 2018-11-02:
> I would appreciate if one or more of you could pull the code in #6098 and verify it works for you before I merge it in.

@pabender I pulled your pabender:master to my local system and tried it out. Tested EngineDriver with 2 phones, each with 6 throttles, and set speed units to 128. Plus one JMRI throttle window set to "real time tracking". Loconet Simulator.
Acquired same loco address on all 13 throttles.  Tried various combinations of sliding slider, jumping to specific speed on slider, and using buttons instead of slider. 
Seems to work fine. I can outrun the responses, but they are queued and the "other" sliders catch up within a few seconds.
Hopefully, I'll get a chance to repeat this with actual hardware. I have a Digitrax system and a PiSPROG I can connect to my layout.

BHoffman351, 2018-11-02:
> I would appreciate if one or more of you could pull the code in #6098 <https://github.com/JMRI/JMRI/pull/6098> and verify it works for you before I merge it in.

I pulled master just now which included the most recent changes and this video (unless it got removed from this message) shows one slide of the throttle from bottom to top.  That is really all it took, the video is not edited or clipped.  The UI is being updated from JMRI and staggering the control back down again.  I also have it on my YouTube channel at https://youtu.be/xqMislkWggM <https://youtu.be/xqMislkWggM>

It exhibits the exact same behavior as before.

When you get a moment, perhaps you can examine the branch:
BH_SpeedUpdates <https://github.com/BHoffman351/JMRI/tree/BH_SpeedUpdates>
I have only four lines of changes and it eliminates all jitter by not sending the property change back to the device that sent it.  (Not exactly, but it does not send the value to a device if that value matches the last one received from that device.)

Steve does have some concerns, but they appear to be related to the eStop command, which may be a separate issue that is being exposed by some heavy testing.  (Of 12 throttles, only 3-5 were getting the speed of zero when eStop is issued.)

> On Nov 2, 2018, at 8:45 AM, Paul Bender <notifications@github.com> wrote:
> 
> I am not saying that WiThrottle sends every speedstep. What I am saying is that when you move the slider between two points WiThrottle tends to send more than one speed change.

That is correct and behaves like any other throttle.  TO MY KNOWLEDGE, NO OTHER THROTTLE ONLY SENDS SPEED STEPS WHEN THE USER STOPS MOVING THE KNOB.  This is what allows a user to make speed changes smoothly.

> Since this is the same log provided previously, it does not include the latest updates to #6098. There is a test in this PR that proves the WiThrottle server sends no more property changes than [...]

The number of property changes sent is not the issue, it is when those values are sent.

> I can’t reproduce the extra values you are seeing. Both with the current code in master and the code in #6098, I see a response to every speed request actually sent by the devices.

That is because, as I said, it is an Xcode log of one of my devices.  That is my debugging log.  It is not exposed to the end user.  It is also showing the positioning updates to the drawing values of the UI.

> What I am seeing is that the slider jumps when the responses from the command station that generate the property changes are delayed, relative to when the slider was moved by the user, WiThrottle, [...]

This is exactly the problem as I originally described it.  The backend is not sending more property changes than what your changes are requesting.  It is simply that there is a propagation delay that takes longer than the touch screen updates the display.  The user has moved on, it is correcting to the last sent value.

Upon thinking about this, it is likely that not sending every speed change may be what displays the problem.  If every value was sent (instead of at .25 second intervals) then it would not jitter but would only lag smoothly behind.  Engine Driver appears to send more frequent steps, so it does not appear to jitter.  It can run behind and continue to move the slider after user input has stopped.

> That’s fine, but in my message from last night my last message from last night indicated that the behavior in #6098 can be replicated on the JMRI throttle with real time tracking enabled. Whatever [...]

It can be there as well, my change is very non-intrusive.

Regards,
Brett

pabender, 2018-11-03:
> When you get a moment, perhaps you can examine the branch: BH_SpeedUpdates <https://github.com/BHoffman351/JMRI/tree/BH_SpeedUpdates> I have only four lines of changes and it eliminates all jitter [...]

That is similar to the approach I am taking, but I have been comparing with the property change value.

> Steve does have some concerns, but they appear to be related to the eStop command, which may be a separate issue that is being exposed by some heavy testing. (Of 12 throttles, only 3-5 were getting the speed of zero when eStop is issued.)

I think I have the same concern Steve does with your solution.  Some systems we support don’t have an emergency stop command.  The result of sending a negative speed is that the speed is set to 0, so if either a throttle never sent a speed or if the last speed a particular throttle sent was actually 0, both newSpeed and the throttle’s speed will be 0, and WiThrottle throttle will not be updated.

I was thinking about this today while teaching a class today, and I came up with a solution that just might fix this.  

Instead of storing single values and filtering out property changes until we get to that value, we store a list of values sent in a queue.  

Since the property changes come in the same order as the speed change requests, we just remove the head of the list when the property change matches that value. 

When the list is empty, we forward the property changes to the throttle.

I am going to work on getting this in place

BHoffman351, 2018-11-03:
Okay, I give up.  I presented a solution that works.  It was simple.  It solved the original issue you raised.

I can see my opinion is not wanted.

> On Nov 2, 2018, at 8:09 PM, Paul Bender <notifications@github.com> wrote:
> 
> When you get a moment, perhaps you can examine the branch: BH_SpeedUpdates https://github.com/BHoffman351/JMRI/tree/BH_SpeedUpdates <https://github.com/BHoffman351/JMRI/tree/BH_SpeedUpdates> I have only four lines of changes and it eliminates all jitter by not sending the property change back to the device that sent it. (Not exactly, but it does not send the value to a device if that value matches the last one received from that device.)
> 
> That is similar to the approach I am taking, but I have been comparing with the property change value.
> 
> Steve does have some concerns, but they appear to be related to the eStop command, which may be a separate issue that is being exposed by some heavy testing. (Of 12 throttles, only 3-5 were getting the speed of zero when eStop is issued.)
> 
> I think I have the same concern Steve does with your solution. Some systems we support don’t have an emergency stop command. The result of sending a negative speed is that the speed is set to 0, so if either a throttle never sent a speed or if the last speed a particular throttle sent was actually 0, both newSpeed and the throttle’s speed will be 0, and WiThrottle throttle will not be updated.
> 
> I was thinking about this today while teaching a class today, and I came up with a solution that just might fix this.
> 
> Instead of storing single values and filtering out property changes until we get to that value, we store a list of values sent in a queue.
> 
> Since the property changes come in the same order as the speed change requests, we just remove the head of the list when the property change matches that value.
> 
> When the list is empty, we forward the property changes to the throttle.
> 
> I am going to work on getting this in place
>

pabender, 2018-11-03:
@BHoffman351 

> Okay, I give up. I presented a solution that works. It was simple. It solved the original issue you raised. I can see my opinion is not wanted.

I really do want you opinion.

There was just a small issue with the solution you presented because it didn't account for a throttle that had never had a speed set, which is what @mstevetodd had pointed out.

By using a queue to save more than one value, we avoid that issue, so I really am just expanding on what you suggested.

BHoffman351, 2018-11-03:
> On Nov 2, 2018, at 9:17 PM, Paul Bender <notifications@github.com> wrote:
> 
> There was just a small issue with the solution you presented because it didn't account for a throttle that had never had a speed set, which is what @mstevetodd <https://github.com/mstevetodd> had pointed out.

My apologies, I see what you are referring to.  An update will not be sent to any device that matches the last value received from it, even after the UI has moved to a new location.

Regards,
Brett

BHoffman351, 2018-11-03:
> On Nov 2, 2018, at 11:28 PM, Paul Bender <notifications@github.com> wrote:
> 
> I just updated #6098 <https://github.com/JMRI/JMRI/pull/6098> again with the queue implemented. This seems to work well in my testing.

So if I understand:
If a device sends an update, we should not send that property change value back to the phone as it causes the jitter due to delays.

If a device is already set to that speed value (which could be a result of above) we should not send the property change value to that device.  We do not need to change it, it is already correct.

BH_SpeedUpdates <https://github.com/BHoffman351/JMRI/tree/BH_SpeedUpdates>
It does not address the z21 issues, but handles the property changes in a delightful fashion.
I had previously missed including a line to set the last value to the property change value.

Regards,
Brett

pabender, 2018-11-03:
Sent from my iPad

> On Nov 3, 2018, at 9:10 AM, Brett Hoffman <notifications@github.com> wrote:
> > On Nov 2, 2018, at 11:28 PM, Paul Bender <notifications@github.com> wrote:
> > I just updated #6098 <https://github.com/JMRI/JMRI/pull/6098> again with the queue implemented. This seems to work well in my testing.
> 
> So if I understand:
> If a device sends an update, we should not send that property change value back to the phone as it causes the jitter due to delays.
> 
> If a device is already set to that speed value (which could be a result of above) we should not send the property change value to that device. We do not need to change it, it is already correct.

Both of those are correct, but there is one other caveat.  When the device sends more than one speed step that hasn’t been acknowledged due to the delays, we don’t want to update the UI before the last one is sent.

In other words, if we see a sequence like this:
Throttle sends speed step 20
Throttle sends speed step 40
Throttle sends speed step 60
Property change for speed step 20
Property change for speed step 40
Property change for speed step 60

We don’t want to forward the property changes for 20 or 40.  Sending the 60 is ok, but not required ( as you pointed out )

One of the things I added in #6098 was a test that kind of sequence produces the expected results.

> BH_SpeedUpdates <https://github.com/BHoffman351/JMRI/tree/BH_SpeedUpdates>
> It does not address the z21 issues, but handles the property changes in a delightful fashion.
> I had previously missed including a line to set the last value to the property change value.

I’ll pull that down when I get home later today.  It definitely is a simpler change than what I have done.

Paul

pabender, 2018-11-04:
@BHoffman351 

I pulled your branch, and I see the jumping with it when connected to the Z21.

I'm attaching two files.  

The first is a PDF of the stream between the iPhone and the WiThrottle Server. If this worked right data from the server is in blue and data from the iPhone is in red.

The second file is zip file containing a wireshark capture export of the TCP/IP conversation involved.  I'm attaching this just in case there is an issue with the PDF.

[WiThrottleZ21BHoffman.pdf](https://github.com/JMRI/JMRI/files/2545574/WiThrottleZ21BHoffman.pdf)

[WiThrottleZ21BHoffman.zip](https://github.com/JMRI/JMRI/files/2545575/WiThrottleZ21BHoffman.zip)

The important piece of the strean is the following lines:

```
MTAL1234<;>V4
MTAL1234<;>V110
MTAL1234<;>V126
MTAL1234<;>V4

*
MTAL1234<;>V110

MTAL1234<;>V126
```

The first three lines are the speeds sent by the iPhone in response to swiping from bottom to top on the slider.  The other V lines are the responses generated after the property change is sent to the server.

Paul

pabender, 2018-11-04:
and yes, I fixed the Z21 code in what I pulled from your branch before generating that capture.

pabender, 2018-11-04:
Here are two more trace files.  These are with the #6098 also talking to a Z21 with the same scenario as before (one slide quickly up the speed slider)

The first one I went quickly enough on the slider that the iphone only sent two speed changes.
In the second one, I went more slowly, and the iphone generated about 10 speed changes.

[WiThrottleZ21-6098-1.pdf](https://github.com/JMRI/JMRI/files/2545583/WiThrottleZ21-6098-1.pdf)
[WiThrottleZ21-6098-2.pdf](https://github.com/JMRI/JMRI/files/2545584/WiThrottleZ21-6098-2.pdf)

In neither case did the server actually send out a speed update as a result of property changes.

pabender, 2018-11-08:
I am ready to pull the WIP from #6098.  I think I have proven sufficiently that it eliminates the jumping WiThrottle/Engine Driver slider problem on all the systems I have access to.

pabender, 2018-11-09:
Closed via #6068 and #6098

closed 2018-11-09 by pabender