JMRI/JMRI #6078: SignalMast logic edit screen. Buttons are off the bottom of small screens.

Pugwash1, 2018-10-27:
User reports that on a 1290 x 800 screen you have to move the task bar to the side to get the update, apply buttons. I think it might need a borderlayout? or some such layout, to get it to resize nicely across all screen sizes

bobjacobsen, 2018-10-28:
How do you get this screen to display?

silverailscolo, 2018-10-28:
This pane can be opened by choosing "Edit Logic" from the Signal Mast Table and from that table select "Add Signal Logic". It clearly needs a bit of "Glue" - and perhaps a scroll bar around the main content - to keep the buttons in view on a smaller screen. Will look at it (soon).
[image]

Pugwash1, 2018-10-28:
This should be done with, I think,border layout. The buttons are fixed at bottom and the main tabs resize down to one line.     A  lot of autosizeing, no fixed sizes., and then it should scale from 640x480 to 4K.

On October 28, 2018 4:10:39 AM EDT, Egbert Broerse <notifications@github.com> wrote:
>This pane can be opened by choosing "Edit Logic" from the Signal Mast
>Table and from that table select "Add Signal Logic". It clearly needs a
>bit of "Glue" - and perhaps a scroll bar around the main content - to
>keep the buttons in view on a smaller screen. Will look at it (soon).
>![edit sml screen
>clipped](https://user-images.githubusercontent.com/14811327/47613508-d47e6880-da90-11e8-8860-cf080267fa89.png)

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

bobjacobsen, 2018-10-30:
Did #6087 fix this?

Pugwash1, 2018-10-31:
@bobjacobsen  Now works on a small screen linux lxde, so should be ok for windows

closed 2018-10-31 by Pugwash1