JMRI/JMRI #6107: Rotating a scaled icon is broken in 4.13.4

DanielBoudreau, 2018-11-04:
If you try to rotate a scaled icon (50% for example) certain number of degrees (45 for example), part of the icon image is chopped off.  Looking at the image, the rotation code doesn't keep the image centered in the view box.  Rotating an image using the "Orthogonal" tool seems to restore the image.  This bug was introduced after 4.13.3.  I believe the broken code is in NamedIcon.java

jcomuzzi, 2018-11-06:
This is due to my rotate/blinking change. I'll debug it.

jcomuzzi, 2018-11-06:
I think I have a fix for this. Running JUnit tests and some other testing and will submit a pull request shortly.

jcomuzzi, 2018-11-06:
I think I created pull request #6112 with the fix... Not sure though

jcomuzzi, 2018-11-06:
ok - Not clear why that failed JUnit test didn't fail for me. Let me debug that...

jcomuzzi, 2018-11-09:
Pull request #6112 has passed all checks and fixes this issue.