JMRI/JMRI #6119: Multiple common tests disabled in jmrix.loconet.LnSensorTest

bobjacobsen, 2018-11-08:
java/test/jmri/jmrix/loconet/LnSensorTest.java overrides and ignores a number of tests from the jmri.implementation.AbstractSensorTestBase common set of tests.

This _might__ be because the setUp is not complete.

danielb987, 2018-11-08:
LnSensor.setKnownState() sends a command on the LocoNet and the state of the sensor is not updated unless there is a LocoNet simulator that receives this message and handle it correctly. For me it seems that the LnSensorTest is not running a LocoNet simulator, but I'm not sure.

pabender, 2018-11-08:
> On Nov 8, 2018, at 9:43 AM, Daniel <notifications@github.com> wrote:
> 
> LnSensor.setKnownState() sends a command on the LocoNet and the state of the sensor is not updated unless there is a LocoNet simulator that receives this message and handle it correctly. For me it seems that the LnSensorTest is not running a LocoNet simulator, but I'm not sure.
> 
That is true, but I also don’t believe it should.

Bob knows this code better than I do, but I believe the LocoNet code is expecting to see the messages at least echoed back to it. 

Rather than using the simulator, the LnSensorTest should override the methods and use the LocoNetInterfaceScaffold to echo the messages back at the desired point in the test.  In particular, after a message is sent, the forwardMessage method can be used to force a command to be echoed back to the class under test.

Using this method gives us better control over message timing.

danielb987, 2018-11-08:
Thanks. The following code solved the problem.

```
// The minimal setup for log4J
@Before
@Override
public void setUp() {
    JUnitUtil.setUp();
    // prepare an interface
    lnis = new LocoNetInterfaceScaffold() {
        @Override
        public void sendLocoNetMessage(LocoNetMessage m) {
            log.debug("sendLocoNetMessage [{}]", m);
            // save a copy
            outbound.addElement(m);
            sendTestMessage(m);
        }
    };
    t = new LnSensor("LS042", lnis, "L");
}
```

bobjacobsen, 2018-11-18:
The fix by @danielb987 fixed this nicely.

closed 2018-11-18 by bobjacobsen