JMRI/JMRI #6124: HW Networked Connection Status main Panel

icklesteve, 2018-11-10:
Somewhere between 4.13.1 and 4.13.2 , the reporting of network connection status in main JMRI panel stopped working.

In 4.13.1 the text changes red with "ERROR" inserted into the connection when there's an error, in.
From 4.13.2 and in latest Head, the text remains black without the insertion when the connection fails.

Tested on actual MERG CBUS Network Connection, likely to also affect other CAN network connections?

icklesteve, 2018-11-18:
Have been doing a bit of tracing on this one,
It looks like jmrix/ConnectionStatus.java is getting 2 different connection names used with it.

ConnectionLabel.java (ConnectionConfig) connection.getConnectionName() stores and monitors changes to eg. "CAN via MERG Network Interface" ( I believe incorrect, should be something like "MERG 2"? )

while AbstractNetworkPortController.java  getusername() updates ConnectionStatus with eg. "MERG 2" ( 
I believe correct name to use ).

I spotted that jmrix\can\adapters\gridconnect\net\MergConnectionConfig.java
does not extend, MergConnectionConfig extends ConnectionConfig
so changed this to
MergConnectionConfig extends jmri.jmrix.AbstractNetworkConnectionConfig
and added a getConnectionName override, however has made no difference and the result is still "CAN via MERG Network Interface", not "MERG 2".

Any thoughts?  
My next move might be to submit a ConnectionStatus.java with more debug statements to it to properly demonstrate the issue?

silverailscolo, 2018-11-18:
The connectionName is used to create the Connection specific menu title and some of its item labels, so I would agree the long name is not very practical for that. This description belongs to some combobox on the config pane only. Can you compare with the way jmrix.lenz is structured?

icklesteve, 2018-12-12:
Thanks @silverailscolo , I compared with jmrix.lenz but still couldn't spot anything with the MERG connection.

I then spotted that Apps.java uses a different name to other jmrix which sets ConnectionStatus.java so swapped this to match the others on #6292 tested with network + USB connection

icklesteve, 2019-01-02:
Confirm fixed by #6292

closed 2019-01-02 by icklesteve